arendst/Tasmota #9081: [REQUEST] Zeroconf enabled by default in tasmota-zbbridge to allow automatic mDNS network discovery of Sonoff ZBBridge WiFi bridge

Hedda, 2020-08-13:
Please consider Zeroconf enabling automatic mDNS network discovery of tasmota-zbbridge on Sonoff ZBBridge.

Reason for me asking this feature is that it would allow an automagical initialization routine of the adapter configuration in ZHA (Zigbee Home Automation) integration component for Home Assistant which would make that installation procedure much more user friendly than having to manually enter URL for socket protocol plus IP-address and port of the Sonoff ZBBridge WiFi bridge with Tasmota:

https://www.home-assistant.io/integrations/zha/ 

**Have you looked for this feature in other issues and in the docs?**  

Yes

**Is your feature request related to a problem? Please describe.**  

Not a problem per say, however, I believe this would make ZHA's config flow with more user-friendly with tasmota-zbbridge.

**Describe the solution you'd like**  

Please enable mDNS for the tasmota-zbbridge image by default so it is always available in that firmware. @s-hadinger maybe?

Copy to HA ZHA maintainers: @dmulcahey or @Adminiuga Please consider enabling Zeroconf discovery in ZHA config flow.

**Describe alternatives you've considered**  

An alternative is to during ZHA config flow enter Sonoff ZBBridge "socket://[IP]:[PORT]" for example "socket://192.168.1.11:8888", so not a major effort but automagically discovering the WiFi-bridge would be awesome! 

It should be noted there is currently no information in the ZHA config flow that describes that you manually enter a socket URL, so I would argue that at least that part in the ZHA config flow is the opposite of being user-friendly.

**Additional context** 

As I understand it, Tasmota has mDNS function capability for Zeroconf automatic network discovery but it is disabled by default. I'm assuming that is because it is not normally used by many users, but I believe it makes perfect sense to have mDNS enabled by default in the tasmota-zbbridge version, especially when using it on the Sonoff ZBBridge with Tasmota serial-bridge together with the ZHA component for Home Assistant.  

Suggest to checkout his guide on how the configuration of adapters in ZHA is currently done manually and you can probably understand how the ZHA config flow could be made even more user-friendly if it could detect the bridge automatically:

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

**(Please, remember to close the issue when the problem has been addressed)**

Hedda, 2020-08-13:
Please also see this similar request for ZiGate WiFi bridge which has the exact same issue -> https://github.com/zigpy/zigpy-zigate/issues/36

blakadder, 2020-08-13:
If I understood correctly you're requesting mdns to be enabled by default to support a feature that doesn't exist yet in ZHA?

You're also forgetting the part that you need to enable TCP bridge in ZBBridge by using rules making the, potentially, automatic discovery still require manual configuration not to mention you cannot deal with the situation when a user changes the port in the rule.

s-hadinger, 2020-08-13:
I'm not in favor of this. From long timer experience MDNS on Tasmota generates a lot of problems. We will be focusing more on Z2T for out-of-the-box configuration.

Jason2866, 2020-08-13:
Since most routers supports static hosts entrys there is even no need for setting up a DNS server.
See example: 
[image]

Hedda, 2020-08-13:
> If I understood correctly you're requesting mdns to be enabled by default to support a feature that doesn't exist yet in ZHA?

That is correct! 

ZHA and zigpy developers were very fast to implement support for tasmota-zbbridge after it was added to Tasmota by @s-hadinger

It should be noted that Home Assistant already has mDNS support, it is just not yet implemented in the ZHA integration component.

> Since most routers supports static hosts entrys there is even no need for setting up a DNS server.

I don't think it is the mDNS part alone that is that only need but that in combination with Zeroconf (Zero-configuration networking) techniques which could possibly pass on port and port speeds for the serial-bridge service that ZHA users want it to connect automatically to

Jason2866, 2020-08-13:
@Hedda mDNS is NOT working reliable with Arduino ESP8266. As long this issue is not solved from Arduino ESP crew there will be no zbbridge precompiled build with activated mDNS. Since zigbee needs a fast reacting ESP82xx to work without issues, mDNS is a no go. With mDNS activated Tasmota is often not so responsive as it has to be. You can get not everything with a ESP82xx. Keep in mind this little CPU has very little resources and limited performance. Compromises have to been done.

Hedda, 2020-08-14:
> @Hedda mDNS is NOT working reliable with Arduino ESP8266. 

OK I understand. Do you know if you would work better om ESP32? I know at least one person is working a Zigbee ESP32 bridge

Jason2866, 2020-08-14:
About ESP32 and mDNS. I dont know. Since ZigbeeBridge is a ESP82xx device there will no work done on mDNS.

Hedda, 2020-08-14:
@s-hadinger Do you know if Zigbee2Tasmota will be supported at all on ESP32 when such Zigbee WiFi bridges become available?

I know some new Zigbee WiFi bridges is currently in the development and some are planned to have ESP32 and Silicon Labs EFR32.

Example of an existing one with ESP32 and Silabs is Wyze/Lockin lock gateway device https://lockinsmarthome.com/pages/lockE

Jason2866, 2020-08-14:
There is no tasmota-zbbridge ESP32 version.

arendst, 2020-08-14:
As the requested information has been provided I hereby close this issue.

closed 2020-08-14 by arendst

Hedda, 2021-04-26:
FYI, ZHA developers have now added Zeroconf discovery of Tube's Zigbee gateways based on ESP32 running ESPHome firmware:

https://github.com/home-assistant/core/pull/48420

I see @s-hadinger posted in https://github.com/tube0013/tube_gateways/issues/1 about adding Tasmota32 ZBBridge firmware support to that hardware.

@tube0013 makes both EFR32/EZSP and CC2652/ZStack variants based on his "Tube's Zigbee Gateways" open source hardware:

https://github.com/tube0013/tube_gateways

https://www.tubeszb.com/shop/coordinators/2

s-hadinger, 2021-04-26:
Correct. The Tube's  cc2652p2 is now officially supported. The EFR32 will be shortly. 

MDNS is untested on Tasmota32 and I'm not willing to try. So if you want to try on your side you are more than welcome.

Hedda, 2022-05-10:
FYI, PR for ZiGate's Zeroconf has recently been merged in Home Assistant's core as an example for adding additional to their side:

https://github.com/home-assistant/core/pull/68577

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/manifest.json#L85-L88

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/config_flow.py#L198-L201

https://github.com/home-assistant/core/blob/dev/tests/components/zha/test_config_flow.py#L80-L107

https://github.com/home-assistant/core/blob/dev/homeassistant/generated/zeroconf.py#L384-L389