arendst/Tasmota #9084: ESP32 [E][WebServer.cpp:617] _handleRequest(): request handler not found

ionultd, 2020-08-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I get this error when booting but booting, otherwise all seems to work.
**[E][WebServer.cpp:617] _handleRequest(): request handler not found**

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP Wroom 32
- [X] Tasmota binary firmware version number used: 8.4.0.1
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: ESPTool.py
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
19:16:09 RSL: stat/tasmota_02A374/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
19:16:10 RSL: stat/tasmota_02A374/RESULT = {"Module":{"1":"ESP32-DevKit"}}
19:16:10 RSL: stat/tasmota_02A374/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

19:17:51 RSL: stat/tasmota_02A374/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_02A374","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:17:51 RSL: stat/tasmota_02A374/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software reset CPU","Uptime":"0T00:16:22","StartupUTC":"2020-08-13T18:01:29","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2020-08-13T19:01:38","SaveCount":13}}
19:17:51 RSL: stat/tasmota_02A374/STATUS2 = {"StatusFWR":{"Version":"8.4.0.1(sensors)","BuildDateTime":"2020-08-12T17:09:02","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","CpuFrequency":80,"Hardware":"ESP32","CR":"357/699"}}
19:17:51 RSL: stat/tasmota_02A374/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WIFI",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000"]}}
19:17:51 RSL: stat/tasmota_02A374/STATUS4 = {"StatusMEM":{"ProgramSize":1118,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","00152005","B7FFBFCF","01DA9BC4","60360CC7","00084052"],"Drivers":"1,2,3,4,5,7,8,9,10,12,14,17,20,21,24,29,34","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,64,66,67,74"}}
19:17:51 RSL: stat/tasmota_02A374/STATUS5 = {"StatusNET":{"Hostname":"tasmota_02A374-0884","IPAddress":"10.10.0.45","Gateway":"10.10.0.1","Subnetmask":"255.255.255.0","DNSServer":"10.10.0.1","Mac":"30:AE:A4:05:A3:74","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:17:51 RSL: stat/tasmota_02A374/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_02A374","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:17:51 RSL: stat/tasmota_02A374/STATUS7 = {"StatusTIM":{"UTC":"2020-08-13T18:17:51","Local":"2020-08-13T19:17:51","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:41","Sunset":"20:08"}}
19:17:51 RSL: stat/tasmota_02A374/STATUS10 = {"StatusSNS":{"Time":"2020-08-13T19:17:51"}}
19:17:51 RSL: stat/tasmota_02A374/STATUS11 = {"StatusSTS":{"Time":"2020-08-13T19:17:51","Uptime":"0T00:16:22","UptimeSec":982,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"WIFI","BSSId":"A0:63:91:23:52:AF","Channel":13,"RSSI":100,"Signal":-19,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Termite console output here:

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
00:00:00 CFG: Loaded, Count 10
00:00:00 Project tasmota Tasmota Version 8.4.0.1(sensors)-1_0_4
00:00:00 WIF: Connecting to AP1 WIFI in mode 11  as tasmota_02A374-0884...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota_02A374-0884 with IP address 10.10.0.43
00:00:06 RSL: tele/tasmota_02A374/INFO1 = {"Module":"ESP32-DevKit","Version":"8.4.0.1(sensors)","FallbackTopic":"cmnd/DVES_02A374_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:06 RSL: tele/tasmota_02A374/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_02A374-0884","IPAddress":"10.10.0.43"}
00:00:06 RSL: tele/tasmota_02A374/INFO3 = {"RestartReason":"Software reset CPU"}
19:01:42 RSL: tele/tasmota_02A374/STATE = {"Time":"2020-08-13T19:01:42","Uptime":"0T00:00:13","UptimeSec":13,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":192,"MqttCount":0,"Wifi":{"AP":1,"SSId":"WIFI","BSSId":"A0:63:91:23:52:AF","Channel":13,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:06"}}
**[E][WebServer.cpp:617] _handleRequest(): request handler not found**
19:06:41 RSL: tele/tasmota_02A374/STATE = {"Time":"2020-08-13T19:06:41","Uptime":"0T00:05:12","UptimeSec":312,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"WIFI","BSSId":"A0:63:91:23:52:AF","Channel":13,"RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:06"}}
19:11:29 APP: Serial logging disabled

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Upload latest firmware tasmpota32-sensors.bin Tasmota Version 8.4.0.1(sensors)-1_0_4 then check boot messages

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Boot without any errors

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-08-14:
The error is caused by (mis)handling of not found links like in the below case ``favicon.ico``:

[image]

Tasmota does not provide a favicon.ico as it is overhead we won't need.

For some reason the ESP32 core webserver (that tasmota builds upon) errors out instead of handling it as a not found link.

I need to investigate how to handle this as the ESP8266 core webserver (again, that tasmota builds upon) seems to handle it well.

arendst, 2020-08-14:
The latest commit disables all ESP32 core debug messages like the ESP8266 does.

ionultd, 2020-08-14:
Thanks again for your awesome work!

closed 2020-08-14 by ionultd