arendst/Tasmota #9086: Support for multiple MAX31865 sensors (PT100/PT1000)

brvdboss, 2020-08-13:
Hello,

I would like to use multiple MAX31865 boards so I can use multiple temperature sensors of that type.  The current implementation only supports a single one.

**Have you looked for this feature in other issues and in the docs?**  

Yes
If I understood correctly, this is something that should be implemented in the driver for the sensorboard. ([xsns_47_max31865.ino](https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_47_max31865.ino))

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

I would like to use multiple sensors of this type with a single ESP board.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

I started with a quick implementation on this, yet as I'm not familiar with the code base I do have some questions and would like some advise before proceeding and submitting a PR.

Fr one device the following pins are used:
- Pin(GPIO_SSPI_CS),
- Pin(GPIO_SSPI_MOSI),
- Pin(GPIO_SSPI_MISO),
- Pin(GPIO_SSPI_SCLK)

An additional device needs a separate CS pin and the others are shared across devices.

Is is possible to configure multiple pins as CS and detect that from code which are assigned?  For testing purposes I have made an implementation which uses the defined GPIO_SSPI_DC as alternative, but I'm not sure if that's the best solution.  And what if we want more than two sensors, which others to use?

Apart from which pins to use, the basic adjusted implementation does work with two devices.  I tried finding other sensors in the repository that allowed multiple devices (I think the [DHT](https://github.com/arendst/Tasmota/blob/f4adce5fb15c4aa2375b8558ee66a73087a0d0b4/tasmota/xsns_06_dht.ino) one does) and tried to keep it similar.

Assumptions in this implementation:
- Same sensors are used (so not mixing the PT100 and PT1000 variant)
- All of them with the same configuration (not mixing 2/3/4 wire versions)

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Alternative: use multiple ESP devices, one for every sensor.  In my use-case they are all connected to the same heatpump system, so it makes sense to connect them to a single Tasmota device.

**Additional context**  
_Add any other context or screenshots about the feature request here._

N/A; see start of implementation attached. Would like advice on how to proceed so it aligns with project style guide and guidelines.
I did consult the [Sensor API documentation](https://tasmota.github.io/docs/Sensor-API/)

**(Please, remember to close the issue when the problem has been addressed)**  

[xsns_47_max31865.ino.txt](https://github.com/arendst/Tasmota/files/5071558/xsns_47_max31865.ino.txt)

device111, 2020-08-13:
Isn't it easier to you use the I2C sensors DS1624(1) or MCP9808? 
You can use up to 8 Sensors with 2 wires and you don't need a CS-Pin for every Sensor.

brvdboss, 2020-08-13:
> Isn't it easier to you use the I2C sensors DS1624(1) or MCP9808?
> You can use up to 8 Sensors with 2 wires and you don't need a CS-Pin for every Sensor.

The PT100 probes are installed and I cannot replace those with other temperature sensors. So, in my case no. (Nor do I think that the ones you refer to are water-proof).  Anyway, the sensors are there, And I want to read them :)

arendst, 2020-08-14:
The current implementation makes use of the Adafruit MAX31865 library which only supports 1 sensor. 

Supporting more would need a total rewrite of library and driver.

So for now you'll need an ESP8266 for every sensor.

brvdboss, 2020-08-14:
Uhm, not sure if we're saying the same. It's perfectly possible to create multiple instances in code and thus have multiple sensors connected to the board. (Each with a dedicated CS pin).  See the driver code attachment I added to the initial issue where I created an array of instances (2 in this case) and I can successfully use this. **I am currently running a working test setup using the two sensors for development.**: One ESP8266, multiple MAX31865 devices connected, and each reading the temperature.  (One Tasmota driver that manages multiple instances of the library). See screenshots of running version
[image]
[image]

This scales as long as you have a pin on the board you can use for each sensor.  I am willing to implement this and create a PR for you to review.

Practical question for me to continue:
- In the user interface I can assign multiple times the CS pin, but I don't think those can individually be used/identified later?
  - I can just use other identifiers already available, but I would assume that's not the best practice in this case.
  - Additional (generic)identifiers could be used, but I'm not sure if this is desirable (GPIO_SSPI_CS_0, GPIO_SSPI_CS_1, GPIO_SSPI_CS_2 etc)
  - or dedicated pin names like for the MAX31855; thus something like GPIO_MAX31865CS_0, GPIO_MAX31865CS_1 etc and GPIO_MAX31865MISO, GPIO_MAX31865MOSI, GPIO_MAX31865SCLK).  **Personally I think this is the cleanest option.**
- If different types of sensors (2/3/4 wire and PT100 and PT1000) are mixed, a specific define for the parameters for each would be required and more work is needed (and would probably make the code more messy). If we consider the scenario of multiple identical sensors, then there's not really an issue.

Thanks for your advice

joba-1, 2020-08-14:
Regarding pins: I would use generic miso, mosi and sclk (analog to i2c sda/scl) and a sensor instance specific cs.

brvdboss, 2020-08-15:
I provided a pull request with an implementation of this: https://github.com/arendst/Tasmota/pull/9103

support for up to 6 sensors (6 somewhat arbitrary because that's what I need right now)
dedicated pin names as suggested by @joba-1. Except maintaining the first pin using the standard CS-pin to avoid breaking existing installations that have it configured like that.

Assumes all sensors are of the same type and no mixing is in place (only one type of 2/3/4 wire and only PT100 or PT1000)

ascillato2, 2020-08-16:
> I provided a pull request with an implementation of this: #9103

Thanks.

closed 2020-08-16 by ascillato2

ChristopherGriffin, 2020-12-09:
This works amazingly well.  I have 3 sensors running right now.  I would request and enhancement to allow an error correction of + or - 5C in .1C or .5C increments.  I have noticed there is some variation between the cheaper probe manufacturers.

pebl-hank, 2021-04-17:
It does however not work for MAX6675. Can you implement the same solution for SetOption94=1?

pebl-hank, 2021-04-17:
> I provided a pull request with an implementation of this: #9103
>
> support for up to 6 sensors (6 somewhat arbitrary because that's what I need right now)
> dedicated pin names as suggested by @joba-1. [...]

It does however not work for MAX6675. Can you implement the same solution for SetOption94=1?

helmar74, 2023-07-06:
Hi I also want to use Multiple PT1000 sensors. I have connected 2 MAX31865 and I'm using 2-wire sensors.
My settings look like this:
[image]

I only get values of one of the sensors. The MAX31865 are working, as I did a cross check.
[image]