arendst/Tasmota #9087: ESP32-WROOM-32 dont work

christianwis, 2020-08-14:
### PROBLEM DESCRIPTION
I have an esp32-wroom-32 devkit v1.

Perform the following steps indicated in the readme:
-----------------------
These files are required to update Tasmota32 with esptool.py on an ESP32.

Command syntax to update Tasmota32 firmware on ESP32 via Esptool (replace COM port number!):

esptool.py --chip esp32 --port COM5 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dout --flash_freq 40m --flash_size detect 0x1000 bootloader_dout_40m.bin 0x8000 particitions.bin 0xe000 boot_app0 .bin 0x10000 tasmota32.bin
----------------

The transfer was successful, but it does not work, the tasmota32 AP does not appear in wifi to connect.

I think the bootloader is in loop.

### REQUESTED INFORMATION

It is not a sonnof, it is only an esp32 which I want to install the tasmota to use to activate and deactivate a relay.
```
  Configuration output here:

  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

see attach image
```

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

see attach image
[image]
[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

closed 2020-08-14 by close-issue-app[bot]

mahdy89, 2021-01-23:
same here
did you know how to fix it?

christianwis, 2021-01-25:
yes I can solve..

Open the .bin file with the notepad, if you can see HTML code it is
because the download is wrong.

You should not click "save as" to download the binary.

I hope it works for you ..
Christian

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>