arendst/Tasmota #9089: Add a new Status 0 Reporting set for shutters

stefanbode, 2020-08-14:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
Shutter configuration is most times not clear because the configuration is not reported with status 0
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Instead of reporting shutter configuration during startup it could be helpful if there is a new status message that outputs the shutter configuration on demand. 
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Catch the logfile from the beginning
**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-08-16:
Ok I will go for it. Just want to make sure that there are no complains by the master architect to do so.

stefanbode, 2020-08-20:
@arendst : If there a method that converts a number into the binary as a string. For example 6 -> "011". The rest is ready to submit. I want to make the options more readable.

arendst, 2020-08-20:
No ther is not. This (https://stackoverflow.com/questions/111928/is-there-a-printf-converter-to-print-in-binary-format) looks very useful at a minimal code cost. Especially this:

```
#define BYTE_TO_BINARY_PATTERN "%c%c%c%c%c%c%c%c"
#define BYTE_TO_BINARY(byte)  \
  (byte & 0x80 ? '1' : '0'), \
  (byte & 0x40 ? '1' : '0'), \
  (byte & 0x20 ? '1' : '0'), \
  (byte & 0x10 ? '1' : '0'), \
  (byte & 0x08 ? '1' : '0'), \
  (byte & 0x04 ? '1' : '0'), \
  (byte & 0x02 ? '1' : '0'), \
  (byte & 0x01 ? '1' : '0') 
printf("Leading text "BYTE_TO_BINARY_PATTERN, BYTE_TO_BINARY(byte));
For multi-byte types

printf("m: "BYTE_TO_BINARY_PATTERN" "BYTE_TO_BINARY_PATTERN"\n",
  BYTE_TO_BINARY(m>>8), BYTE_TO_BINARY(m));
```

stefanbode, 2020-08-20:
#9135 pull request generated

closed 2020-08-20 by stefanbode