arendst/Tasmota #9091: Increase measurement duration for windmeter (Sensor68)

Ingenieur89, 2020-08-14:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
Right now the measurement duration is fixed to 1 second. This means when there are no counts within 1 second the wind speed will be 0 even when the anemometer cups are rotating slowly.
I use a Davis anemometer which has a measurement duration of 2.5 seconds (according to datasheet). The minimum resolution for this gauge is 0.444 rotations per second (1.6 km/h).
With the current measurement duration of windmeter my minimum wind speed is 3.6 km/h.
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Be able to set the measurement duration with sensor settings.
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I thought about a solution but unfortunately I am not an experienced programmer (I could try anyway with some help).
One way to solve it would be to increase a counter in the main loop which will be executed every second (FUNC_EVERY_SECOND -> WindMeterEverySecond).
When a set amount of counts is reached the main loop to calculate wind speed is executed. The set amount of counts is the measurement duration in 1 second steps which can be set by a sensor setting.
**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

curzon01, 2020-08-20:
this is what the wind speed average value is intended for

the wind speed value is a momentary value, the avg is an exponentially weighted average over the [Teleperiod](https://tasmota.github.io/docs/Commands/#teleperiod) setting period (if Teleperiod is not set, the constant value WINDMETER_WEIGHT_AVG_SAMPLE is used - defaults to 150 sec).
The exponentially weighted average is a fairly smooth value over the period of Teleperiod seconds, which do not jump because 'sample counter is reset to 0' - once reache, the number of samples for avg calc is not reset.

so you can use the avg value and set the number of seconds used for that using the Teleperiod setting

ascillato2, 2020-08-20:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-08-20 by ascillato2

Ingenieur89, 2020-08-25:
I guess we have a missunderstanding here.

In my setup I use an anemometer with a higher radius for the cups (approx. 2.2 times bigger).
This means I will get less counts per second than with the default value (smaller radius).

If I understand it right, the momentary wind speed value is calculated every second (WindMeterEverySecond).

When I have less than 1 count per second the momentary wind speed is 0. With my bigger radius I have a big resolution gap between 0 and 1 count per second (which is quite much already).
By stretching the measuring duration for momentary wind speed from 1 second to like 3 seconds I can increase my lowest resolution because I can measure something like 1 count per 3 seconds.

I don't know how to explain it better. I downloaded the code and edited a little. The fix is working for me. Unfortunately I don't know how I can show the code here.

curzon01, 2020-08-26:
does it work legacy when setting WINDMETER_DEF_MEAS_DURATION to 1?
so you can make a PR (pull request) triggered from your repository

MrZ0r0, 2022-03-05:
Would like to try this code for a home weather station I want to build but am somewhat new to using Tasmota. How to I add/compile the above code to the tasmota-sensors.bin?

curzon01, 2022-03-05:
that's described in [TX20/TX23/WS2300-15 anemometer](https://tasmota.github.io/docs/TX2x/):
define sensor type you want in `user_config_override.h` and [compile your own build](https://tasmota.github.io/docs/Compile-your-build/)

PR #19542 (2023-10-06): Windmeter: add measure interval setting
## Description:
Original implementation was designed for a small cup anemometer (approx 6cm radius) where measuring speed every second was sufficient. As reported by @Ingenieur89 several months ago, when using an anemometer with a higher radius, is useful to delay speed computation, otherwise the speed may not get computed below a certain value.

New setting `windmeter_measure_intvl`, by default set to 1s (like before), allows to increase the resolution of the speed measurement. Using command `Sensor68 6, X` (where is X from 1 to 255), you can specify to [... cut]