arendst/Tasmota #9097: MQTT-SN support

ajlennon, 2020-08-14:
**Have you looked for this feature in other issues and in the docs?**  

Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

I work with LPWA (low power wide area) networking systems. We have very little bandwidth and need to minimise what we send.  MQTT is great for local WiFi but not so good over LPWA. MQTT-SN was developed as a lighter-weight solution to this problem.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

I'd like to see support for MQTT-SN which could be switched to from MQTT essentially transparently with just a #define 
 
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

There are other REST based systems, some which Tasmota may already support, but I don't see anything particularly viable as an alternative to MQTT-SN at this time

**Additional context**  
_Add any other context or screenshots about the feature request here._

I'm  happy to put some time into this. I am raising the issue as I don't want to reinvent the wheel is others are already looking at this or there are better alternatives to what I am thinking.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-08-15:
Could you give more details about the setup you foresee?

Esp8266 only speaks wifi which means you will need an intermediate gateway that could use MQTT-SN.

ajlennon, 2020-08-15:
So, the project leverages Nick O’Leary's PubSub Client for MQTT which is availabile in the Platform.io registry.

I think the first thing I should do is to take the Eclipse Paho reference implementation of MQTT-SN by Ian Craggs and modify it so it is available in the Platform.io registry. This feels a useful job of work in its own right.

MQTT is built on TCP. MQTT-SN is built on UDP. MQTT-SN is designed to have the same API to MQTT in as far as is possible.

So for next steps I am thinking I might modify  xdrv_02_mqtt to use the MQTT-SN API via some #defines. 

In terms of ESP8266 MQTT-SN is designed to work on anything that supports message sending and receiving, UDP being an obvious example so I see no issue in supporting MQTT-SN on ESP8266/ESP32 and similar devices.

ajlennon, 2020-08-15:
Had a bit of a rethink on using Paho and instead leveraged some great work by @arobenko at https://github.com/arobenko/mqtt-sn

The resulting library is here https://platformio.org/lib/show/10990/mqtt-sn

arendst, 2020-08-17:
I'm open for a PR. 

In addition to a new MQTT-SN driver I expect many changes to the current MQTT JSON response messages so I don't see aneasy implemntation.

ajlennon, 2020-08-18:
Thanks @arendst. I've been working away here and I have some alpha code which wraps the MQTT-SN protocol using the PubSubClient API so in theory it should just slot into what you already have for MQTT.

As I say it's alpha so I'm going to try to get some feedback on any issues before I suggest a PR.

Library is here (Platform IO publishing is broken at the moment but I'll publish this up asap)

https://github.com/DynamicDevices/mqtt-sn

closed 2020-09-20 by stale[bot]