arendst/Tasmota #9106: ESP32: Differentiate mqtt message for different switch

qwertangel, 2020-08-16:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
I am using ESP32 wroom Nodemcu module. I flashed tasmota32 and trying to configure is with different reed switches I have. When the reed switch triggers, I would like a MQTT message to be triggered. Currently even if I number buttons 1,2,and so on, the final MQTT message publishes everything under */POWER and it is not possible to distinguish which GPIO pin got triggered. On searching interwebs, it seems people are doing workaround like setting dummy relays etc.

**Describe the solution you'd like**  
1) The MQTT message should publish the button number. Tasmota configuration UI exposes the number 1-4, if that could be included somewhere as part of mqtt body message that will allow differentiation on which GPIO got triggered). 
2) Currently the number only ranges from 1-4. SInce esp32 has tons of GPIOs, it would be nice for this range to be extended to allow more than 4 inputs.

**Describe alternatives you've considered**  
Maybe the topic itself could include the number (e.g. POWER1, POWER2 etc)

**Additional context**  
[image]

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-08-16:
Hi,

You can make that behaviour using rules. Please, check the docs and if you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-08-16 by ascillato2

qwertangel, 2020-08-16:
Thanks. I did see people are managing it with dummy relays/rules. However, any reason devs think it should not be part of core feature ? For a device which has tons of input GPIO's, I believe it should be part of core tasmota functionality. Users should have to jump through hoops (rules) to read values from more than 1 input of same type.

ascillato, 2020-08-16:
Some users wants more mqtt messages, some other users want less mqtt messages. All depends on your setup.

Tasmota allows you to adapt its behaviour by the use of rules. Inside a rule you can set whatever you want.

If you only want a switch to be sent by mqtt, please check the command SWITCHTOPIC in the docs. Be aware of its fallback functionality too!

ascillato, 2020-08-16:
The power topic has nothing to do with switches. POWER is for RELAYs outputs. If you have define none, you need rules or switchtopic defined