arendst/Tasmota #9107: ESP32: Hall effect sensor

qwertangel, 2020-08-16:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
ESP32 includes a hall effect sensor. I don't see a way to read the value. 

**Describe the solution you'd like**  
It will be great for tasmota to read the value and be able to make it available (I will be sending it over MQTT if it exceeds a threshold using rules),

**Describe alternatives you've considered**  
None

**Additional context**  
None

**(Please, remember to close the issue when the problem has been addressed)**

Staars, 2020-08-28:
Just out of curiosity, what would your hardware setup look like?

I did a quick and dirty proof-of-concept driver to see the behaviour of the sensor. After applying a (Kalman) filter the readings are quite stable and I could sense a magnet through a piece of wood (like a table).
[image]

The graph shows the values with and without the filter (multiplied by 100). I am still playing ...

qwertangel, 2020-08-29:
i have an esp in garage to track/control my garage door/garage lights/parking sensors/mailbox. Since esp is housed near doorbell transformer, planning on using the hall effect sensor to detect doorbell ringing and send notifications.

Staars, 2020-09-13:
If you are still interested, here is a test version:
https://gist.github.com/Staars/09c24fe92880c5453d719df0372ee812

Just replace the file inside Tasmota and compile for ESP32.

There are a few things, that could be done differently. So I multiplied the measured hall sensor value by 10, after applying the filter, which be or may not be useful.

RULE would work like this:
`on ANALOG#Hall>100 do `...` endon`

samuelalexmclean, 2020-10-20:
Thanks stalebot, still a valid FR.

sfromis, 2020-10-20:
Did you try the test version posted by Staars?

samuelalexmclean, 2020-10-20:
Yup, worked for me. I was hoping this might become part of the main branch as a first class compile customization. Is that likely or is this a bit niche?

Staars, 2020-10-21:
If we find a niche in the source code ... 😁 

We will do some internal talks about it. This demo driver binds the hall sensor to the ADC-driver, which would make sense for me. An extra driver would be overkill for my taste.
I guess there is a chance to include this, as the ESP32 is no too space limited and it is a built-in chip feature.

Was the scaling okay for you? It was just a spontaneous decision and later I thought it might be better to go down by a power of ten.

closed 2020-12-02 by ascillato2

JustusW, 2021-06-27:
If you're not gonna support these sensors I suggest you remove them. Right now all the documentation does is trap people into thinking it _is_ supported when it really really isn't. You can't poll them outside of general status polling, you can't read them, and that means you can't normalize the very jittery values so it's unusable.

ErnieFR, 2024-01-14:
> If you are still interested, here is a test version: https://gist.github.com/Staars/09c24fe92880c5453d719df0372ee812
>
> Just replace the file inside Tasmota and compile for ESP32.
>
> There are a few [...]

How did you know the sensor was called "analog#hall" in Tasmota? I can't find any description of this sensor in the Tasmota docs, other than enabling gpio36 and gpio39 for halleffect-1 and halleffect-2. The sensor works allright, but if I try to publish it, nothing happens and I suspect that a rule like "on analog#hall#state do publish bla bla bla endon,  is wrong syntax?

barbudor, 2024-01-14:
Because staars added that mode into that modified file, it's not part of standard Tasmota code base
Note that this file is now very old and copying it roughly into current Tasmota wmay not work, or even prevent compiling

ErnieFR, 2024-01-14:
> Because staars added that mode into that modified file, it's not part of standard Tasmota code base Note that this file is now very old and copying it roughly into current Tasmota wmay not work, or even prevent compiling

Hey Barbador, how are you? Nice to see somebody is alive here :-) great. Found out the right syntax only don't understand why a rule does not react with "on ESP32#halleffect#state" as trigger

Staars, 2024-01-14:
Note the date:
created this gist on Sep 13, 2020.

I will delete this very old GIST, it is confusing at best nowadays.

ErnieFR, 2024-01-14:
> Note the date: created this gist on Sep 13, 2020.
> 
> I will delete this very old GIST, it is confusing at best nowadays.

THANKS at least that's solved !

barbudor, 2024-01-14:
@Staars May be it could be usefull to update that code and merge it ?

sfromis, 2024-01-14:
The standard built-in Hall effect support (different from what Staars did) is indeed `ESP32#HallEffect`, as you can see based on the SENSOR payload published, and also printed in the console.

Staars, 2024-01-14:
@barbudor  I assumed, that my old (and forgotten) demo  code is not needed anymore as this is now integrated as @sfromis pointed out.

ErnieFR, 2024-01-14:
> The standard built-in Hall effect support (different from what Staars did) is indeed `ESP32#HallEffect`, as you can see based on the SENSOR payload published, and also printed in the console.

Hey Sfromis, did not know I was being watched so closely :-)    But can you tell me why this does not work please : 
rule3 on esp32#halleffect#state do publish domoticz/in {"idx":190,"nvalue":0,"svalue":"%value%"} endon
rule3 on
==> I have a SG90 servo on a gpio pwm output, it changes position with a little magnet on its arm near the Wemos-Lolin

ErnieFR, 2024-01-14:
https://github.com/arendst/Tasmota/assets/60966258/2ab77575-2a82-49e0-a724-e662305bb14e

Here's little demo, if you think I should not do stuff like this here on github but rather do this at DiscordApp, please let me know. I do not want to do something wrong here.

barbudor, 2024-01-14:
Oh, that's for th ESP32 internal Hall sensor ?
I thought it was about an external one connected to an analog input

barbudor, 2024-01-14:
@ErnieFR Discussion is fine but not a good idea to dig up old discussion
Better to start a new one if you want

Discussion or Discord is up to you

Staars, 2024-01-14:
Only the built-in one.
What a lost code snippet can bring up 😁 .
(No problem, but now let's close this)

sfromis, 2024-01-14:
`rule3 on esp32#halleffect#state do` means that you are not using the suggested `ESP32#HallEffect`. Having `#state` is **only** relevant for certain types not appearing in JSON, and only as specifically suggested in the docs. The JSON parsing is very generic, and there `#state` is not relevant.

ErnieFR, 2024-01-14:
> `rule3 on esp32#halleffect#state do` means that you are not using the suggested `ESP32#HallEffect`. Having `#state` is **only** relevant for certain types not appearing in JSON, and only as specifically suggested in the docs. The JSON parsing is very generic, and there `#state` is not relevant.

Okay thanks for explaining that, I'll keep using "on Tele-ESP32#halleffect " then, it seems that there is always a few seconds delay anyway before the Hall Effect sensor shows the right value. The min logging of 10secs simply forces me to wait for the right value. Good thing I'm retired, haha. Cheers

s-hadinger, 2024-01-14:
Please keep in mind that Espressif is dropping support for the internal Hall effect. So it will soon be not available anymore.

sfromis, 2024-01-14:
I'd expect the feature to stay in the first ESP32 series (along with Ethernet support), but indeed not in the newer ESP32-Sx/Cx/Hx/whatever. And it's usually not a great spot for a hall effect sensor, anyway.