arendst/Tasmota #9114: Power1 toggle turns off i2c display (SSD1306) if display is on

DavinKD, 2020-08-17:
### PROBLEM DESCRIPTION
If the display is currently on and power1 is toggled, the display is turned off.  PWM1 exists which power1 should toggle.  

Turning on SetOption73 and using rules to toggle power1 results in the same issue.

This happens when toggling power1 through the web interface or by pressing button1

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): NodeMCU
- [ ] Tasmota binary firmware version number used: 8.4.0
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X ] IDE / Compiler used: Gitpod (needed binary with display and sensors)
- [X ] Flashing tools used: Tasmotizer
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```

20:26:09 MQT: stat/DE195D/RESULT = {"NAME":"DD-TEMP","GPIO":[17,255,18,255,6,5,255,255,46,4,255,56,255],"FLAG":15,"BASE":18}
20:26:09 MQT: stat/DE195D/RESULT = {"Module":{"0":"DD-TEMP"}}
20:26:10 MQT: stat/DE195D/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"18":"Button2"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"46":"PWM1i"},"GPIO13":{"4":"DS18x20"},"GPIO14":{"0":"None"},"GPIO15":{"56":"Led1i"},"GPIO16":{"0":"None"}}
20:26:13 MQT: tele/DE195D/STATE = {"Time":"2020-08-16T20:26:13","Uptime":"0T03:15:04","UptimeSec":11704,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER1":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dameron_smart","BSSId":"76:83:C2:B1:A0:12","Channel":6,"RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:03"}}
20:26:13 MQT: tele/DE195D/SENSOR = {"Time":"2020-08-16T20:26:13","BME280":{"Temperature":74.8,"Humidity":43.7,"DewPoint":51.2,"Pressure":727.5},"PressureUnit":"mmHg","TempUnit":"F"}
20

  Configuration output here:

```
- [X ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

20:28:05 MQT: stat/DE195D/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":376,"Free":135,"Rules":"on Rules#Timer=1 do Var16 0 endon on tele-BME280#Temperature do var1 %value% endon on tele-BME280#Humidity do var2 %value% endon on button2#state do add16 1 endon on tele-BME280#Pressure do var3 %value% endon on tele-BME280#DewPoint do var4 %value% endon on Var16#State==6.000 do Var16 0 endon on Var16#State==0 do displaytext [o] endon on button1#State do power1 toggle endon"}
20:28:05 RUL: Rule2 on Var16#State==1.000 do backlog displaytext [z]; displaytext [O]; displaytext [f2y1]Pool; displaytext [f2y21]Temp; displaytext [f2y42]%Var1%; displaytext[f0x90]~F8F; RuleTimer1 20 endon on Var16#State==2.000 do backlog displaytext [z]; displaytext [O]; displaytext [f2y1]Outside; displaytext [f2y21]Temp; displaytext [f2y42]%Var1%; displaytext[f0x90]~F8F; RuleTimer1 20 endon on Var16#State==3.000 do backlog displaytext [z]; displaytext [O]; displaytext [y1]Outside; displaytext [f2y21]Humid; displaytext
20:28:05 RUL: Rule2 +[f2y42]%Var2%%; RuleTimer1 20 endon
20:28:05 MQT: stat/DE195D/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":542,"Free":355,"Rules":"on Var16#State==1.000 do backlog displaytext [z]; displaytext [O]; displaytext [f2y1]Pool; displaytext [f2y21]Temp; displaytext [f2y42]%Var1%; displaytext[f0x90]~F8F; RuleTimer1 20 endon on Var16#State==2.000 do backlog displaytext [z]; displaytext [O]; displaytext [f2y1]Outside; displaytext [f2y21]Temp; displaytext [f2y42]%Var1%; displaytext[f0x90]~F8F; RuleTimer1 20 endon on Var16#State==3.000 do backlog displaytext [z]; displaytext [O]; displaytext [y1]Outside; displaytext [f2y21]Humid; displaytext [f2y4..."}
20:28:05 MQT: stat/DE195D/RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":360,"Free":151,"Rules":"on Var16#State==4.000 do backlog displaytext [z]; displaytext [O]; displaytext [f2y1]Outside; displaytext [f2y21]Press; displaytext [f2y42]%Var3% ; RuleTimer1 20 endon on Var16#State==5.000 do backlog displaytext [z]; displaytext [O]; displaytext [f2y1]Outside; displaytext [f2y21]Dew Pnt; displaytext [f2y42]%Var4%; displaytext[f0x90]~F8F; RuleTimer1 20 endon"}

```
- [X ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

20:28:31 MQT: stat/DE195D/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"DE195D","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:28:31 MQT: stat/DE195D/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T03:17:22","StartupUTC":"2020-08-17T00:11:09","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2020-08-15T23:50:53","SaveCount":187,"SaveAddress":"F9000"}}
20:28:31 MQT: stat/DE195D/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-08-15T21:51:11","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"379/699"}}
20:28:31 MQT: stat/DE195D/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dameron_smart",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["01008109","2805C8000100060000005A00000000000000","00800000","00006000"]}}
20:28:31 MQT: stat/DE195D/STATUS4 = {"StatusMEM":{"ProgramSize":643,"Free":360,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FFAE797","043683A1","000004CD","010013C0","C000F981","00004024"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,13,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,9"}}
20:28:31 MQT: stat/DE195D/STATUS5 = {"StatusNET":{"Hostname":"DE195D-6493","IPAddress":"192.168.0.90","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"CC:50:E3:DE:19:5D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:28:31 MQT: stat/DE195D/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.40","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DE195D","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:28:31 MQT: stat/DE195D/STATUS7 = {"StatusTIM":{"UTC":"2020-08-17T03:28:31","Local":"2020-08-16T20:28:31","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"-07:00","Sunrise":"21:45","Sunset":"12:03"}}
20:28:31 MQT: stat/DE195D/STATUS10 = {"StatusSNS":{"Time":"2020-08-16T20:28:31","BME280":{"Temperature":75.5,"Humidity":43.0,"DewPoint":51.4,"Pressure":727.5},"PressureUnit":"mmHg","TempUnit":"F"}}
20:28:31 MQT: stat/DE195D/STATUS11 = {"StatusSTS":{"Time":"2020-08-16T20:28:31","Uptime":"0T03:17:22","UptimeSec":11842,"Heap":19,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER1":"ON","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dameron_smart","BSSId":"76:83:C2:B1:A0:12","Channel":6,"RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [X ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

20:29:03 CMD: weblog 4
20:29:03 MQT: stat/DE195D/RESULT = {"WebLog":4}
20:29:04 CFG: Saved to flash at F8, Count 188, Bytes 4096
20:29:09 APP: Button2 multi-press 1
20:29:09 RUL: BUTTON2#STATE performs "add16 1"
[... 33 lines cut ...]
20:29:12 CMD: Group 0, Index 1, Command "POWER", Data "toggle"
20:29:12 MQT: stat/DE195D/RESULT = {"POWER1":"OFF"}
20:29:12 MQT: stat/DE195D/POWER1 = OFF
20:29:13 MQT: stat/DE195D/BUTTON1 = {"ACTION":"SINGLE"}
20:29:13 CFG: Saved to flash at F7, Count 189, Bytes 4096
[... cut ...]
_Steps to reproduce the behavior:_
Use a device with i2c display and 2 buttons and 1 pwm channel.  Turn the display on then toggle power1, display will turn off.

### EXPECTED BEHAVIOUR
State of the display should not change if power1 is toggled when there is a pwm channel controlled by power1.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

DavinKD, 2020-08-17:
Wanted to add another observation.  This only occurs when using displaytext [O] while power2 is off.  If instead I leave power2 on, toggling power1 does NOT turn off the display.

DavinKD, 2020-08-17:
I think the issue can be summed up in that using displaytext [O] and displaytex[o] does not update the corresponding power state.  Therefore when any of the powerstates are toggled, it thinks it should be on/off and changes it to match.  I've worked around this by changing my rules to use power2 1 and power2 0 instead of displaytext [O] and displaytext [o]

arendst, 2020-08-17:
What do you use the PWM for?

arendst, 2020-08-17:
Thx for your thorough debugging information. The latest commit in the development branch should solve your issue.

DavinKD, 2020-08-17:
The PWM is for some LEDs.  The board is installed in a dark place and it helps to have a bit of light handy.  Thanks for fix.

closed 2020-08-17 by ascillato2

sushantr5, 2021-02-26:
I am facing similar issue with ESP32. I have custom compiled 9.3.1 with USE_SCRIPT and USE_POWER_SECTION

Following is simple script:

`>D
>P
if pwr[1]>0
then
=>DisplayText Power1 ON
else
=>DisplayText Power1 OFF
endif

if pwr[2]>0
then
=>DisplayText Power2 ON
else
=>DisplayText Power1 OFF
endif

>BS

>R
print restarting now`

and following is configuration

Display I2C SSD_1306 128x32

GPIO17= Relay_i:1
GPIO18= Relay_i:2
GPIO25= I2C SCL
GPIO26= I2C SDA
GPIO27= Button_n:1
GPIO33= Button_n:2

Issue: Display Turns off when I use button to toggle power. Happens even when I turn of using web ui.

psi-4ward, 2023-12-13:
I've a similar issue using wemos d1 mini a pir and a I2C display. :(

ouinouin, 2024-01-14:
Hi, same behavior here with an ATOMS3 with 0,85 inch display and the universal driver, my guessing is that the buttons are binded to the  pin driving the display backlight. , my guessing is that the button has an action on the "backlight" pin (pin 16 selected as "backlight" in my case). changing the button to button 2 instead of button 1 doesnt fix the thing. setoption73 is neither working
21:09:14.310 RSL: RESULT = {"NAME":"ATOMS3DISPLAY","GPIO":[0,1,1,0,1056,1,1,1,6210,0,0,0,0,0,0,768,992,736,0,0,0,704,800,1024,0,0,0,640,608,0,32,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
@arendst should i open a new bug for that, or do I miss an undocumented option to not have this behavior ?