arendst/Tasmota #9126: Gosund mixed colors (red and green)

Fexiven, 2020-08-19:
### PROBLEM DESCRIPTION
The red and green color on my Gosund RGB Strip is switched

### REQUESTED INFORMATION
How can i switch the colors

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund LED Controller
- [x] Tasmota binary firmware version number used: 8.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
06:46:45 RSL: stat/tasmota/RESULT = {"NAME":"Tasmota","GPIO":[17,255,255,255,255,0,0,0,38,39,37,255,0],"FLAG":0,"BASE":18}
06:46:46 RSL: stat/tasmota/RESULT = {"Module":{"0":"Tasmota"}}
06:46:46 RSL: stat/tasmota/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"38":"PWM2"},"GPIO13":{"39":"PWM3"},"GPIO14":{"37":"PWM1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
06:45:46 RSL: stat/tasmota/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
06:45:46 RSL: stat/tasmota/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
06:45:47 RSL: stat/tasmota/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
06:48:01 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:48:01 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T08:23:12","StartupUTC":"2020-08-18T21:24:49","Sleep":50,"CfgHolder":4617,"BootCount":31,"SaveCount":357,"SaveAddress":"FB000"}}
06:48:01 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0.2(tasmota)","BuildDateTime":"2019-12-30T19:07:34","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"316/699"}}
06:48:01 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Coole WG",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008008","2805C8000100060000005A00000000000000","00000000","00000000"]}}
06:48:01 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":452,"Free":548,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0F400082","04048000","00000005","000000C0","00000080"],"Drivers":"1,2,3,4,9,10,12,16","Sensors":"1,2,3"}}
06:48:01 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-2005","IPAddress":"192.168.0.134","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"D8:BF:C0:0A:A7:D5","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
06:48:01 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0AA7D5","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
06:48:01 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Wed Aug 19 05:48:01 2020","Local":"Wed Aug 19 06:48:01 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"05:49","Sunset":"19:58"}}
06:48:01 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-08-19T06:48:01"}}
06:48:01 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-08-19T06:48:01","Uptime":"0T08:23:12","UptimeSec":30192,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":100,"Color":"EA00FF","HSBColor":"295,100,100","Channel":[92,0,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Coole WG","BSSId":"E0:28:6D:57:54:86","Channel":6,"RSSI":98,"Signal":-51,"LinkCount":2,"Downtime":"0T00:00:19"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just use a Gosund LED Strip with tasmota

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
red is red
green is green
blue is blue

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

mechanarchy, 2020-08-19:
The [suggested template](https://templates.blakadder.com/gosund_led_strip.html) will probably fix the issue. In the template you are currently using two of the PWM channels are swapped.

**Yours:** `[17,255,255,255,255,0,0,0,38,39,37,255,0]`, Note the `38,39,37`
**Template:** `[17,0,0,0,0,0,0,0,37,39,38,0,0]`, Note the `37,39,38`

Fexiven, 2020-08-19:
That was the problem. Thank you, have a nice day

closed 2020-08-19 by Fexiven