arendst/Tasmota #9127: IrReceived sent incorrectly after processing IRsend from MQTT

mechanarchy, 2020-08-19:
### PROBLEM DESCRIPTION
I have an IR bridge/blaster running Tasmota `8.4.0(ir)`. I can get it to send IR signals without issue using `IRsend` in the web console, via the HTTP endpoint, and over MQTT.

However, when using MQTT to run the command there is a erroneous `IrReceived` message generated. This is causing automations in Home Assistant to be triggered by that IR message when they shouldn't be (because no signal was actually received).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): [CSH-IRREM](https://templates.blakadder.com/connect_smarthome_CSH-IRREM.html)
- [x] Tasmota binary firmware version number used: `8.4.0(ir)`
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tuya-convert and then 'firmware upgrade by web server' to the IR version using the web console
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
06:37:09 CMD: Backlog Template; Module; GPIO 255
06:37:10 MQT: irbridge/stat/RESULT = {"NAME":"CSH IR Bridge","GPIO":[255,255,255,255,56,51,0,0,0,17,8,0,0],"FLAG":0,"BASE":62}
06:37:10 MQT: irbridge/stat/RESULT = {"Module":{"0":"CSH IR Bridge"}}
06:37:10 MQT: irbridge/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"56":"Led1i"},"GPIO5":{"51":"IRrecv"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"17":"Button1"},"GPIO14":{"8":"IRsend"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
- [x] Provide the output of this command: `Status 0`:
```
06:37:53 CMD: Status 0
06:37:53 MQT: irbridge/stat/STATUS = {"Status":{"Module":0,"DeviceName":"IRBridge","FriendlyName":["IRBridge"],"Topic":"irbridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:37:53 MQT: irbridge/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-ir.bin","RestartReason":"Power On","Uptime":"5T03:38:22","StartupUTC":"2020-08-14T01:59:31","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2020-08-13T12:51:23","SaveCount":31,"SaveAddress":"F9000"}}
06:37:53 MQT: irbridge/stat/STATUS2 = {"StatusFWR":{"Version":"8.4.0(ir)","BuildDateTime":"2020-07-29T15:22:16","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"427/699"}}
06:37:53 MQT: irbridge/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RedactedWifi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00000200","00006000"]}}
06:37:53 MQT: irbridge/stat/STATUS4 = {"StatusMEM":{"ProgramSize":568,"Free":432,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":0,"Features":["00000809","0F5A0582","04000001","00000002","00000000","00000000","00000000"],"Drivers":"1,2,4,5,7,9,10,12","Sensors":""}}
06:37:53 MQT: irbridge/stat/STATUS5 = {"StatusNET":{"Hostname":"irbridge-1517","IPAddress":"192.168.0.75","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"220.233.0.3","Mac":"2C:F4:32:70:25:ED","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
06:37:53 MQT: irbridge/stat/STATUS6 = {"StatusMQT":{"MqttHost":"homeassistant.local","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7025ED","MqttUser":"tasmota","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
06:37:53 MQT: irbridge/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-08-19T05:37:53","Local":"2020-08-19T06:37:53","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:49","Sunset":"19:57"}}
06:37:53 MQT: irbridge/stat/STATUS10 = {"StatusSNS":{"Time":"2020-08-19T06:37:53"}}
06:37:53 MQT: irbridge/stat/STATUS11 = {"StatusSTS":{"Time":"2020-08-19T06:37:53","Uptime":"5T03:38:22","UptimeSec":445102,"Vcc":3.485,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"Wifi":{"AP":1,"SSId":"RedactedWifi","BSSId":"50:78:B3:F3:7F:23","Channel":9,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
06:23:25 CMD: weblog 4
06:23:25 MQT: irbridge/stat/RESULT = {"WebLog":4}
06:23:26 CFG: Saved to flash at FA, Count 30, Bytes 4096

06:23:40 MQT: Received Topic "irbridge/cmnd/IRsend", Data Size 48, Data "{"Protocol":"NEC","Bits":32,"Data":"0x00FD956A"}"
06:23:40 SRC: MQTT
06:23:40 CMD: Group 0, Index 1, Command "IRSEND", Data "{"Protocol":"NEC","Bits":32,"Data":"0x00FD956A"}"
06:23:40 IRS: protocol 3, bits 32, data 0x16618858 (00FD956A), repeat 0
06:23:40 MQT: irbridge/stat/RESULT = {"IRSend":"Done"}
06:23:40 MQT: irbridge/tele/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FD956A","DataLSB":"0x00BFA956","Repeat":0}}

06:24:12 CMD: IRsend {"Protocol":"NEC","Bits":32,"Data":"0x00FD956A"}
06:24:12 SRC: WebConsole from 192.168.0.80
06:24:12 CMD: Group 0, Index 1, Command "IRSEND", Data "{"Protocol":"NEC","Bits":32,"Data":"0x00FD956A"}"
06:24:12 IRS: protocol 3, bits 32, data 0x16618858 (00FD956A), repeat 0
06:24:12 MQT: irbridge/stat/RESULT = {"IRSend":"Done"}

06:27:20 HTP: Command
06:27:20 SRC: WebCommand from 192.168.0.80
06:27:20 CMD: Group 0, Index 1, Command "IRSEND", Data "{"Protocol":"NEC","Bits":32,"Data":"0x00FD956A"}"
06:27:20 IRS: protocol 3, bits 32, data 0x16618858 (00FD956A), repeat 0
06:27:20 MQT: irbridge/stat/RESULT = {"IRSend":"Done"}

06:27:38 MQT: Received Topic "irbridge/cmnd/IRsend", Data Size 48, Data "{"Protocol":"NEC","Bits":32,"Data":"0x00FD956A"}"
06:27:38 SRC: MQTT
06:27:38 CMD: Group 0, Index 1, Command "IRSEND", Data "{"Protocol":"NEC","Bits":32,"Data":"0x00FD956A"}"
06:27:38 IRS: protocol 3, bits 32, data 0x16618858 (00FD956A), repeat 0
06:27:38 MQT: irbridge/stat/RESULT = {"IRSend":"Done"}
06:27:38 MQT: irbridge/tele/RESULT = {"IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0x00FD956A","DataLSB":"0x00BFA956","Repeat":0}}

06:27:52 CMD: weblog 2
06:27:52 SRC: WebConsole from 192.168.0.80
06:27:52 CMD: Group 0, Index 1, Command "WEBLOG", Data "2"
06:27:52 MQT: irbridge/stat/RESULT = {"WebLog":2}
```

### TO REPRODUCE
1. Install Tasmota IR firmware onto an IR blaster.
1. Use MQTT to call the `IRsend` command.

### EXPECTED BEHAVIOUR
The `IRsend` command executes and no `IrReceived` message is sent over MQTT.

### ADDITIONAL CONTEXT
The console output block above shows me performing the following sequence of events: running `weblog 4`; IRsend via MQTT; IRsend via web console; IRsend via HTTP endpoint; IRsend via MQTT; running `weblog 2`.

I was running these commands from a different room, and the position of the IR blaster and the other objects in the room didn't change. This rules out it reflecting off something.

Jason2866, 2020-08-19:
Probably your device receives parts of your just sended signal. 
To verify disable the IR_RECEIVE configured GPIO in your setup.
If there is still something received it is a issue. If not your device needs a better decoupling from
Ir transmitter and receiver

mechanarchy, 2020-08-20:
If I disable the IRrecv GPIO, the problem disappears. I'm not sure it's decoupling because the problem is extremely repeatable: it never appears with webconsole or HTTP endpoint calls, only when `irsend` is caused by MQTT. I've even tested some more exotic scenarios:

* MQTT `irbridge/cmnd/backlog` `irsend {...}` - problem exists
* Console `backlog irsend {...}` - no issue
* MQTT `irbridge/cmnd/publish` `irbridge/cmnd/irsend {...}` - problem exists
* Console `publish irbridge/cmnd/irsend {...}` - problem exists

When I involve a second Tasmota device:
* From the other device: `WebSend [irblaster] irsend {...}` - no issue
* MQTT `irbridge/cmnd/websend` `[tasmota2] publish irbridge/cmnd/irsend {...}` - problem exists
* MQTT `irbridge/cmnd/websend` `[tasmota2] websend [irblaster] irsend {...}` - intermittent
  * Interestingly, the problem exists but is not 100% repeatable here. By my theory it shouldn't happen, because the final `irsend` is running via the web... but I'm guessing it does, because the MQTT driver is active.

I'm confused, but I think these results are enough for reasonable doubt that poor decoupling is the root cause.

s-hadinger, 2020-08-20:
There is no difference in the code whether yiu send the command via console or MQTT. It is possible however that the physical environment is different and generates IR echoes. Or something else.

closed 2020-08-20 by ascillato2