arendst/Tasmota #9129: multi-client support for tasmota-client

garci66, 2020-08-19:
**Have you looked for this feature in other issues and in the docs?**  
Yes - found nothing (looking both for tasmota-slave / tasmota-client). The recent renaming makes it a bit more complex

**Is your feature request related to a problem? Please describe.**  
Its a feature request. I am doing some roller blinds automation but im using an optical encoder + servo (continuous) on each blind. The idea is to control 5 blinds in a row using a single tasmota. Each blind would have an arduino (maybe attiny85 if code fits) and tasmota send commands to each of the blinds.

**Describe the solution you'd like**  
Ideally.. I would love to use the tasmota-client where multiple clients can be configured on the bus. Using the serial bus in a multi-drop mode, we would need to identify each client with an ID and then each client responds to the id in question. Multiple reset lines (one per client) could be an option to allow for client-flashing and also client-address setting for example. Once tasmota-client supports multiple clients, I could rework the blinds code to send the target positions to each of the blinds, with the local arduino handling everything else

**Describe alternatives you've considered**  
A possibility would be to use the modbus library where tasmota is the client and each blind is its own server with a different modbus address.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-08-26:
Typically today you are not solving this with serial clients. The most common and very stable way to do this is using MQTT. The only thing is: You must install a MQTT broker on one device. A PI Mini is already suitable for it.

garci66, 2020-08-26:
Hi stefan. In my case the idea was to use arduino (pro-mini or simlar) at each curtain and have a "central ESP8266 control these curtains. Its just a group of 4 windows next to each other and for the curtains themselves it was simpler, for me, to code everything in a stand-alone arduino which is then driven by the ESP w/tasmota. the ESP is using mqtt to talk to home assitant.

What I did now is write a quite generic driver that uses sevo motors (but in continuous rotation mode) and an optical encoder to know the position of the curtain at all times. It receives commands over modbus so all 4 curtains are in a single "bus" (they are all next to each other, covering one huge window). So, over modbus, the ESP can send commands such as "move to 10%", move "one step up"  and can also poll the current position, etc... I implemented a few failsafes (such as stopping if the motor is triggered to move but it isnt moving after 1 second, based on the encoder feedback), or stopping if it didnt reach the target position after 90 seconds.

All the parameters are adjustable at runtime and it saves the current position in the eeprom at the end of each move, so that the system always "knows"  where it is at any point in time.

Below are two images on how the motors are building look like and here is the code for the "slave"
https://github.com/garci66/modbus-shutters/blob/move-to-modbus/serial-servo.ino (still a work in progress but anyhow).

[image]
[image]

I think we can close this issue and continue discussing in https://github.com/arendst/Tasmota/issues/9098

closed 2020-08-26 by ascillato2