arendst/Tasmota #9131: Dynamic button press listening time

golddragon007, 2020-08-19:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
I want to reduce the waiting time between the final button press and the action taken. Currently, there's around ~1 sec waiting time if you do a single press if multi-press actions are allowed. This is really long and noticeable when you enter into the room and you press the switch, then you need to wait 'a lot'. The biggest problem is in the hall when I press the button and I'm halfway to between the two switches... (Maybe I need to walk slower? :D )

**Describe the solution you'd like**  
Have the ability to set a moving frame/window for multi press actions. Let say, you have 200ms window to press the button again, You press first, within 200ms you press again, i.e. at 132ms, then you have again 200ms to press it again to have triple press action. If you stop, then the action will be taken after 332ms (132+200ms) from the first button press. On single press the waiting time is 200ms in this example. Therefore the waiting time can be significantly smaller, and the application will be in overall more responsive (at least in feeling). If it's not possible to make another click (because the click count was reached, then the action can be fired immediately). The window time needs to be adjustable.

**Describe alternatives you've considered**  
Don't have in my mind any other.

**Additional context**  
[image]

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-09-10:
Pretty cool idea. I had the same problem but there was only a SINGLE press. Here I was able with some setoptions and rules to overcome the issue. If we change it any way I would also dynamically change the window size. The 200ms could be lowered if a gamer is detected. e.g. define 300ms. then second press is in 130ms->window changes to 1.2x130ms. From the physics the second press take longest. All following are normally shorter.

Anyhow. Really good idea, because I also struggle with the multi-press and the "long time" the device needed to start the action.

stefanbode, 2020-09-11:
I checked last night a bit the button (support_button.ino) controller. Added a lot of debugging information to get clear timing information.

First finding: The Window for a second press is already assigned at the press of the button and the next press must be in the window. If you, for example only use 10% of the window the 90% gets resetted. so the behavior is already as you show in your second graph. 
Second finding: The window size is fixed to 0.5second and it will take at the end 0.5 seconds before anything happens.
All tested with DEFAULT setoptions and BUTTON1 defined.

golddragon007, 2020-09-11:
Then maybe the only thing that needs to be done is to allow the user to set the window length. For me, it feels actually more than .5s...

stefanbode, 2020-09-11:
I made one small change to coupe it with setoption32. A window size of 0.25 is unreachable. you cannot click that fast. 0.3 to 0.35 is possible. Anyhow you only get 0.15 seconds benefit. Not more. I checked with the internal LED D4 and the standard functionality. I'am pretty sure that your delay is coming from something else. Not from the ESP. Just use the D4 and assign it to RELAY3. then 3times pushing the button toggle the led. There is really not much delay. additionally the 0.5 seconds start counting when you pressing the button. when releasing it there is often already 0.15 gone

golddragon007, 2020-09-11:
I'm using these: https://templates.blakadder.com/moes_WS-EUY1.html and https://templates.blakadder.com/moes_WS-EUY2.html
I directly turning on the light with single press as the default is. And in that case, I feel an ~1-sec delay after I press the button. If I turn off the multi-press function with the correct option, then the light will turn on immediately.

stefanbode, 2020-09-11:
Her you see what the software is doing. The rest is your hardware. https://www.youtube.com/watch?v=cDf8MXArR9Y

golddragon007, 2020-09-11:
Can I get a bin file to install it on my spare switch to test it?

effelle, 2020-09-11:
Stefan is right, when I've updated the buttons I left the window set to 0.5. I've done some test with 0.4 and found pretty impossible to get more than 3 button press with lower values.  
Also if I'm correct your devices are using electro mechanical relays, then you will always have some sort of delay depending from the model, even if pretty minimal. The Sonoff Basic first version for example had a delay of about 20-25ms (it's a lot) but the latest R3 is just on a more average 5-10ms. 
Is not a big delay, in fact is just 0.01 sec max, but as Stefan said look how you hardware is assembled. Could it have an RC filter or any other sort of additional logic? 
To resume: multi press window +  command execution + relay delay = bigger time. 
Where I need a fast switch I use solid state relays or simply I don't use the multi press feature.

golddragon007, 2020-09-11:
"I've done some test with 0.4 and found pretty impossible to get more than 3 button press with lower values."
That means the window time is not being reset, as I said in the description of this issue. Every button press you need to have the same amount of time to press it again, except if you reached the maximum button press when the action will be taken immediately.
"Also if I'm correct your devices are using electro mechanical relays, [...]"
This doesn't explain why the action is fired immediately if the multi-press if turned off in the software!

golddragon007, 2020-09-11:
[button_demo.zip](https://github.com/arendst/Tasmota/files/5209720/button_demo.zip)
Here is it, open in a modern browser.

stefanbode, 2020-09-12:
You're simple wrong. Take a look at windows size in the code. This will set to 10 and count to 0. Count happens every 50ms. It will be set on button pressed. Not on button release. This is 100% sure

closed 2020-10-24 by stale[bot]