arendst/Tasmota #9132: Energy today gets reset on power cycle. (pzem004t - v3)

gammerboy, 2020-08-20:
### PROBLEM DESCRIPTION
The energy today gets reset on manual power cycle (turning off/on the power supply). We do have power cuts in our region like few times a day so the data keeps getting reset. The value in the energy yesterday stays unaffected (given that the power cut happens after it has reached 0000hrs). 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [.] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [.] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [.] Device used (e.g., Sonoff Basic): D1 mini
- [.] Tasmota binary firmware version number used: 8.4.0.2 
  - [ ] Pre-compiled
  - [.] Self-compiled
    - [ ] IDE / Compiler used: PlatformIO v1.10.0 on visual studio code v1.48.0 (Windows 10)
- [.] Flashing tools used: tasmotizer-1.1a
- [.] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
11:18:11 MQT: stat/PowerMonitoring_4B278B/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:18:11 MQT: stat/PowerMonitoring_4B278B/RESULT = {"Module":{"0":"Generic"}}
11:18:11 MQT: stat/PowerMonitoring_4B278B/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"62":"PZEM0XX Tx"},"GPIO2":{"0":"None"},"GPIO3":{"98":"PZEM016 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
11:19:38 MQT: stat/PowerMonitoring_4B278B/STATUS = {"Status":{"Module":0,"DeviceName":"PowerMonitoring1","FriendlyName":["PowerMonitoring1"],"Topic":"PowerMonitoring_4B278B","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:19:38 MQT: stat/PowerMonitoring_4B278B/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:17","StartupUTC":"2020-08-20T05:30:21","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2020-08-19T17:32:10","SaveCount":16,"SaveAddress":"F4000"}}
11:19:38 MQT: stat/PowerMonitoring_4B278B/STATUS2 = {"StatusFWR":{"Version":"8.4.0.2(tasmota)","BuildDateTime":"2020-08-19T15:01:44","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"412/699"}}
11:19:38 MQT: stat/PowerMonitoring_4B278B/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Home_Devices",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
11:19:38 MQT: stat/PowerMonitoring_4B278B/STATUS4 = {"StatusMEM":{"ProgramSize":545,"Free":456,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FCA2797","04100001","000000CD","010013C0","C0000800","00004000","00000000"],"Drivers":"1,2,3,4,5,8,9,10,12,20,21,29,35","Sensors":"1,2,3,5,6"}}
11:19:38 MQT: stat/PowerMonitoring_4B278B/STATUS5 = {"StatusNET":{"Hostname":"PowerMonitoring_4B278B-1931","IPAddress":"192.168.30.178","Gateway":"192.168.30.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.30.1","Mac":"2C:F4:32:4B:27:8B","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:19:38 MQT: stat/PowerMonitoring_4B278B/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.30.101","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4B278B","MqttUser":"hamqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:19:38 MQT: stat/PowerMonitoring_4B278B/STATUS7 = {"StatusTIM":{"UTC":"2020-08-20T05:49:38","Local":"2020-08-20T11:19:38","StartDST":"2020-10-25T02:00:00","EndDST":"2020-10-25T02:00:00","Timezone":99,"Sunrise":"10:20","Sunset":"00:25"}}
11:19:38 MQT: stat/PowerMonitoring_4B278B/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:19:38 MQT: stat/PowerMonitoring_4B278B/STATUS10 = {"StatusSNS":{"Time":"2020-08-20T11:19:38","ENERGY":{"TotalStartTime":"2020-08-19T17:32:10","Total":0.572,"Yesterday":0.558,"Today":0.014,"Power":3,"ApparentPower":18,"ReactivePower":18,"Factor":0.19,"Frequency":50,"Voltage":238,"Current":0.075}}}
11:19:38 MQT: stat/PowerMonitoring_4B278B/STATUS11 = {"StatusSTS":{"Time":"2020-08-20T11:19:38","Uptime":"0T00:19:17","UptimeSec":1157,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Home_Devices","BSSId":"8C:3B:AD:D7:C6:67","Channel":2,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
11:21:35 APP: Boot Count 9
11:21:35 CFG: Saved to flash at FA, Count 18, Bytes 4096
11:21:42 DMP: 01 04 14 09 5D 00 4A 00 00 00 20 00 00 0F 3C 00 01 01 F4 00 12 00 00 EF 5E
11:21:43 DMP: 01 04 14 09 5E 00 4A 00 00 00 21 00 00 0F 3C 00 01 01 F4 00 13 00 00 F8 5B
11:21:44 DMP: 01 04 14 09 5E 00 4A 00 00 00 21 00 00 0F 3C 00 01 01 F4 00 13 00 00 F8 5B
11:21:45 DMP: 01 04 14 09 5E 00 4A 00 00 00 22 00 00 0F 3C 00 01 01 F4 00 13 00 00 FD 98
11:21:46 DMP: 01 04 14 09 5E 00 4A 00 00 00 20 00 00 0F 3C 00 01 01 F4 00 12 00 00 AB 1A
11:21:47 DMP: 01 04 14 09 5F 00 4A 00 00 00 1F 00 00 0F 3C 00 01 01 F4 00 11 00 00 33 19
11:21:48 DMP: 01 04 14 09 5F 00 4A 00 00 00 21 00 00 0F 3C 00 01 01 F4 00 13 00 00 C5 A7
11:21:48 WIF: Checking connection...
11:21:49 DMP: 01 04 14 09 5F 00 4A 00 00 00 21 00 00 0F 3C 00 01 01 F4 00 13 00 00 C5 A7
11:21:50 DMP: 01 04 14 09 5F 00 4A 00 00 00 20 00 00 0F 3C 00 01 01 F4 00 12 00 00 96 E6
11:21:51 DMP: 01 04 14 09 5E 00 4A 00 00 00 20 00 00 0F 3C 00 01 01 F4 00 12 00 00 AB 1A
11:21:52 DMP: 01 04 14 09 60 00 4A 00 00 00 20 00 00 0F 3C 00 01 01 F4 00 12 00 00 8E F7
11:21:53 DMP: 01 04 14 09 60 00 4A 00 00 00 20 00 00 0F 3C 00 01 01 F4 00 12 00 00 8E F7
11:21:54 DMP: 01 04 14 09 60 00 4A 00 00 00 22 00 00 0F 3C 00 01 01 F4 00 13 00 00 D8 75
11:21:55 DMP: 01 04 14 09 60 00 4A 00 00 00 21 00 00 0F 3C 00 01 01 F4 00 13 00 00 DD B6
11:21:56 DMP: 01 04 14 09 60 00 4A 00 00 00 20 00 00 0F 3C 00 01 01 F4 00 12 00 00 8E F7
11:21:57 DMP: 01 04 14 09 60 00 4A 00 00 00 20 00 00 0F 3C 00 01 01 F4 00 12 00 00 8E F7
11:21:58 DMP: 01 04 14 09 60 00 4A 00 00 00 20 00 00 0F 3C 00 01 01 F3 00 12 00 00 3B 37
[... 26 lines cut ...]

### TO REPRODUCE
_Steps to reproduce the behavior:_
Poweroff and poweron the power supply to the module. 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Expect it to retain energy today value after a hardware power cycle or get the last reading from HA or similar hub and resume the counter from that value.  

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Before power cycle:
[image]
[image]

After power cycle:
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Using hardware serial. Tried sw serial which was also having issues and switched to hw serial on recommendation found on another post. 
The value does not seem to be affected when using the Software/System restart.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-08-20:
Closing this issue as it is **duplicated**. Sorry.

Please, do a search in issues for full explanation. 

As summary, the today value is saved to flash ONLY at midnight to not wear the flash. The flash memory has a limited amount of writes. 

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-08-20 by ascillato2

gammerboy, 2020-08-20:
Sorry, Was just a bit confused given that the value was being retained with software restart but now that i think about it I guess power is being supplied to retain in the volatile memory, maybe.

Anyways, Thank you for the clarification.

jerry1333, 2022-08-17:
Is there a way to manually save EnergyTotal data (like command)?

sfromis, 2022-08-17:
Anytime settings data are saved, those numbers are saved too, unless you've turned off saving. That includes the SaveData command, and restarting by command. Beware that if you intend to do it many times a day, that this will contribute the flash chip weardown, but that should not be a real issue if only occasionally.

jerry1333, 2022-08-18:
This is a problem, every now and then I have a power cut for a second / two and Tasmota loses today's values etc.
I figured a rule should run every 6 hours and write data (assuming 10,000 write cycles / 4 writes a day is about 7 years old - that's a lot).

By default, SaveData is set to 1, but I assume that if no settings are changed, nothing will be saved. So every 6 hours I have to change an irrelevant option?

I also don't understand why it can't be an option in Tasmota - let everyone decide for themselves whether they want to save data more often or extend memory life.

sfromis, 2022-08-18:
What I do is to just not rely on saved totals, but do my historical energy monitoring based on the payloads reported via MQTT, meaning that I can only miss out on up to TelePeriod seconds for an outage. If I prioritized keeping the local totals up to date, I could have an automation to issue commands to restore values based on last telemetry data reported.

jerry1333, 2022-08-18:
Not using MQTT, data is pulled by telegraf from json into influxdb. Maybe I think about using MQTT instead, will see.
Thanks for reply.