arendst/Tasmota #9134: Shelly 2.5 - MQTT send sensor update on switch change

GiorgioRoma, 2020-08-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _Shelly 2.5____
- [ X] Tasmota binary firmware version number used: __8.4.0___
  - [ X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ X] Flashing tools used: ___esptool__
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
10:49:42 CMD: Backlog Template; Module; GPIO 255
10:49:42 MQT: home/primopiano/lucicorridoio_lucescale/RESULT = {"NAME":"Shelly 2.5","GPIO":[56,255,17,255,21,127,0,0,6,128,5,22,156],"FLAG":2,"BASE":18}
10:49:42 MQT: home/primopiano/lucicorridoio_lucescale/RESULT = {"Module":{"0":"Shelly 2.5"}}
10:49:42 MQT: home/primopiano/lucicorridoio_lucescale/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"Nessuno"},"GPIO2":{"17":"Pulsante1"},"GPIO3":{"0":"Nessuno"},"GPIO4":{"21":"Relè1"},"GPIO5":{"127":"Pulsante2in"},"GPIO9":{"0":"Nessuno"},"GPIO10":{"0":"Nessuno"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"128":"Pulsante3in"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"22":"Relè2"},"GPIO16":{"156":"ADE7953 - IRQ"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
10:51:32 CMD: status 0
10:51:32 MQT: home/primopiano/lucicorridoio_lucescale/STATUS = {"Status":{"Module":0,"DeviceName":"Luci Corridoio Luce Scale Primo Piano","FriendlyName":["Luci Corridoio primo piano","Luce scale primo piano"],"Topic":"primopiano/lucicorridoio_lucescale","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,3,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:51:32 MQT: home/primopiano/lucicorridoio_lucescale/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T15:43:25","StartupUTC":"2020-08-19T17:08:07","Sleep":50,"CfgHolder":4617,"BootCount":39,"BCResetTime":"2020-08-07T17:11:01","SaveCount":295,"SaveAddress":"F5000"}}
10:51:32 MQT: home/primopiano/lucicorridoio_lucescale/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-07-29T12:26:48","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"508/699"}}
10:51:32 MQT: home/primopiano/lucicorridoio_lucescale/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Vodafone-37853567",""],"TelePeriod":60,"Resolution":"55C180C0","SetOption":["000A880F","0A05C8000100060000005A00000000000000","00000200","00006000"]}}
10:51:32 MQT: home/primopiano/lucicorridoio_lucescale/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000410","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
10:51:32 MQT: home/primopiano/lucicorridoio_lucescale/STATUS5 = {"StatusNET":{"Hostname":"LuciCorridoioLuceScalaPrimoPian","IPAddress":"192.168.1.217","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"D8:F1:5B:12:EE:02","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:51:32 MQT: home/primopiano/lucicorridoio_lucescale/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.13","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_12EE02","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:51:32 MQT: home/primopiano/lucicorridoio_lucescale/STATUS7 = {"StatusTIM":{"UTC":"2020-08-20T08:51:32","Local":"2020-08-20T10:51:32","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:50","Sunset":"20:55"}}
10:51:32 MQT: home/primopiano/lucicorridoio_lucescale/STATUS9 = {"StatusPTH":{"PowerDelta":1,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
10:51:32 MQT: home/primopiano/lucicorridoio_lucescale/STATUS10 = {"StatusSNS":{"Time":"2020-08-20T10:51:32","ANALOG":{"Temperature":68.5},"ENERGY":{"TotalStartTime":"2020-08-07T17:11:01","Total":0.010,"Yesterday":0.001,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}}
10:51:32 MQT: home/primopiano/lucicorridoio_lucescale/STATUS11 = {"StatusSTS":{"Time":"2020-08-20T10:51:32","Uptime":"0T15:43:25","UptimeSec":56605,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Vodafone-37853567","BSSId":"2C:91:AB:D3:3F:43","Channel":11,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
10:52:19 CMD: weblog 4
10:52:19 MQT: home/primopiano/lucicorridoio_lucescale/RESULT = {"WebLog":4}
10:52:20 CFG: Salvato nella flash in F4, Conteggio 296, Byte 4096
10:52:20 MQT: home/primopiano/lucicorridoio_lucescale/STATE = {"Time":"2020-08-20T10:52:20","Uptime":"0T15:44:13","UptimeSec":56653,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Vodafone-37853567","BSSId":"2C:91:AB:D3:3F:43","Channel":11,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:05"}}
10:52:20 MQT: home/primopiano/lucicorridoio_lucescale/SENSOR = {"Time":"2020-08-20T10:52:20","ANALOG":{"Temperature":67.6},"ENERGY":{"TotalStartTime":"2020-08-07T17:11:01","Total":0.010,"Yesterday":0.001,"Today":0.000,"Period":0,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}
10:52:20 ADE: 5997430, 4473, [1773, -17, 633, 74], [1773, -10, 633, -2]
10:52:20 ADE: U 5997430, C 4473, I 0 + 0 = 0, P 0 + 0 = 0
10:52:21 ADE: 5999804, 4474, [1773, 2, 633, 55], [1773, -32, 633, -2]
10:52:21 ADE: U 5999804, C 4474, I 0 + 0 = 0, P 0 + 0 = 0
10:52:22 UDP: Packet (119)
10:52:22 ADE: 5999106, 4474, [1773, -45, 633, 12], [1773, 1, 633, -9]
10:52:22 ADE: U 5999106, C 4474, I 0 + 0 = 0, P 0 + 0 = 0
10:52:23 ADE: 6002513, 4474, [1773, 19, 633, 108], [1773, -24, 633, 10]
10:52:23 ADE: U 6002513, C 4474, I 0 + 0 = 0, P 0 + 0 = 0
10:52:23 UDP: Packet (119)
10:52:24 UDP: Packet (119)
10:52:24 ADE: 5998422, 4474, [1773, 42, 633, 60], [1773, 5, 633, 16]
10:52:24 ADE: U 5998422, C 4474, I 0 + 0 = 0, P 0 + 0 = 0
[... 258 lines cut ...]
11:02:23 MQT: home/primopiano/lucicorridoio_lucescale/RESULT = {"POWER2":"OFF"}
11:02:23 MQT: home/primopiano/lucicorridoio_lucescale/POWER2 = OFF
**11:02:27 CMD: PowerDelta 1**
11:02:27 MQT: home/primopiano/lucicorridoio_lucescale/RESULT = {"PowerDelta":1}
11:02:30 MQT: home/primopiano/lucicorridoio_lucescale/STATE = {"Time":"2020-08-20T11:02:30","Uptime":"0T15:54:23","UptimeSec":57263,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Vodafone-37853567","BSSId":"2C:91:AB:D3:3F:43","Channel":11,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:05"}}
**11:02:30 MQT: home/primopiano/lucicorridoio_lucescale/RESULT = {"POWER1":"ON"}
11:02:30 MQT: home/primopiano/lucicorridoio_lucescale/POWER1 = ON
11:02:33 MQT: home/primopiano/lucicorridoio_lucescale/MARGINS = {"MARGINS":{"PowerDelta":10}}
[... cut ...]
11:02:54 MQT: home/primopiano/lucicorridoio_lucescale/STATE = {"Time":"2020-08-20T11:02:54","Uptime":"0T15:54:47","UptimeSec":57287,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Vodafone-37853567","BSSId":"2C:91:AB:D3:3F:43","Channel":11,"RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:05"}}
11:02:54 MQT: home/primopiano/lucicorridoio_lucescale/RESULT = {"POWER2":"OFF"}
11:02:54 MQT: home/primopiano/lucicorridoio_lucescale/POWER2 = OFF

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-08-20:
As I mentioned on Google Groups - A rule using `Power<x>#State` and `Status 8`.

Please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://tasmota.github.io/docs/#/Home) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

GiorgioRoma, 2020-08-20:
Ciao Meingraham, yes, i saw your comment on the google groups, but in my opinion this is a bug, as sending the consumption sensor information is not sent if the state of switch 2 changes, but only if switch 1 changes. do you think it's a bug?

meingraham, 2020-08-20:
Once the switch is off, there is no power consumed through the relay. It's been discussed previously. This `PowerDelta` "behavior" is as designed. I'm surprised that it is sent when either switch changes state. That may actually be the bug. Unsure.

But, if you are looking for consistent behavior for you... Rules.

GiorgioRoma, 2020-08-20:
what you write is clear, but in the command docs there is this: 

Set maximum delta for energy monitoring devices to report on active power load change while the power is ON. PowerDelta will not report when the power turns off. 
0 = disable reporting on power change
1..100 = set reporting **on percentage power change** to send an MQTT telemetry message
101..32000 = set reporting on absolute power change to send an MQTT telemetry message (offset by 100, e.g., 101=1**W**, 207=107**W**)

So it is correct that when the switches are turned off PowerDelta does not generate an update event, it only generates it if there is a change in power (watts)  when the switch is on, or when the switch state changes. But this only applies to switch 1, it doesn't work for switch 2, I hope I was clear.

sfromis, 2020-08-20:
The short version is that it would be reasonable to expect PowerDelta to trigger for energy usage changes on both channel 1 and channel 2. However, it only triggers on channel 1, not when turning on/off a consumer on channel 2.

Ignore all that stuff about switch state changes, that is not part of how Tasmota works for energy monitoring.

The easy way forward is, as already suggested, to create a small rule to trigger some more reporting of the energy consumption numbers.

ascillato2, 2020-08-20:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-08-20 by ascillato2

GiorgioRoma, 2020-08-20:
sorry for my post, where is the original issue post ?? Thanks

sfromis, 2020-08-20:
https://github.com/arendst/Tasmota/issues/7890

GiorgioRoma, 2020-08-21:
Thanks for the reply, however my report is different than the post reported, the bug I described concerns switch2, when its status changes (ON), PowerDelta does not send sensor data. This is the bug