arendst/Tasmota #9136: GPIO2 stop working as button or Switch after installing relese 8.4.0

byte4geek, 2020-08-20:
### PROBLEM DESCRIPTION
After upgrade frmo 8.2.0 to 8.4.0 the GPIO2 dose'nt works anymore as switch or button.
A down grade to 8.2.0 the GPIO2 works again.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff Basic
- [ ] Tasmota binary firmware version number used: 8.4.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: upgrade via OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
17:15:01 CMD: Backlog Template
17:15:01 MQT: stat/test/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
17:15:34 CMD: Module
17:15:34 MQT: stat/test/RESULT = {"Module":{"0":"Generic"}}
17:15:54 CMD: GPIO 255
17:15:54 MQT: stat/test/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"17":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
none

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
17:16:35 CMD: Status 0
17:16:35 MQT: stat/test/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"test","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:16:35 MQT: stat/test/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://192.168.1.2:9541/data/firmwares/tasmota-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:45","StartupUTC":"2020-08-20T15:13:50","Sleep":50,"CfgHolder":4617,"BootCount":40,"BCResetTime":"2020-04-07T14:07:15","SaveCount":243,"SaveAddress":"F8000"}}
17:16:35 MQT: stat/test/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-07-29T12:05:28","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"348/699"}}
17:16:35 MQT: stat/test/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HOME-NET",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000008","00006000"]}}
17:16:35 MQT: stat/test/STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
17:16:35 MQT: stat/test/STATUS5 = {"StatusNET":{"Hostname":"test-4921","IPAddress":"192.168.1.19","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"84:0D:8E:58:B3:39","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:16:35 MQT: stat/test/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_58B339","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:16:35 MQT: stat/test/STATUS7 = {"StatusTIM":{"UTC":"2020-08-20T15:16:35","Local":"2020-08-20T17:16:35","StartDST":"2020-03-29T01:00:00","EndDST":"2020-10-25T02:00:00","Timezone":99,"Sunrise":"06:50","Sunset":"20:55"}}
17:16:35 MQT: stat/test/STATUS10 = {"StatusSNS":{"Time":"2020-08-20T17:16:35"}}
17:16:35 MQT: stat/test/STATUS11 = {"StatusSTS":{"Time":"2020-08-20T17:16:35","Uptime":"0T00:02:45","UptimeSec":165,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXXXXX","BSSId":"DC:39:6F:AB:C2:02","Channel":11,"RSSI":40,"Signal":-80,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
using GPIO2 as button to change status to Relay1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-08-20:
You cannot have two Button1 configured. 8.4.0 is more strict in this.

byte4geek, 2020-08-20:
ok, configuring only one button fix this, but in 8.2.0 it works with this configuration, why now only one button must be present?

arendst, 2020-08-20:
Try a button2.

It makes no sense to name different physical buttons the same. Previous versions also did not allow the same device on different GPIOs.

As said, it is more strict now for obvious reasons.

byte4geek, 2020-08-20:
Ok I have configured GPOI0 as Button2 and GPIO2 as Button1 now it work fine, thanks for the support

closed 2020-08-20 by ascillato2