arendst/Tasmota #9153: Tasmota Wi-Fi AP password has to be under 32 characters and without special characters?

ghnu, 2020-08-23:
### PROBLEM DESCRIPTION
For security purposes, my AP password is of course longer than 32 characters also also does contain special characters.

### REQUESTED INFORMATION
Tasmota should support longer than 32 characters as WiFi password, and special characters should be allowed as well

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): BN-LINK BNC-60/U133TJ WIFI SMART PLUG
- [ ] Tasmota binary firmware version number used: 8.0.3.1
  - [ X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

Device never connects to my WiFi, but only continues to spin up its own TASMOTA_xxxx access point.

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Enter a WiFi password longer than 32 chars and with special chars in the AP1 or AP2 section

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
It should connect to my WiFi 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-08-23:
Update to latest Tasmota version. It supports more as 32 characters.
Special characters are still dangerous, no gurantees for

ghnu, 2020-08-23:
Starting with which Tasmota version the full 255 characters are supported?

Gosokan, 2020-08-24:
I'm using 8.3.1 and it's can connect to ssid with special character (eg: "My SSID 👍👍")

golddragon007, 2020-08-24:
@Gosokan the password need to contains special character, that's the OP's problem.

@ghnu Special character doesn't give sensible more entropy and in theory, a true pseudo-random password with [a-zA-Z0-9]{32} is enough for your lifetime if you don't share it on the social platform (and using a proper protocol).
[image]
The only problem is the compatibility whit this.

meingraham, 2020-08-24:
Password encoding is not handled by Tasmota. This is handed off to the underlying Espressif SDK (close sourced)... and thus cannot be altered. Tasmota supports longer passwords now. Special characters cannot be supported.

You may want to put all of your Tasmota devices on a separate SSID/subnet with a correspondingly lengthy password but refrain from using special characters. This is the best you can hope to achieve with the current state of ESP82xx based software. The other devices on your network can connect to an SSID with a password containing special characters.

closed 2020-08-27 by ascillato2