arendst/Tasmota #9154: Websend support HTTPS

ganzerlimino, 2020-08-24:
**Have you looked for this feature in other issues and in the docs?**  
Yes, with no luck
**Is your feature request related to a problem? Please describe.**  
I would like to send https message from Tasmota to Telegram chat.

**Describe the solution you'd like**  
Add a https support

**Describe alternatives you've considered**  
Accept the parameters :443 as port after the hostname. Example : www.pluto.com:443 will drive the web request to the correct https port 

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-08-24:
Tasmota is designed for MQTT first. TLS requires significant ram usage so we favor it for MQTT or very specific usage like Telegram.

ganzerlimino, 2020-08-24:
Sure, MQTT is the solution for home automation. 

But,as I saw sendmail supported (and the companion MQTT_TLS), i was hoping fot https suppor :-)
Having HTTP support (using WebSend) without the possibility to use the https sounded strange to me.

s-hadinger, 2020-08-24:
To reduce code size, TLS in Tasmota only supports 1 cipher and 2 certificate authorities. This works in very specific use cases but will fail if you try random https. General HTTPS support is way out of ESP8266 possibilities. Either use ESP32 or narrow to a specific use case where maybe it would be doable.

ganzerlimino, 2020-08-24:
Wow, if i can narrow the https usage to https://api.telegram.org/ domain, I suppose that will be enough to send message to telegram groups.

The complete link will be something like this :

`https://api.telegram.org/bot[BOT_API_KEY]/sendMessage?chat_id=[MY_CHANNEL_NAME]&text=[MY_MESSAGE_TEXT]`

My goal is to place some text in [MY_MESSAGE_TEXT] based upon Var and Mem, using some Rule.

arendst, 2020-08-24:
You don't need we send. You need telegram support already available in tasmota. Read the docs regarding telegram or look at the header of the telegram driver how to use.

ganzerlimino, 2020-08-24:
I'm embarassed and really sorry, I've not fount something similar, but using "tasmota telegram" instead of "tasmota ssl" I've fount exactly what I was searching for.

Tnx a lot Theo and s-hadinger for the time that both invested on me,
Mino

closed 2020-08-24 by ascillato2

chaveiro, 2021-09-16:
Addind that HTTPS support would be a nice addition to WEBSEND anyway.
I would use it to secure sending data to emoncms on top of using already mqtt with TLS for home assistant.

s-hadinger, 2021-09-16:
HTTPS is now supported with a high degree of customization using Berry Scripting and `webclient`
https://tasmota.github.io/docs/Berry/#webclient-class

It supports POST, custom headers, Authorization

DerDom, 2022-01-30:
Hey, I would love HTTPS support as well. I am trying to call a Google Form to store data in a Google Sheet with a URL like:  

https://docs.google.com/forms/d/e/1234567890/formResponse?entry.12345=temperatureData&submit=Submit

s-hadinger, 2022-01-30:
Esp32 only

wodnik7, 2022-02-20:
> > HTTPS is now supported with a high degree of customization using Berry Scripting...
> 
> This is ESP32 only. A 8266 option would be great. Here is the use case: My little solar panel is monitored by a smart power plug with tasmota. I'd like to send the output to [https://pvoutput.org](https://pvoutput.org/help/api_specification.html#get-examples). I wouldn't mind compiling a version and swapping the telegram API for the https target I need.

Hello, any news on making this possible? Thanks!

josemariaaraujo, 2022-07-03:
> Hello, any news on making this possible? Thanks!

pvoutput.org doesn't need HTTPS, works fine with HTTP

chaveiro, 2022-07-03:
Http is insecure.

Kugeleis, 2022-07-18:
I just found that pvoutput wants date and time as separate strings. like `d=20220718 t=10:32`. Can tasmota on ESP3266 transform date/time strings?

sfromis, 2022-07-18:
Not really, with the scripting language you may have some options, but be prepared for a steep learning curve if you go in that direction.

More generally, to send data to external servers, I'd not want to do it directly from each Tasmota device, but send the request to some backend server (could even be Tasmota on ESP32), able to deal with the requirements of the external service, including https.

barbudor, 2022-07-25:
For sure, feel free to add this to the docs by clicking on the pencil button at the top right.

Kugeleis, 2022-07-25:
done

barbudor, 2022-07-25:
Excellent
Thank you for contributing

TBR-BRD, 2022-08-24:
> Here is my working example how to upload the power reading from a ESP8266 based device to pvoutput.org. This works as long as they allow hppt (without s) connections.
> 
> ```
> >D 42
> ;long string required for key
> y=0
> m=0
> d=0
> key="yourkey"
> id="yourSystemID"
> ws="WebSend [pvoutput.org]"
> et=0
> p=0
> 
> >T
> et=ENERGY#Total
> p=ENERGY#Power
> ; every 5 minutes
> if upsecs%300==0
> then
> y=sb(tstamp 0 4)
> m=sb(tstamp 5 2)
> d=sb(tstamp 8 2)
> =>%ws%/service/r2/addstatus.jsp?key=%key%&sid=%id%&d=%1.0(y)%%2.0(m)%%2.0(d)%&t=%1(sb(tstamp 11 5))%&v2=%s(2.0p)%
> endif
> ```
> 
> This might be worth using as an example in the docs (@arendst ?).

My pvoutput.org upload is not working. Can u help me please?

>D 42
;long string required for key
y=0
m=0
d=0
key="9d31axxxxxxxxxxxxxxxxxbca7e"
id="7xxx"
ws="WebSend [pvoutput.org]"
et=0
p=0

>B
=>sensor53 r
;Set teleperiod to 20sec  
tper=10  
>M 1
+1,3,s,0,9600,Power
1,77070100010800ff@1000,PV_Bezug,kWh,Total_in,4
1,77070100100700ff@1,PV_Leistung,W,Power_curr,0
1,77070100020800ff@1000,PV_Erzeugung,kWh,Total_out,4

>T
et=ENERGY#Total_out
p=ENERGY#Power_curr
; every 5 minutes
if upsecs%300==0
then
y=sb(tstamp 0 4)
m=sb(tstamp 5 2)
d=sb(tstamp 8 2)
=>%ws%/service/r2/addstatus.jsp?key=%key%&sid=%id%&d=%1.0(y)%%2.0(m)%%2.0(d)%&t=%1(sb(tstamp 11 5))%&v2=%s(2.0et)%
endif
#

The response is:
WebSend [pvoutput.org]/service/r2/addstatus.jsp?key=9dxxxe&sid=7xxx&d=20220824&t=15:29&v2=00

v2 is sending with 0, that isn't right.

Thank u for help

Kugeleis, 2022-08-27:
First, the sections in the script start with `>`. This is missing in your example.
For debugging you cold just send a fixed string like `v2=123`. My example uses[ string conversion](https://tasmota.github.io/docs/Scripting-Language/#special-variables) and formatting` v2=%s(2.0et)%` - maybe this is not suitable for your numbers. Just try `v2=%yourvariable%`.

TBR-BRD, 2025-02-03:
> First, the sections in the script start with `>`. This is missing in your example. For debugging you cold just send a fixed string like `v2=123`. My example uses[ string [...]

Thank you, Kugeleis.
I have now tested, nothing works.
script?<PVErzeugunggesamt> result ? (see picture); MQTT pub works

[image]

TBR-BRD, 2025-02-03:
my new script:

`>D 42
;long string required for key
y=0
m=0
d=0
key="9d..."
id="7..."
ws="WebSend [pvoutput.org]"
et=0
p=0

>B
=>sensor53 r
;Set teleperiod to 20sec  
tper=20  
>M 1
+1,3,s,0,9600,PV
1,77070100010800ff@1000+195000,PVBezug,kWh,PVBezug,4
1,77070100100700ff@1,PVLeistung,W,PVLeistung,0
1,77070100020800ff@1000+51660000,PVErzeugunggesamt,kWh,PVErzeugunggesamt,4

>T
et=ENERGY#PVErzeugunggesamt
p=ENERGY#PVLeistung
; every 5 minutes oder 1min
if upsecs%60==0
then
y=sb(tstamp 0 4)
m=sb(tstamp 5 2)
d=sb(tstamp 8 2)
=>%ws%/service/r2/addstatus.jsp?key=%key%&sid=%id%&d=%1.0(y)%%2.0(m)%%2.0(d)%&t=%1(sb(tstamp 11 5))%&v2=%PVErzeugunggesamt%
endif
#`

TBR-BRD, 2025-02-04:
Please help me

TBR-BRD, 2025-02-26:
now i have a simple version, console output is different. CMD: script?<Total_out> => no value
RSL Sensor => Total_out: 10089 => right value

picture follows

TBR-BRD, 2025-02-26:
[image]

TBR-BRD, 2025-02-26:
any ideas?