arendst/Tasmota #9155: Alexa device discovery not working with Hue emulation.

kgwatson, 2020-08-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Alexa not discovering tasmota devices with Hue emulation

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff S2X
- [x] Tasmota binary firmware version number used: 8.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
09:59:22 CMD: Backlog Template; Module; GPIO 255
09:59:22 SRC: WebConsole from 192.168.1.172
09:59:22 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
09:59:22 SRC: Backlog
09:59:22 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
09:59:22 MQT: stat/tasmota_1C076F/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
09:59:22 SRC: Backlog
09:59:22 CMD: Group 0, Index 1, Command "MODULE", Data ""
09:59:22 MQT: stat/tasmota_1C076F/RESULT = {"Module":{"8":"Sonoff S2X"}}
09:59:23 SRC: Backlog
09:59:23 CMD: Group 0, Index 1, Command "GPIO", Data "255"
09:59:23 MQT: stat/tasmota_1C076F/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
0
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
10:02:25 CMD: status 0
10:02:25 SRC: WebConsole from 192.168.1.172
10:02:25 CMD: Group 0, Index 1, Command "STATUS", Data "0"
10:02:25 MQT: stat/tasmota_1C076F/STATUS = {"Status":{"Module":8,"DeviceName":"Printer","FriendlyName":["Printer"],"Topic":"tasmota_1C076F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:02:25 MQT: stat/tasmota_1C076F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:03","StartupUTC":"2020-08-24T08:55:22","Sleep":50,"CfgHolder":4617,"BootCount":42,"BCResetTime":"2020-06-29T18:49:18","SaveCount":254,"SaveAddress":"F9000"}}
10:02:26 MQT: stat/tasmota_1C076F/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-07-29T12:05:28","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"399/699"}}
10:02:26 MQT: stat/tasmota_1C076F/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Sweet16",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00800000","00006000"]}}
10:02:26 MQT: stat/tasmota_1C076F/STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
10:02:26 MQT: stat/tasmota_1C076F/STATUS5 = {"StatusNET":{"Hostname":"tasmota_1C076F-1903","IPAddress":"192.168.1.215","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"CC:50:E3:1C:07:6F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:02:26 MQT: stat/tasmota_1C076F/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.9","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1C076F","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:02:26 MQT: stat/tasmota_1C076F/STATUS7 = {"StatusTIM":{"UTC":"2020-08-24T09:02:26","Local":"2020-08-24T10:02:26","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:56","Sunset":"19:47"}}
10:02:26 MQT: stat/tasmota_1C076F/STATUS10 = {"StatusSNS":{"Time":"2020-08-24T10:02:26"}}
10:02:26 MQT: stat/tasmota_1C076F/STATUS11 = {"StatusSTS":{"Time":"2020-08-24T10:02:26","Uptime":"0T00:07:04","UptimeSec":424,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":35,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Sweet16","BSSId":"64:CC:22:F4:8D:8D","Channel":1,"RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

Alexa device discovery invoked from Amazon Echo Dot:
09:12:14 UPP: Hue 3 response packets sent to 192.168.1.108:50000
09:12:14 HTP: Hue setup
09:12:14 HTP: Hue API (/1c076f/lights)
09:12:14 HTP: Hue User-Agent: , gen=1
09:12:14 HTP: Hue Result ({"29390577":{"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Printer","modelid":"Generic","manufacturername":"Tasmota","uniqueid":"cc:50:e3:1c:07:6f:00:11-1"}})
09:12:15 HTP: Hue API (/1c076f/lights/29390577)
09:12:15 /lights path=/lights/29390577
09:12:15 HTP: Hue User-Agent: , gen=1
09:12:15 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Printer","modelid":"Generic","manufacturername":"Tasmota","uniqueid":"cc:50:e3:1c:07:6f:00:11-1"})

Alexa device discovery invoked from Android app:
09:15:16 UPP: Hue 3 response packets sent to 192.168.1.212:50000
09:15:16 HTP: Hue setup
09:15:17 UDP: Packet (119)
09:15:17 HTP: Hue API (/1c076f/lights)
09:15:17 HTP: Hue User-Agent: , gen=1
09:15:17 HTP: Hue Result ({"29390577":{"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Printer","modelid":"Generic","manufacturername":"Tasmota","uniqueid":"cc:50:e3:1c:07:6f:00:11-1"}})
09:15:17 HTP: Hue API (/1c076f/lights/29390577)
09:15:17 /lights path=/lights/29390577
09:15:17 HTP: Hue User-Agent: , gen=1
09:15:17 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Printer","modelid":"Generic","manufacturername":"Tasmota","uniqueid":"cc:50:e3:1c:07:6f:00:11-1"})
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Alexa should discover Tasmota devices with Hue emulation enabled. It does not.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
According to weblogs, Tasmota device is responding to Alexa discovery requests, but subsequently device does not appear in Alexa devices list.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-08-24:
You have to discover for Philips HueBridge V.1. Just tried works without a issue.

kgwatson, 2020-08-24:
That's what I've done (many times). This used to work with one device, but since I added some more, Alexa no longer discovers ANY of my devices. I've disabled Hue emulation on all my devices except one, but Alexa still doesn't find it even though the logs indicate (see above) that the device is responding to Alexa discovery requests.

If it makes any difference, I have Node Red, Mosquitto MQTT and TasmoAdmin all running in Docker containers on a Raspberry Pi 3B+. Everything works fine except for the Alexa device discovery, which used to work, which makes it very frustrating.

kgwatson, 2020-08-24:
Looking at the Hue response packets in the log above, are these in the format that Alexa expects? Could it be something to do with port 50000?

kgwatson, 2020-08-24:
Ok. It's working now. Further investigation shows it's an Amazon issue. I went to the Amazon website and deleted my Smart Home Device history, then invoked Alexa device discovery and it found all my devices.

ascillato2, 2020-08-24:
> It's working now. Further investigation shows it's an Amazon issue.

Great to know that now it is working for you. Closing this issue. Thanks.

closed 2020-08-24 by ascillato2

slim0287, 2020-08-29:
I have the exact same problem that kgwatson described and unfortunately, deleting Smart Home Device history from the Amazon website did not resolve the issue.  Should I open a new issue or post the requested information here?

s-hadinger, 2020-08-29:
@slim0287 Please provide `Weblog 4` logs of the pairing.

slim0287, 2020-08-29:
@s-hadinger 
Here are the Console Weblog 4 responses to discovery from an Echo (the 192.168.152.76 IP):

```
15:05:13 UPP: Hue 3 response packets sent to 192.168.152.76:50000
15:05:13 HTP: Hue setup
15:05:13 HTP: Hue API (/d5c72e/lights)
15:05:13 HTP: Hue User-Agent: , gen=1
15:05:13 HTP: Hue User-Agent: , gen=1
15:05:13 HTP: Hue Result ({"224162529":{"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3a","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-1"},"224162530":{"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3b","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-2"}})
15:05:13 UDP: Packet (94)
15:05:13 UDP: Packet (101)
15:05:13 HTP: Hue API (/d5c72e/lights/224162529)
15:05:13 /lights path=/lights/224162529
15:05:13 HTP: Hue User-Agent: , gen=1
15:05:13 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3a","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-1"})
15:05:14 HTP: Hue API (/d5c72e/lights/224162530)
15:05:14 /lights path=/lights/224162530
15:05:14 HTP: Hue User-Agent: , gen=1
15:05:14 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3b","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-2"})
15:05:15 UPP: Hue 3 response packets sent to 192.168.152.99:35334
15:05:16 HTP: Hue setup
```

And here are the logs from discovery done through the Android app (odd that there is no IP for an Echo here.  The 192.168.152.99 is a Home Assistant installation):

```
15:07:18 UPP: Hue 3 response packets sent to 192.168.152.99:45165
15:07:21 UDP: Packet (119)
15:07:21 UDP: Packet (119)
15:07:21 UDP: Packet (119)
15:07:21 UDP: Packet (107)
15:07:21 UDP: Packet (119)
15:07:21 UDP: Packet (119)
15:07:23 HTP: Hue setup
15:07:23 HTP: Hue API (/d5c72e/lights)
15:07:23 HTP: Hue User-Agent: , gen=1
15:07:23 HTP: Hue User-Agent: , gen=1
15:07:23 HTP: Hue Result ({"224162529":{"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3a","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-1"},"224162530":{"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3b","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-2"}})
15:07:23 HTP: Hue API (/d5c72e/lights/224162529)
15:07:23 /lights path=/lights/224162529
15:07:23 HTP: Hue User-Agent: , gen=1
15:07:23 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3a","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-1"})
15:07:23 HTP: Hue API (/d5c72e/lights/224162530)
15:07:23 /lights path=/lights/224162530
15:07:23 HTP: Hue User-Agent: , gen=1
15:07:23 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3b","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-2"})
15:07:26 UDP: Packet (119)
```

Any ideas?  Thanks for the help.  I'm happy to provide the other information of an issue report if that will help.

s-hadinger, 2020-08-29:
@slim0287 From the first log, Alexa did successfully discover 2 devices with internal Ids: 224162529 and 224162530.

However short after a request is made by HA and I think it interferes with the discovery. Pleas disable HA Hue client and try again.

slim0287, 2020-08-29:
@s-hadinger 
Its funny you should ask me to do it with the HA client disabled.  I actually first did it with the HA powered off and got the same result.

Here's the log from discovery from the Echo:
```
14:52:05 UPP: Hue 3 response packets sent to 192.168.152.76:50000
14:52:06 HTP: Hue setup
14:52:06 HTP: Hue API (/d5c72e/lights)
14:52:06 HTP: Hue User-Agent: , gen=1
14:52:06 HTP: Hue User-Agent: , gen=1
14:52:06 HTP: Hue Result ({"224162529":{"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3a","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-1"},"224162530":{"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3b","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-2"}})
14:52:06 HTP: Hue API (/d5c72e/lights/224162529)
14:52:06 /lights path=/lights/224162529
14:52:06 HTP: Hue User-Agent: , gen=1
14:52:06 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3a","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-1"})
14:52:06 HTP: Hue API (/d5c72e/lights/224162530)
14:52:06 /lights path=/lights/224162530
14:52:06 HTP: Hue User-Agent: , gen=1
14:52:06 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3b","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-2"})
14:52:07 WIF: Checking connection...
14:52:11 UPP: Multicast disabled
14:52:11 MQT: Attempting connection...
14:52:16 MQT: Connect failed to 192.168.152.99:1883, rc -2. Retry in 10 sec
14:52:17 UPP: Multicast (re)joined
```

And here's the log with discovery from the Android app:
```
14:54:29 UPP: Hue 3 response packets sent to 192.168.152.76:50000
14:54:29 HTP: Hue setup
14:54:29 HTP: Hue API (/d5c72e/lights)
14:54:29 HTP: Hue User-Agent: , gen=1
14:54:29 HTP: Hue User-Agent: , gen=1
14:54:29 HTP: Hue Result ({"224162529":{"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3a","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-1"},"224162530":{"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3b","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-2"}})
14:54:29 HTP: Hue API (/d5c72e/lights/224162529)
14:54:29 /lights path=/lights/224162529
14:54:29 HTP: Hue User-Agent: , gen=1
14:54:29 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3a","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-1"})
14:54:29 HTP: Hue API (/d5c72e/lights/224162530)
14:54:29 /lights path=/lights/224162530
14:54:29 HTP: Hue User-Agent: , gen=1
14:54:29 HTP: Hue Result ({"state":{"on":false,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"OUT_Outlet3b","modelid":"Teckin SS31","manufacturername":"Tasmota","uniqueid":"2c:f4:32:d5:c7:2e:00:11-2"})
14:54:32 WIF: Checking connection...
14:54:34 UPP: Multicast disabled
14:54:34 MQT: Attempting connection...
14:54:39 MQT: Connect failed to 192.168.152.99:1883, rc -2. Retry in 10 sec
14:54:40 UPP: Multicast (re)joined
14:54:50 UPP: Multicast disabled
```

It looks similar, although now instead of a Hue response to HA we see an attempt for an MQTT connection which fails because HA is powered down.

This is frustrating, up until just a few days ago, it was working well.

s-hadinger, 2020-08-29:
Discovery looks good. What Echo device are you using? It looks that Echo generation detection does not work anymore, and that could explain the issue.

Can you please update to the latest version in the dev branch from http://ota.tasmota.com/tasmota/

Then remove the devices from Alexa and do a fresh new discovery. I believe it will work better now.

slim0287, 2020-08-29:
I have two Echo devices.  The first is the Echo 1st generation, with software version 653619520.  The second is the Echo Dot 2nd generation software version 655711020.  And I should give you one more piece of information, I do have a few Tasmota-flashed smartplugs with Belkin WeMo emulation and these are discovered by Alexa without issue.

I have a firmware update question.  I am using Tasmota 8.4, which I got by using the "Upgrade by web server" button several weeks ago.  Currently `http://thehackbox.org/tasmota/release/tasmota.bin` is in the field.  Are you asking me to use the `tasmota.bin` file of the dev branch or a different file?  Second, as I understand this will take me from core 2.7.2.1 to core 2.7.4.1, and this may work better?  Sorry for the questions but I always have used stable releases. 

Lastly, the devices are not currently in Alexa because they haven't been discovered, so I don't need to remove them.

s-hadinger, 2020-08-30:
Ah some good news. The Echo 1st generation does not do Hue discovery at all, and the Echo Dot 2nd gen needs a tweak on the latest version. You need to use `SetOption109 1` if you have only older generation Echo device, otherwise On/Off device will not get recognized.

slim0287, 2020-08-30:
That's interesting, because Hue Emulation has worked properly for about a year and I never made that tweak; you write "on the latest version" so perhaps you mean Tasomta.  I also do have a Fire Tablet HD 10, 7th generation on the network so I'm not sure if that would make a difference.

I issued `SetOption109 1` on the Teckin SS31 that is not being discovered and ran discovery from the Echo Dot, the Amazon App from a Samsung Tab S6, and from `https://alexa.amazon.com/` and none of the three discovered the device.  I am using the dev built on 20200829, would that make a difference?

To make matters a little more complicated, I just added a Teckin SP10 smart outlet inside and using Belkin WeMo emulation and Alexa won't recognize it either.  I have several other SP10s that have been discovered just fine.  This one is still running Tasmota 8.4.0 so perhaps I should move it to the dev branch as well, even though the ones that have been discovered are on Tasmota 8.4.0?

I am in a weird state here.  I have 2 Teckin SS31s running Tasmota 8.4.0.3 which are discovered by Alexa and 1 Techkin SS31 running Tasmota 8.4.0.3 which is not, and I have 6 Teckin SP10s running Tasmota 8.4.0 which are discovered by Alexa and 1 Teckin SP10 running Tasmota 8.4.0 which is not.  In the past, all of these devices were discovered.

I feel like I am running out of options here.  I flashed all of these devices with Tasmota because they were originally Smart Life and I wanted out of the cloud and more control.  However at this point I have odd semi-broken functionality.  Fortunately my wife doesn't like to talk to Alexa and controls these things with her phone otherwise I'd be in big trouble :) However I'd very much like to have it function properly with Alexa as well because I use it.

Jason2866, 2020-08-30:
Amazon has changed stuff how it works. Discovery now will work only with a Echo Dot 2nd Gen. if latest Tasmota development version is flashed and Setoption 109 is activated on all your devices you want to use with Alexa. (EchoDot 1st Gen. is not capable to discover).
To have a clean setup. Delete all devices in Alexa. Delete Smart Home device history.
Restart Echo Dot 2nd Gen. Start device discovery in Alexa App

slim0287, 2020-08-30:
@Jason2866 
Thanks.  OK, just so I understand the steps:
  1.  Flash the 20200830 development version on all of the Teckin SP10s and Teckin SS31s
  2. Run `SetOption109 1` on all flashed devices
  2. Delete all devices in Alexa
  3. Delete Smart Home Device History
  4. Restart the echo Dot 2nd Generation (i.e., power off, then on)
  5. Start the device discovery on the Alexa App (running on my Samsung Tab S6)

It is interesting that discovery must be done in the app rather than the Echo Dot.

I will give it a go.  Please tell me if I have that incorrect.

s-hadinger, 2020-08-30:
`SetOption109 1` is not needed for Wemo, only for Hue emulation and only for simple switches (RGB or dimmer lights are fine).

Also the discovery issue may be caused by a corrupt Alexa skill. Make sure also that you disable any SmartHome Alexa skill, do the discovery and re-enable any SmartHome skill. You don't need to disable any other skill like radio, news, fun...

slim0287, 2020-08-30:
Heh.  Should I through some Eye-of-Newt into the cauldron and do a prayer dance at midnight during full moon for the gods of IoT as well?  Disabling any SmartHome Alexa skill means Philips Hue and Sonos skills, etc.  Got it.  In process, we'll see how it goes.

Jason2866, 2020-08-30:
@slim0287  Played around. Had the issue too that no new devices are discovered.
Discovery works reliable if and only then:
**Shut down every device where you have a Alexa App running!**

slim0287, 2020-08-30:
@Jason2866 
I am right near the end of this, so I should be clear on the process.  I am powering off and on the Echo Dot now.  Should I shut down all devices running Alexa App and do the discovery from the Dot?  Previously, you had written I should do the discovery from the Alexa App.

Jason2866, 2020-08-30:
Yeah, before it worked (better) with the App...  NOW i had to shutdown all Alexa Apps on my devices
#blameAmazon

slim0287, 2020-08-30:
This hasn't worked out well for me.  I went through the steps listed above, and out of the 10 Tasmota-flashed devices I have, only 4 are discovered.  I actually made steps backward, when I started this morning 8 had been discovered and I was only having trouble with 2.  Now I am missing 6.

Jason2866, 2020-08-30:
😟 Out of ideas. Just tried again. Deleted 10 devices. All where discovered in first try.
Have you tried Wemo Emulation for the switches too?

slim0287, 2020-08-30:
I am doing a little better now.  All WeMo emulation devices have been discovered; I had four originally and as I re-enabled the Hue skill and the Sonos skill, the other three WeMo emulation devices came back up.  Now I am missing just the three Hue emulation devices.  These three have the latest firmware and I ran `SetOption109 1` on them.

Jason2866, 2020-08-30:
I think Amazon is changing stuff in your region. The behaviour is strange and the software in the devices is the same... Maybe you just wait a few days and try again.

slim0287, 2020-08-30:
Believe it or not I actually had lunch with Jeff Bezos 10 years ago.  Little did I know at that time I'd eventually be frustrated by an API into one of his devices!  

I'll give it a go in a few days and see if I can pick out the Hue emulation devices.  I'm thinking about opening a new issue as well given this is still a problem.

Jason2866, 2020-08-30:
Cool, for sure intersting to talk about his ideas and visions.
General it is easier and more responsive helping in Discord

slim0287, 2020-08-30:
Yea, I tried this topic in Discord and didn't get anywhere.  Between you (@Jason2866) and @s-hadinger this GitHub thread has been much more helpful.

martin1338, 2021-01-17:
Hi all, sorry for adding comments here - but I have the same issue and am kind of desperate because Alexa won't find my tasmota devices. 
What I did so far: 

- deleted all devices via alexa.amazon.de
- searched for devices -> found 1 Tasmota device (yeeha) but not the other 3
- tried Wemo for the other 3 tasmota devices -> works fine
- updated one of the tasmota devices to the newest version -> not succesful
- unplugged all alexa devices except for 1 Echo Dot 2nd Gen
- tried searching for devices and got the following results: 

Device 1 with Tasmota 9.2.0:
`14:29:25 UPP: Hue 3 response packets sent to 192.168.0.209:50000
14:29:26 UDP: Packet (119)
14:29:26 UDP: Packet (119)
14:29:26 UDP: Packet (119)
14:29:26 UDP: Packet (119)
14:29:26 UDP: Packet (119)
14:29:26 UDP: Packet (119)
14:29:26 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:27 UDP: Packet (119)
14:29:28 UDP: Packet (119)
14:29:28 UDP: Packet (119)
14:29:28 UDP: Packet (119)
14:29:28 UDP: Packet (119)
14:29:28 UDP: Packet (119)
14:29:28 UDP: Packet (119)
14:29:28 UDP: Packet (119)
14:29:28 UDP: Packet (119)
14:29:28 UDP: Packet (119)
14:29:30 WIF: Checking connection...
14:29:31 UDP: Packet (119)
14:29:32 UDP: Packet (119)
14:29:32 UDP: Packet (119)
14:29:32 UDP: Packet (119)
14:29:33 HTP: Hue setup
14:29:34 HTP: Hue API (/79466f/lights) from 192.168.0.209
14:29:34 HTP: Hue Result ({"127166193":{"state":{"on":false,"bri":153,"colormode":"hs","xy":[0.52570,0.45438],"hue":8010,"sat":254,"ct":153,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Ikea Lampe 1","modelid":"Maxcio YXL01P","manufacturername":"Tasmota","uniqueid":"80:7d:3a:79:46:6f:00:11-1"}})
14:29:34 HTP: Hue API (/79466f/lights/127166193) from 192.168.0.209
14:29:34 /lights path=/lights/127166193
14:29:34 HTP: Hue Result ({"state":{"on":false,"bri":153,"colormode":"hs","xy":[0.52570,0.45438],"hue":8010,"sat":254,"ct":153,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Ikea Lampe 1","modelid":"Maxcio YXL01P","manufacturername":"Tasmota","uniqueid":"80:7d:3a:79:46:6f:00:11-1"})
14:29:34 HTP: Hue API (/79466f/lights) from 192.168.0.209
14:29:34 HTP: Hue Result ({"127166193":{"state":{"on":false,"bri":153,"colormode":"hs","xy":[0.52570,0.45438],"hue":8010,"sat":254,"ct":153,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Ikea Lampe 1","modelid":"Maxcio YXL01P","manufacturername":"Tasmota","uniqueid":"80:7d:3a:79:46:6f:00:11-1"}})
14:29:36 UDP: Packet (119)`

Device 2 with Tasmota 8.2.0
`14:29:24 UPP: Hue 3 response packets sent to 192.168.0.209:50000
14:29:25 UDP: Packet (199)
14:29:25 UDP: Packet (199)
14:29:25 UDP: Packet (199)
14:29:26 WIF: Checking connection...
14:29:26 UDP: Packet (199)
14:29:26 UDP: Packet (199)
14:29:26 UDP: Packet (199)
14:29:26 UDP: Packet (199)
14:29:26 UDP: Packet (127)
14:29:26 UDP: Packet (127)
14:29:26 UDP: Packet (129)
14:29:26 UDP: Packet (199)
14:29:26 UDP: Packet (199)
14:29:26 UDP: Packet (199)
14:29:26 UDP: Packet (199)
14:29:26 UDP: Packet (199)
14:29:26 UDP: Packet (199)
14:29:26 UDP: Packet (199)
14:29:26 UDP: Packet (199)
14:29:26 UDP: Packet (199)
14:29:30 UDP: Packet (123)
14:29:31 UDP: Packet (133)
14:29:31 UDP: Packet (127)
14:29:31 UDP: Packet (129)
14:29:31 UDP: Packet (129)
14:29:31 UDP: Packet (124)
14:29:32 HTP: Hue setup
14:29:32 HTP: Hue API (/618e30/lights)
14:29:32 HTP: Hue User-Agent: , gen=1
14:29:32 HTP: Hue Result ({"102294273":{"state":{"on":true,"bri":179,"colormode":"hs","xy":[0.52570,0.45438],"hue":8032,"sat":254,"ct":153,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Ikea Lampe 2","modelid":"LCT007","uniqueid":"2c:f4:32:61:8e:30:00:11-1","swversion":"5.50.1.19085"}})
14:29:32 HTP: Hue API (/618e30/lights/102294273)
14:29:32 /lights path=/lights/102294273
14:29:32 HTP: Hue User-Agent: , gen=1
14:29:32 HTP: Hue Result ({"state":{"on":true,"bri":179,"colormode":"hs","xy":[0.52570,0.45438],"hue":8032,"sat":254,"ct":153,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Ikea Lampe 2","modelid":"LCT007","uniqueid":"2c:f4:32:61:8e:30:00:11-1","swversion":"5.50.1.19085"})
14:29:32 HTP: Hue API (/618e30/lights)
14:29:32 HTP: Hue User-Agent: , gen=1
14:29:32 HTP: Hue Result ({"102294273":{"state":{"on":true,"bri":179,"colormode":"hs","xy":[0.52570,0.45438],"hue":8032,"sat":254,"ct":153,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Ikea Lampe 2","modelid":"LCT007","uniqueid":"2c:f4:32:61:8e:30:00:11-1","swversion":"5.50.1.19085"}})
14:29:35 UDP: Packet (123)`

The search was executed from alexa.amazon.de, but I also tried Android app, Alexa command, iOS App -> all without success.

Any advice would be appreciated :)

slim0287, 2021-01-17:
I ended up buying an Echo Dot, 3rd generation which made things go smoother.  I can't say for sure a 2md generation won't work but I recall struggling with it so I just got a new Dot.

Also make sure you don't have the Alexa app running on a tablet or smartphone somewhere.  Either quit the app or shut the tablet/smartphone off.

I disabled Alexa Smart Home skills (in my case, Hue and Sonos) prior to doing the discovery.  After the discovery, I re-enable the skills.

Good luck.  It was real irritating for me when I went through the process but has been working smoothly since.  Oh, I do the discovery either from the Dot itself or from alexa.amazon.com.

BoShek76, 2021-03-30:
Dont know if it is helpful. I am using two switches with hue bridge emulation for shutter control they ran tasmota 6.7.1 and an echo dot 3. I bought an Echo Studio but with it th voice control wasn't working anymore. I upgraded the switches to 9.1.0 but still not working. After reading the thread I turned on the dot3 and everything was working fine again. I found out that I only need to have the dot3 powered up (running in an other room on a other floor) than voice control on my studio is working. When turn off my dot3 it isn't working.

bwims, 2021-04-01:
> You have to discover for Philips HueBridge V.1. Just tried works without a issue.

How do you do this?  

Thanks.

bwims, 2021-04-01:
Never mind, I figured it out.  My zbbridge is invisible.

RolfWojtech, 2021-04-05:
After spending an hour trying all permutations at a friend's house, I must conclude that this is not fixed.

Hardware:
- Gosund SP211 Module with Tasmota 9.3.1 (the one I am trying to add)
- many existing tasmota single plugs with Tasmota 9.3.1
- Echo Show 2nd Gen
- Echo 1st Gen

I tried with:
- SetOption109 on or off
- All devices connected
- Only an Echo Show 2nd Gen with Software 663648120 connected, searching from App, Website and Device
- Only an Echo 1st Gen with Software 658655620 connected, searching from App, Website and Device

In the end I gave up and decided to at least get one of the power outlets working by using Wemo mode, only to find out that the documentation on https://tasmota.github.io/docs/Alexa/ is wrong and using Belkin WeMo actually exposes both relays to Alexa. The documentation says: "Use Belkin WeMo for devices with a single relay or Hue Bridge for devices with one or more relays or for lights." which is not the best advice given how many issues people have with Hue Bridge mode.

So I recommend to either fix the Hue Bridge issues with Alexa or change that part of the documentation to recommend Belkin WeMo for Alexa users.

slim0287, 2021-04-05:
@RolfWojtech  
In order to make Hue Bridge emulation work, I needed to get an Echo Dot 3rd generation.  I suspect you are running into issues because you are using the Echo Show 2nd generation and Echo 1st generation.

It was painful to figure it out as I tried many permutations prior to purchasing the Echo Dot 3rd generation, but after I did so it went smoothly.

RolfWojtech, 2021-04-05:
Thanks for the recommendation slim0287. Maybe this Merge has something to do with it: https://github.com/arendst/Tasmota/pull/11415

Now that everything is working i'm a bit sceptical to try it out, Alexa is being so sensitive to device changes. But I think Tasmota is awesome. Since it is not the first time that Amazon introduced such changes, I recommended to change the documentation to recommend WeMo even for multi-relay devices (#11594)

slim0287, 2021-04-05:
It has been quite some time since I had these issues, but I don't think I ever tried WeMo for multi-relay devices.  I should have thought of that because it may have helped.  I'm glad to learn the tip from you.

I don't think my problem was related to Merge #11415 because I had this issue back in January and the merge is recent.

You are right Tasmota is awesome.  I have 9.3.1 running on dozens of devices around my house.  In fact I am about to move and I'm dreaming up the way I'm going to create my next "smart home."

Jason2866, 2021-04-05:
Switching to Belkin is not option since it is only for relais. HueEmulation does support lights with color control.

Mighty-Mick, 2021-04-13:
I was watching this issue closely as I exprienced the same problem with two newly flashed devices - both on Tasmota 9.3.1. Alexa simply could not discover these devices running with HUE Emulation but more than 30 devices are running fine since some years in my smart home. I also scanned the offiical Alexa forum and followed the recommendation to remove all devices. Afterwards none of my old devices was discovered again. However Amazon changed something (see here [Forum]( https://www.amazonforum.com/s/feed/0D54P000084CnuYSAS?t=1616965077174) and afterwards old devices previously deleted were discovered again. Unfortunately new devices are still not recognized. Any idea would be highly appreciated.

When using "Wemo Emulation" all deives are discovered without any problem but this does not help with a RGB light.

Jason2866, 2021-04-14:
For latest development! there is feedback that it works again.

Mighty-Mick, 2021-04-14:
> For latest development! there is feedback that it works again.

This indeed would be good news. I searched discussions/issues but with no concrete result to get more concrete feedback. So I will wait. Thank you!

s-hadinger, 2021-04-14:
This issue is closed and obsolete. The problem is solved in latest version

ascillato, 2021-04-14:
Please, update to latest tasmota. You can use OTAurl to http://ota.tasmota.com/tasmota/tasmota.bin.gz

You can update from the Tasmota Web Interface in FIRMWARE UPDATE

or go to the console and type

```
OTAurl  http://ota.tasmota.com/tasmota/tasmota.bin.gz
upgrade 1
```

Mighty-Mick, 2021-04-14:
Confirmed: it is working now! Alexa discovers new HUE devices. Thanks a lot for this great support! Awesome!

robluca, 2021-04-19:
Hello Guys,
I have same problem today. Here Console log:
09:26:57.681 CMD: Backlog Template; Module; GPIO 255
09:26:57.684 SRC: WebConsole from 192.168.188.75
09:26:57.688 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
09:26:57.738 SRC: Backlog
09:26:57.741 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
09:26:57.747 RSL: RESULT = {"NAME":"LedMonocolore","GPIO":[0,0,0,418,0,32,0,0,416,417,0,0,0,0],"FLAG":0,"BASE":73}
09:26:57.964 UDP: Packet (119)
09:26:57.968 UDP: HUE
09:26:57.970 UDP: Packet (119)
09:26:57.974 SRC: Backlog
09:26:57.977 CMD: Group 0, Index 1, Command "MODULE", Data ""
09:26:57.982 RSL: RESULT = {"Module":{"0":"LedMonocolore"}}
09:26:58.217 SRC: Backlog
09:26:58.220 CMD: Group 0, Index 1, Command "GPIO", Data "255"
09:26:58.228 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"418":"PWM3"},"GPIO4":{"0":"None"},"GPIO5":{"32":"Button1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"416":"PWM1"},"GPIO13":{"417":"PWM2"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

09:29:26.851 CMD: Backlog Rule1; Rule2; Rule3
09:29:26.855 SRC: WebConsole from 192.168.188.75
09:29:26.858 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
09:29:26.940 UDP: Packet (119)
09:29:26.943 UDP: HUE
09:29:26.948 SRC: Backlog
09:29:26.951 CMD: Group 0, Index 1, Command "RULE", Data ""
09:29:26.957 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":64,"Free":447,"Rules":"ON Switch1#State DO Backlog Mem1=%value%; Ruletimer1 0.001 ENDON"}
09:29:27.189 SRC: Backlog
09:29:27.192 CMD: Group 0, Index 2, Command "RULE", Data ""
09:29:27.199 RSL: RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":329,"Free":182,"Rules":"ON Rules#Timer=1 DO IF (Mem1==2) Power1 2 ENDIF ENDON ON Rules#Timer=1 DO IF ((Mem1==6) AND (Mem2==0)) Mem2=1 ELSEIF ((Mem1==6) AND (Mem2==1)) Mem2=0 ENDIF ENDON ON Rules#Timer=1 DO IF (((Mem1==4) OR (Mem1==7)) AND (Mem2==0)) Dimmer+ ENDIF ENDON ON Rules#Timer=1 DO IF (((Mem1==4) OR (Mem1==7)) AND (Mem2==1)) Dimmer- ENDIF ENDON"}
09:29:27.447 SRC: Backlog
09:29:27.451 CMD: Group 0, Index 3, Command "RULE", Data ""
09:29:27.456 RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

09:30:56.008 CMD: status 0
09:30:56.011 SRC: WebConsole from 192.168.188.75
09:30:56.014 CMD: Group 0, Index 1, Command "STATUS", Data "0"
09:30:56.020 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"LuciStanzetta","FriendlyName":["Luce Stanzetta"],"Topic":"tasmota_9E9DDB","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[11,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:30:56.028 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T10:07:17","StartupUTC":"2021-04-18T21:23:39","Sleep":50,"CfgHolder":4617,"BootCount":32,"BCResetTime":"2021-04-18T14:59:06","SaveCount":272,"SaveAddress":"FB000"}}
09:30:56.034 RSL: STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-03-09T16:12:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"426/699"}}
09:30:56.037 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":1,"LogHost":"","LogPort":514,"SSId":["Vodafone_40308150_Fritz",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008001","0F05C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
09:30:56.046 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
09:30:56.054 RSL: STATUS5 = {"StatusNET":{"Hostname":"LuceStanzetta","IPAddress":"192.168.188.180","Gateway":"192.168.188.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.188.1","Mac":"48:3F:DA:9E:9D:DB","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:30:56.063 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-04-19T07:30:56","Local":"2021-04-19T09:30:56","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"06:50","Sunset":"20:47"}}
09:30:56.069 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-04-19T09:30:56"}}
09:30:56.076 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-04-19T09:30:56","Uptime":"0T10:07:17","UptimeSec":36437,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":70,"Color":"B30077","HSBColor":"320,100,70","Channel":[70,0,47],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Vodafone_40308150_Fritz","BSSId":"E8:DF:70:95:96:F2","Channel":11,"RSSI":100,"Signal":-40,"LinkCount":2,"Downtime":"0T00:01:37"}}}

Finally Console log of Alexa device discovery invoked from Android app:
09:38:05.850 UPP: Hue 3 response packets sent to 192.168.188.64:50000
09:38:05.853 UDP: Packet (119)
09:38:05.856 UDP: Packet (119)
09:38:05.889 UDP: Packet (119)
09:38:05.891 UDP: Packet (119)
09:38:05.894 UDP: Packet (119)
09:38:06.242 HTP: Hue setup
[... cut ...]
09:38:16.798 HTP: Hue Authentication Result ([{"success":{"username":"9e9ddb"}}])
09:38:16.889 HTP: Hue API (/9e9ddb/lights) from 192.168.188.64
09:38:16.901 HTP: Hue Result ({"166321585":{"state":{"on":false,"bri":179,"colormode":"hs","xy":[0.52308,0.17444],"hue":58253,"sat":254,"aler...
09:38:16.987 UDP: Packet (119)

I have others devices that were discovered and are in use from few months ago and works but I tried to remove one of them from Alexa and I was not able to discover it again :-(

Could you help me if I missing something?

Thank you!

Roberto

robluca, 2021-04-19:
Update: With latest development release 

Program Version | 9.3.1.3(tasmota)
-- | --
2021-04-18T14:26:49
2_7_4_9/2.2.2-dev(38a443e)

It works again!
Thank you!
Roberto

danilodene, 2021-04-20:
> Update: With latest development Version
> 
> Program Version	9.3.1.3(tasmota)
> 2021-04-18T14:26:49	
> 2_7_4_9/2.2.2-dev(38a443e)	
> It works again!
> Thank you!
> Roberto

Thanks