arendst/Tasmota #9156: Device Does Not Start Properly with GPIO2 External Switch Closed

law1964, 2020-08-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
While power cycling, if the external switch connected to GPIO2 is already CLOSED, the device activates Relay 1 and Relay 2 and then hangs. This is done before the device connects to WIFI.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): __Sonoff 4CH PRO___ 
- [x ] Tasmota binary firmware version number used: __8.2.0 and 8.4.0___
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: __Native WebUI___
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
08:29:41 CMD: Backlog Template; Module; GPIO 255
08:29:41 SRC: WebConsole from 192.168.20.20
08:29:41 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
08:29:42 SRC: Backlog
08:29:42 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
08:29:42 MQT: stat/sonoff4chpro/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
08:29:42 SRC: Backlog
08:29:42 CMD: Group 0, Index 1, Command "MODULE", Data ""
08:29:42 MQT: stat/sonoff4chpro/RESULT = {"Module":{"23":"Sonoff 4CH Pro"}}
08:29:42 SRC: Backlog
08:29:42 CMD: Group 0, Index 1, Command "GPIO", Data "255"
08:29:42 MQT: stat/sonoff4chpro/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"11":"Switch3"},"GPIO3":{"12":"Switch4"},"GPIO4":{"23":"Relay3"},"GPIO5":{"22":"Relay2"},"GPIO9":{"18":"Button2"},"GPIO10":{"19":"Button3"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"20":"Button4"},"GPIO15":{"24":"Relay4"},"GPIO16":{"0":"None"}}
08:29:42 UDP: Packet (119)

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
NA - Not using rules.

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
08:31:52 SRC: WebConsole from 192.168.20.20
08:31:52 CMD: Group 0, Index 1, Command "STATUS", Data "0"
08:31:52 MQT: stat/sonoff4chpro/STATUS = {"Status":{"Module":23,"DeviceName":"Garage Door One","FriendlyName":["Garage Door One","Garage Door Two","GD1 is OPEN","GD2 is OPEN"],"Topic":"sonoff4chpro","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,1,1,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:31:52 MQT: stat/sonoff4chpro/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:27:27","StartupUTC":"2020-08-24T12:04:25","Sleep":50,"CfgHolder":4617,"BootCount":270,"BCResetTime":"2020-05-07T16:23:49","SaveCount":35711,"SaveAddress":"F6000"}}
08:31:52 MQT: stat/sonoff4chpro/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-07-29T12:05:28","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"375/699"}}
08:31:52 MQT: stat/sonoff4chpro/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["neuronet",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00408009","0105C8000100060000005AFF000000000000","00000001","00006000"]}}
08:31:52 MQT: stat/sonoff4chpro/STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
08:31:52 MQT: stat/sonoff4chpro/STATUS5 = {"StatusNET":{"Hostname":"sonoff4chpro-2957","IPAddress":"192.168.20.42","Gateway":"192.168.20.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.20.1","Mac":"DC:4F:22:80:2B:8D","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
08:31:52 MQT: stat/sonoff4chpro/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.20.4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_802B8D","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:31:52 MQT: stat/sonoff4chpro/STATUS7 = {"StatusTIM":{"UTC":"2020-08-24T12:31:52","Local":"2020-08-24T08:31:52","StartDST":"2020-03-08T00:00:00","EndDST":"2020-11-01T00:00:00","Timezone":"-04:00","Sunrise":"06:14","Sunset":"19:53"}}
08:31:52 MQT: stat/sonoff4chpro/STATUS10 = {"StatusSNS":{"Time":"2020-08-24T08:31:52","Switch3":"ON","Switch4":"OFF"}}
08:31:52 MQT: stat/sonoff4chpro/STATUS11 = {"StatusSTS":{"Time":"2020-08-24T08:31:52","Uptime":"0T00:27:27","UptimeSec":1647,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"neuronet","BSSId":"14:CC:20:B8:F2:85","Channel":8,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:06"}}}
08:32:01 UDP: Packet (119)

```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
08:34:49 SRC: WebConsole from 192.168.20.20
08:34:49 CMD: Group 0, Index 1, Command "WEBLOG", Data "4"
08:34:49 MQT: stat/sonoff4chpro/RESULT = {"WebLog":4}
08:34:49 UDP: Packet (119)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Connect external switch to GPIO2 (SDA on the Sonoff 4CH Pro) and configure to Switch 3.
2. Close Switch 3 and keep closed.
3. Unplug and plug power supply.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
When the steps to reproduce are used, Relay 1 and Relay 2 are activated. In this case my garage doors open. LEDS for Switch 1, Switch 2 and Switch 3 are lit and stay lit. The device hangs before connecting to Wifi because it is unresponsive when I try the WebUI.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Switches 1 and 2 are used to control my garage doors. Switches 3 and 4 (GPIO2 and GPIO3 respectively) are connected to external reed switches and are used to detect whether the doors are open. The problem only occurs when GPIO2 is closed during power cycle. When GPIO3 is closed during power cycle, the boot up of the device is normal. The problem can only be cleared when the device is power cycled and GPIO2 is OPEN. 

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-08-24:
Hi,

Sorry, That is not Tasmota related. That is a limitation of the ESP8266 chip and can not be addressed by software due to are the BOOT mode selection for the chip.

Please check the docs at: https://tasmota.github.io/docs/Peripherals/#usable-pins for more information on available GPIOs and Restrictions.

ascillato2, 2020-08-24:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-08-24 by ascillato2

law1964, 2020-08-24:
Thank you for the prompt reply, Adrian.