arendst/Tasmota #9157: Xiaomi Zigbee door sensor MCCGQ01LM battery status is not reported

Yveaux, 2020-08-24:
### PROBLEM DESCRIPTION
The Xiaomi Zigbee door sensor reports its battery status once in a while, or when its button is pressed shortly.
However, Xiaomi uses a custom message type that is currently not supported by Tasmota.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Zb Bridge (but same behavior on Wemos D1 mini + CC2530)
- [x] Tasmota binary firmware version number used: 8.4.0.3
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
19:30:35 MQT: tasmota/stat/ZigbeeCoordinator/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:30:35 MQT: tasmota/stat/ZigbeeCoordinator/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
19:30:36 MQT: tasmota/stat/ZigbeeCoordinator/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"165":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"166":"Zigbee Rx"},"GPIO4":{"215":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"158":"LedLinki"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"17":"Button1"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:31:49 MQT: tasmota/stat/ZigbeeCoordinator/STATUS = {"Status":{"Module":75,"DeviceName":"Zigbee Coordinator","FriendlyName":["Zigbee Coordinator"],"Topic":"ZigbeeCoordinator","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:31:49 MQT: tasmota/stat/ZigbeeCoordinator/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T02:45:15","StartupUTC":"2020-08-24T15:46:34","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2020-08-24T16:31:18","SaveCount":17,"SaveAddress":"F4000"}}
19:31:49 MQT: tasmota/stat/ZigbeeCoordinator/STATUS2 = {"StatusFWR":{"Version":"8.4.0.3(zbbridge)","BuildDateTime":"2020-08-23T16:30:46","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"402/699"}}
19:31:49 MQT: tasmota/stat/ZigbeeCoordinator/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xyz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006002","00000000"]}}
19:31:49 MQT: tasmota/stat/ZigbeeCoordinator/STATUS4 = {"StatusMEM":{"ProgramSize":528,"Free":472,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F100786","04500001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,21,23,41","Sensors":""}}
19:31:49 MQT: tasmota/stat/ZigbeeCoordinator/STATUS5 = {"StatusNET":{"Hostname":"ZigbeeCoordinator","IPAddress":"192.168.1.129","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.x","Mac":"A4:CF:12:xx:xx:xx","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:31:49 MQT: tasmota/stat/ZigbeeCoordinator/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.x","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D7F05A","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:31:49 MQT: tasmota/stat/ZigbeeCoordinator/STATUS7 = {"StatusTIM":{"UTC":"2020-08-24T18:31:49","Local":"2020-08-24T19:31:49","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:56","Sunset":"19:47"}}
19:31:49 MQT: tasmota/stat/ZigbeeCoordinator/STATUS10 = {"StatusSNS":{"Time":"2020-08-24T19:31:49"}}
19:31:49 MQT: tasmota/stat/ZigbeeCoordinator/STATUS11 = {"StatusSTS":{"Time":"2020-08-24T19:31:49","Uptime":"0T02:45:15","UptimeSec":9915,"Vcc":3.466,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxxx","BSSId":"78:8A:20:xx:xx:xx","Channel":11,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
* Join the MCCGQ01LM to the Zb Bridge
* Short press the button on the MCCGQ01LM
* Output on Tasmota console:
```
18:39:19 ZIG:
{"ZbEZSPReceived":"45000004010000010100010000D3FFFB210AFFFF1D1C3412090A02FF4C0600100021F90B21A813240600000000214F002059"}
18:39:19 MQT: tasmota/tele/ZigbeeCoordinator/SENSOR = {"ZbReceived":{"0x0A21":{"Device":"0x0A21","0000/FF02":null,"0000/F921":605268001,"DateCode":0,"DateCode2":null,"ZCLVersion":79,"Endpoint":1,"LinkQuality":216}}}
```
* The raw data 02FF4C0600100021F90B holds the battery voltage

### EXPECTED BEHAVIOUR
* Correct decoding of 02FF4C0600100021F90B (refer to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1069):
Last word holds battery voltage: 0x F9 0B => 0x0BF9 = 3065 milliVolts = 3.065 Volts
See also https://raw.githubusercontent.com/veeceeoh/xiaomi-hubitat/master/devicedrivers/xiaomi-door-window-sensor-hubitat.src/xiaomi-door-window-sensor-hubitat.groovy (search for 0xFF02)
* Showing the battery percentage/voltage in the WebUI
* Reporting the battery percentage/voltage through MQTT

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
Side question: Is it possible to expose the raw zigbee data through MQTT? (the {"ZbEZSPReceived":"45000004010000010100010000D4FFFC210AFFFF1D1C34120A0A02FF4C0600100021F90B21A813240100000000214F00205A"} string)

s-hadinger, 2020-08-25:
Thanks for the detailed logs. Indeed Z2T doesn't support the ZCL `struct` (0x4C) data type from `4C0600100021F90B`. This is why the rest of the fields are corrupt. This is non-trivial so I need to find out a solution for it.

s-hadinger, 2020-08-25:
Here is the message you will receive for the example above:
```
SENSOR = {"0x0A21":{"Device":"0x0A21","0000/FF02":"0600100021F90B21A813240600000000214F002059","BatteryVoltage":3.065,"BatteryPercentage":100,"Endpoint":1,"LinkQuality":216}}
```

I have left the whole struct content `0600100021F90B21A813240600000000214F002059`, I don't know what are the other parameters. The battery status is the first attribute of type `21`.

It looks that the byte number 4 `00` is the status of the door sensor. Can you please confirm?

Yveaux, 2020-08-25:
Thanks for the insanely fast fix! :+1:
Is there a place where I can quicly get the binary firmware from your branch?

closed 2020-08-25 by s-hadinger

s-hadinger, 2020-08-25:
The latest binary can be found here: http://thehackbox.org/tasmota/

PR is now merged, you may need to wait for up to 4 hours for the binary to get updated. Just check the commit id. Wait until it shows `e101d08` or a later commit.

Jason2866, 2020-08-25:
You find the firmware here too. Just no OTA possible.
https://github.com/arendst/Tasmota/tree/firmware/firmware/tasmota

Yveaux, 2020-08-27:
> I'm interested in the same logs with the latest version. Do you get the sensor value in other messages?

Here are two of the hourly battery status messages (firmware 8.4.0.3(e101d08-zbbridge)):
```
2020-08-27T17:22:39.721Z        tasmota/tele/ZigbeeCoordinator/SENSOR   {"ZbReceived":{"WoonkamerAchterDeur":{"Device":"0x0A21","Name":"WoonkamerAchterDeur","0000/FF02":"0600100021E50B21A813240100000000214F00205C","BatteryVoltage":3.045,"BatteryPercentage":100,"Endpoint":1,"LinkQuality":200}}}
2020-08-27T17:58:11.645Z        tasmota/tele/ZigbeeCoordinator/SENSOR   {"ZbReceived":{"WoonkamerAchterRaam":{"Device":"0xF5B0","Name":"WoonkamerAchterRaam","0000/FF02":"0600100021E50B21A813240100000000213800205A","BatteryVoltage":3.045,"BatteryPercentage":100,"Endpoint":1,"LinkQuality":197}}}
```

And some regular switching messages:
```
2020-08-27T18:24:01.674Z        tasmota/tele/ZigbeeCoordinator/SENSOR   {"ZbReceived":{"WoonkamerAchterDeur":{"Device":"0x0A21","Name":"WoonkamerAchterDeur","Power":1,"Endpoint":1,"LinkQuality":174}}}
2020-08-27T18:24:02.898Z        tasmota/tele/ZigbeeCoordinator/SENSOR   {"ZbReceived":{"WoonkamerAchterDeur":{"Device":"0x0A21","Name":"WoonkamerAchterDeur","Power":0,"Endpoint":1,"LinkQuality":187}}}
2020-08-27T18:24:04.537Z        tasmota/tele/ZigbeeCoordinator/SENSOR   {"ZbReceived":{"WoonkamerAchterRaam":{"Device":"0xF5B0","Name":"WoonkamerAchterRaam","Power":1,"Endpoint":1,"LinkQuality":158}}}
2020-08-27T18:24:06.090Z        tasmota/tele/ZigbeeCoordinator/SENSOR   {"ZbReceived":{"WoonkamerAchterRaam":{"Device":"0xF5B0","Name":"WoonkamerAchterRaam","Power":0,"Endpoint":1,"LinkQuality":160}}}
```

I've not received any other messages so far.

Battery status is also displayed nicely on WebUI now:

[image]

Thanks again!

s-hadinger, 2020-08-29:
@Yveaux thanks for the feedback.

PR #9162 (2020-08-25): Zigbee adding decoder for Xiaomi door sensor battery
## Description:

**Related issue (if applicable):** fixes #9157 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.1
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change must pass CI tes [... cut]