arendst/Tasmota #9159: Rule on APDS9960 gesture does not fire

rowanalex123, 2020-08-24:
### PROBLEM DESCRIPTION
Rule example for when a gesture happens [given in the docs](https://tasmota.github.io/docs/APDS-9960/) does not fire
E.g: 
Rule1 on Tele-APDS9960#Up=1 do backlog displaymode 0; DisplayText [zs4l1c1d]Test; endon

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): NodeMCU
- [ ] Tasmota binary firmware version number used: 8.4.0
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [ ] IDE / Compiler used: __gitpod___
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
20:04:51 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
20:04:51 RSL: stat/tasmota_502452/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
20:04:52 SRC: Backlog
20:04:52 CMD: Group 0, Index 1, Command "MODULE", Data ""
20:04:52 RSL: stat/tasmota_502452/RESULT = {"Module":{"18":"Generic"}}
20:04:52 SRC: Backlog
20:04:52 CMD: Group 0, Index 1, Command "GPIO", Data "255"
20:04:52 RSL: stat/tasmota_502452/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"46":"PWM1i"},"GPIO13":{"0":"None"},"GPIO14":{"8":"IRsend"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
20:03:37 RSL: stat/tasmota_502452/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":80,"Free":431,"Rules":"on Tele-APDS9960#Up do backlog displaymode 0; DisplayText [zs4l1c1d]Test; endon"}
20:03:38 SRC: Backlog
20:03:38 CMD: Group 0, Index 2, Command "RULE", Data ""
20:03:38 RSL: stat/tasmota_502452/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":101,"Free":410,"Rules":"on APDS9960#Up=1 do backlog power1 1; pulsetime1 1; displaymode 0; DisplayText [zs4l1c1d]Test; endon"}
20:03:38 SRC: Backlog
20:03:38 CMD: Group 0, Index 3, Command "RULE", Data ""
20:03:38 RSL: stat/tasmota_502452/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
20:04:26 CMD: Group 0, Index 1, Command "STATUS", Data "0"
20:04:26 RSL: stat/tasmota_502452/STATUS = {"Status":{"Module":18,"DeviceName":"espgesture1","FriendlyName":["espgesture1","Tasmota2"],"Topic":"tasmota_502452","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":10,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:04:26 RSL: stat/tasmota_502452/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T01:03:01","StartupUTC":"2020-08-24T18:01:25","Sleep":50,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2020-08-16T16:13:11","SaveCount":111,"SaveAddress":"F7000"}}
20:04:26 RSL: stat/tasmota_502452/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-08-14T05:23:30","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"393/699"}}
20:04:26 RSL: stat/tasmota_502452/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["abcd","abcd2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000"]}}
20:04:26 RSL: stat/tasmota_502452/STATUS4 = {"StatusMEM":{"ProgramSize":664,"Free":336,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":2,"Features":["00000809","8FFAE797","043683A1","420000CD","8D0013C0","C000F981","00004064"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,13,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,22,27,45,46,50,67"}}
20:04:26 RSL: stat/tasmota_502452/STATUS5 = {"StatusNET":{"Hostname":"espgesture1","IPAddress":"192.168.1.70","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"48:3F:DA:50:24:52","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:04:26 RSL: stat/tasmota_502452/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_502452","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:04:26 RSL: stat/tasmota_502452/STATUS7 = {"StatusTIM":{"UTC":"2020-08-24T19:04:26","Local":"2020-08-24T20:04:26","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:56","Sunset":"19:47"}}
20:04:27 RSL: stat/tasmota_502452/STATUS10 = {"StatusSNS":{"Time":"2020-08-24T20:04:26"}}
20:04:27 RSL: stat/tasmota_502452/STATUS11 = {"StatusSTS":{"Time":"2020-08-24T20:04:27","Uptime":"0T01:03:02","UptimeSec":3782,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"POWER1":"OFF","Dimmer":74,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"abcd","BSSId":"D4:6E:0E:36:CA:C8","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
20:07:49 Up
20:07:49 RSL: tele/tasmota_502452/SENSOR = {"Time":"2020-08-24T20:07:49","APDS9960":{"Up":1}}

```

### TO REPRODUCE
Set `Sensor27 1`
Define a rule on a APDS9960 gesture and trigger the gesture (up, down etc). Rule is not run.

### EXPECTED BEHAVIOUR
Rule is run once per gesture

ascillato, 2020-08-25:
Hi,

From your log:

> {"Rule2":"OFF",

Your rule is NOT ENABLED

You need to turn it on by command `rule2 1`

rowanalex123, 2020-08-25:
According to the documentation, the rule syntax is on Tele-APDS9960#Up=1. But a rule in that form (rule 1) or using `on Tele-APDS9960#Up` (skipping the `=1`) does not fire. 

Does the documentation need to be fixed to show the syntax as `on APDS9960#Up=1 do` instead of using `tele-APDS9960`?

ascillato, 2020-08-25:
> Does the documentation need to be fixed to show the syntax as on APDS9960#Up=1 do instead of using tele-APDS9960?

Sorry, I don't understand. The docs already explains that the trigger without the condition is ignored. What do you want to change?

ascillato2, 2020-08-25:
Hi, I think I found the confusion. The docs were updated with the correct information 2 days ago.

Now they are ok with the correct trigger.

Thanks.

closed 2020-08-25 by ascillato2

rowanalex123, 2020-08-25:
@ascillato please reopen this issue, dont think old documentation is the issue.

This rule does not fire - `rule1 on Tele-APDS9960#Up=1 do backlog pulsetime1 1;power1 1; endon`

i've tried multiple variants, only one that does fire is `rule1 on APDS9960#Up=1 do backlog pulsetime1 1;power1 1; endon`

I'm trying to understnd if the the documentation wrong or if this a bug. 

In my build I have these two enabled, if that's relevant.

`#define  USE_EXPRESSION
#define  SUPPORT_IF_STATEMENT
`

ascillato, 2020-08-25:
Hi,

Checking the code, the mqtt message is sent to the `tele/...` topic but in the rule processing routine (that is called right after that), it is being checked only the JSON argument, so, the `tele` prefix is not needed anymore for this sensor.

So, the docs needs an update for this sensor.

Are you able to do the docs editing?

rowanalex123, 2020-08-26:
@ascillato Thanks for fixing the docs. When using `Rule on APDS9960#Long=1`, the rule is being triggered twice.

`13:00:01 HTP: Console
13:00:05 Up
13:00:05 RSL: tele/tasmota_502452/SENSOR = {"Time":"2020-08-26T13:00:05","APDS9960":{"Up":1}}
13:00:05 RUL: APDS9960#UP=1 performs "backlog pulsetime1 1;power1 1;"
13:00:05 SRC: Rule
13:00:05 CMD: Group 0, Index 1, Command "BACKLOG", Data "pulsetime1 1;power1 1;"
13:00:05 SRC: Backlog
13:00:05 CMD: Group 0, Index 1, Command "PULSETIME", Data "1"
13:00:05 RSL: stat/tasmota_502452/RESULT = {"PulseTime1":{"Set":1,"Remaining":1}}
13:00:06 SRC: Backlog
13:00:06 CMD: Group 0, Index 1, Command "POWER", Data "1"
13:00:06 RSL: stat/tasmota_502452/RESULT = {"POWER1":"ON"}
13:00:06 RSL: stat/tasmota_502452/POWER1 = ON
13:00:06 SRC: PulseTimer
13:00:06 RSL: stat/tasmota_502452/RESULT = {"POWER1":"OFF"}
13:00:06 RSL: stat/tasmota_502452/POWER1 = OFF
13:00:06 RUL: APDS9960#UP=1 performs "backlog pulsetime1 1;power1 1;"
13:00:06 SRC: Rule
13:00:06 CMD: Group 0, Index 1, Command "BACKLOG", Data "pulsetime1 1;power1 1;"
13:00:06 SRC: Backlog
13:00:06 CMD: Group 0, Index 1, Command "PULSETIME", Data "1"
13:00:06 RSL: stat/tasmota_502452/RESULT = {"PulseTime1":{"Set":1,"Remaining":1}}`

ascillato2, 2020-08-27:
> the rule is being triggered twice.

Just for reference to anyone reaching this, the thread continued in https://github.com/arendst/Tasmota/issues/9169 and this reported bug is fixed now.