arendst/Tasmota #9167: FUNC_COMMAND not triggered on xnrg devices

hallard, 2020-08-26:
### PROBLEM DESCRIPTION

The API documentation https://tasmota.github.io/docs/API/ states that for `xnrg` devices the `FUNC_COMMAND` is passed to device with order `3`

Looks likes this is never happening, I even tried to Log in this case and whatever command I place, I never log anything

```cpp
/*********************************************************************************************\
 * Interface
\*********************************************************************************************/
bool Xnrg15(uint8_t function)
{
    bool result = false;
    
    switch (function)
    {
        case FUNC_EVERY_250_MSECOND:
            if (uptime > 4) { TInfoEvery250ms(); }
            break;
        case FUNC_JSON_APPEND:
            TInfoShow(1);
            break;
    #ifdef USE_WEBSERVER
        case FUNC_WEB_SENSOR:
            TInfoShow(0);
            break;
    #endif  // USE_WEBSERVER
        case FUNC_INIT:
            TInfoInit();
            break;
        case FUNC_PRE_INIT:
            TInfoDrvInit();
            break;
      case FUNC_COMMAND:
        AddLog_P2(LOG_LEVEL_DEBUG, PSTR(D_LOG_TELEINFO "FUNC_COMMAND"));
        result = DecodeCommand(kTeleinfoCommands, TeleinfoCommand);
        break;

    }
    return false;
}
```

full source code on this device is on this branch here 
https://github.com/hallard/Tasmota/blob/teleinfo/tasmota/xnrg_15_teleinfo.ino

closed 2020-08-26 by close-issue-app[bot]

arendst, 2020-08-27:
Charles, as implemented in xnrg drivers it uses a subset of the full FUNC_COMMAND. The main goal here wat to satisfy calibration commands with at less code overhead as possible.

Search for FUNC_COMMAND and see how it is implemnted using function ``XnrgCall(FUNC_COMMAND)``

hallard, 2020-08-27:
Theo,
Thanks, I'll look into that ASAP and get back here if for question.