arendst/Tasmota #9171: Error on command

stehof2, 2020-08-27:
### PROBLEM DESCRIPTION
Hi,

First my setup:
- NodeMCU v3
- BH1750 light sensor
- Tasmota 8.4.0 (tasmota-sensor.bin)

I’m having an issue when I try to command an option, this feature is quite new, it is from issue #8016 https://github.com/arendst/Tasmota/issues/8016

When I command Sensor10 1 to change the resolution, I get response Command:error without further explanation what went wrong.

This is the output from the console:
22:30:43 CMD: Sensor10 1
22:30:43 SRC: WebConsole from 192.168.0.147
22:30:43 CMD: Group 0, Index 10, Command "SENSOR", Data "1"
22:30:43 MQT: stat/LuxKeuken/RESULT = {"Command":"Error"}
22:30:58 WIF: Checking connection...

Same result when I try to set the option over a http request as from the console.

Any idea why this isn’t working for me?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): NodeMCU v3
- [ ] Tasmota binary firmware version number used: 8.4.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x] Flashing tools used: ESPHome for Macosx
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

09:32:06 CMD: Backlog Template; Module; GPIO 255
09:32:06 SRC: WebConsole from 192.168.0.147
09:32:06 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
09:32:06 SRC: Backlog
09:32:06 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
09:32:07 MQT: stat/LuxKeuken/RESULT = {"NAME":"Generic","GPIO":[6,255,5,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
09:32:07 SRC: Backlog
09:32:07 CMD: Group 0, Index 1, Command "MODULE", Data ""
09:32:07 MQT: stat/LuxKeuken/RESULT = {"Module":{"18":"Generic"}}
09:32:07 SRC: Backlog
09:32:07 CMD: Group 0, Index 1, Command "GPIO", Data "255"
09:32:07 MQT: stat/LuxKeuken/RESULT = {"GPIO0":{"6":"I2C SDA"},"GPIO1":{"0":"None"},"GPIO2":{"5":"I2C SCL"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
09:32:18 WIF: Checking connection

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

09:34:36 CMD: Status0
09:34:36 SRC: WebConsole from 192.168.0.147
09:34:36 CMD: Group 0, Index 0, Command "STATUS", Data ""
09:34:36 MQT: stat/LuxKeuken/STATUS = {"Status":{"Module":18,"DeviceName":"LuxKeuken","FriendlyName":["Tasmota"],"Topic":"LuxKeuken","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:34:38 WIF: Checking connection

```
- [ x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

22:30:43 CMD: Sensor10 1
22:30:43 SRC: WebConsole from 192.168.0.147
22:30:43 CMD: Group 0, Index 10, Command "SENSOR", Data "1"
22:30:43 MQT: stat/LuxKeuken/RESULT = {"Command":"Error"}
22:30:58 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Use the same setup

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I would expected to change the resolution from default 0 to 1 (high)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

device111, 2020-08-27:
you must use Bh1750Resolution Command:

Bh1750Resolution1   0..2    - Set BH1750 1 resolution mode

stehof2, 2020-08-27:
> Bh1750Resolution1

Thanks, that seems to do the trick, how did you find out, couldn’t find anything on it in the documentation beside of Sensor10?

And do you also know argument for setting the measurement time? Tried several combination with Bh1750 and Measurement1, MeasurementTime1 but that’s not working :(

device111, 2020-08-27:
Ok, seems there is  not yet in the docs.

Here are the commands from source:

*Bh1750Resolution1   0..2    - Set BH1750 1 resolution mode
 * Bh1750Resolution2   0..2    - Set BH1750 2 resolution mode
 * Bh1750MTime1       30..255  - Set BH1750 1 MT register
 * Bh1750MTime2       30..255  - Set BH1750 2 MT register

stehof2, 2020-08-27:
> Ok, seems there is not yet in the docs.
>
> Here are the commands from source:
>
> * Bh1750Resolution1   0..2    - Set BH1750 1 resolution mode
> * Bh1750Resolution2   0..2    - Set BH1750 2 resolution mode [...]

Thanks, just what I was looking for! It is working now 👍

closed 2020-08-27 by stehof2

device111, 2020-08-27:
have create a Pull Request in Docs for BH1750 commands.