arendst/Tasmota #9178: Build instruction for GZ bin ?

poofyteddy, 2020-08-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I am using a shelly 2.5 and with the lates version of tasmota, my build version with ssl is over 645mb.
Since i though i read that 2MB flash wasn't supported, my next option if gz compressed bin.
I can't find any build instruction to generate such a binary, and if i just gzip -9 the bin file i get a checksum error after uploading.

Can you document the way to build them ?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly 2.5
- [X] Tasmota binary firmware version number used: Building from devel, shelly running minimal firmware
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Vscode + platformio
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
NA

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
NA

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
NA

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
NA

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
NA

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
NA

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
NA

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-08-27:
The default platform.ini file makes the bin and gz files.

Just run the default compilation and you will have both files. 

If you need further assistant on this, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-08-27 by ascillato2

poofyteddy, 2020-08-29:
So to reply to myself after looking without finding
i should look for bin in build_output/firmware and not .pioenvs/tasmota, there i did find the .gz
and in order to have working .gz, and not checksum error, i needed a tasmota-minimal more recent that 8.1.0 that i was using.

Thank you and sorry