arendst/Tasmota #9179: more FEIT BPA800/RGBW/AG/2(P) Template issues

mrneutron42, 2020-08-27:
### PROBLEM DESCRIPTION
I just got 10 of the Feit RGBW A19 bulbs with model number BPA800/RGBW/AG/2P (FCC ID SYW-A19RGBWAG2) and I've loaded the latest pre-compiled Tasmota firmware version:  8.4.0.
I see there have been problems with these Feit template files in the past year:  https://github.com/arendst/Tasmota/issues/6534

The current posted template for these Feit BPA800/RGBW/AG/2P bulbs does not allow control of color - NO color sliders appear in the Tasmota Main web server page, only the CoolWhite-WarmWhite and Intensity sliders.
{"NAME":"BPA800/RGBW/AG/2P","GPIO":[0,0,0,0,37,38,0,0,141,142,140,0,0],"FLAG":0,"BASE":48}
from https://templates.blakadder.com/feit_electric-BPA800RGBWAG2P.html

The Tasmota Component choices 140, 141, & 142 do not appear in the pull-down menu choices for the GPIOs in the Tasmota Configure Template window.  As a result, GPIO boxes 12, 13 and 14 are empty.  According to the template page, GPIO boxes 12, 13, 14 are the SM16716 LED driver chip's i2c Data, Power, and i2s Clock pins.

The end goal is to get color control of the Feit BPA800/RGBW/AD/2P bulbs.
Can Tasmota Component choices 140, 141, & 142 be added to the GUI pull-down choices so they can be selected in the Configure 
[image]
[image]
[image]
[image]

Template GUI? 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

JaimeZX, 2020-08-29:
Try doing the upgrade. I'm running 8.4.0 and
{"NAME":"","GPIO":[0,0,0,0,37,47,0,0,141,142,140,0,0],"FLAG":0,"BASE":48}

mrneutron42, 2020-08-29:
> Try doing the upgrade. I'm running 8.4.0 and the below is giving me satisfactory control:
> {"NAME":"","GPIO":[0,0,0,0,37,47,0,0,141,142,140,0,0],"FLAG":0,"BASE":48}
I already have version 8.4.0 loaded (see 2nd line of 1st post).
I just uploaded the 8.4.0 firmware file again...no change....no color control.
Your template yields the same no-color results.

mrneutron42, 2020-08-29:
OK...problem diagnosed.   I've got color control!
I was following the Migration Path instructions and manually loading the "lite" versions of the bin files.
see https://tasmota.github.io/docs/Upgrading/#migration-path
Apparently, the "lite" versions don't support the SM16716 LED driver chip?
I tried loading the full tasmota.bin file and it failed.
[image]
So, I tried loading the tasmota-minimal.bin.  That worked.
Then, I tried the full tasmota.bin file again, and it worked.

For future reference, during firmware upgrade/migration, can the minimal versions be loaded instead of the "lite" versions?

closed 2020-08-29 by mrneutron42

ascillato, 2020-08-29:
Tasmota minimal is only for upgrading. It is a firmware just to free space and let the full version to be installed by OTA. Minimal does not have any driver and is not for normal usage or first flash. It is just an interim firmware in order to be able to update to the full version.

The lite version is just a Tasmota small version with few drivers intended for devices with poor power regulators. 

For normal usage use tasmota.bin that is the full version.

mrneutron42, 2020-08-29:
For the record, my final template for these Feit BPA800/RGBW/AG/2P RGBW bulbs is:
{"NAME":"BPA800/RGBW/AG/2P","GPIO":[0,0,0,0,37,47,0,0,141,142,140,0,0],"FLAG":0,"BASE":48}

GPIO5 needs to be an inverted PWM to correctly control the CoolWhite-WarmWhite slider.
I also eliminated the leading space in the name BPA800/RGBW/AG/2P, because it caused the last P to be truncated in the Tasmota GUI.

JaimeZX, 2020-08-29:
Glad you got it worked out!

mrneutron42, 2020-08-29:
> Glad you got it worked out! If you figure out how to change the White Color Temp from Home Assistant, let me know. ;)

OK, will do.  I don't have a Home Assistant setup, yet.  Are you using a Raspberry Pi3 or Pi4?

JaimeZX, 2020-08-30:
Just got a Pi4 off Amazon. HA install was easy. Adding devices was easy. These bulbs will automatically give you Color Temp, Color Picker, and Dimmer in the HA GUI. But trying to do a smooth dim or change color / color temp from a script or automation is so far escaping me.