arendst/Tasmota #9181: shelly 1 Tasmota as momentary switch with feedback to Alexa - Rule required

kamilo558, 2020-08-28:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

Hello to all!

Is there a function or setting in Tasmota that I use the shelly 1 as a push button with e.g. impulsetime 10 (sec) and the shelly internally stores this switching state and passes it to Alexa although the push button has switched off the output after one second?

So:

- Alexa switch on
- Switch sets output for one second and switches off again (output goes to a current surge relay) but passes switching state 1 to Alexa

How could you get Alexa to send the switch a pulse signal again? I would have to say "Alexa, switch on" again to generate a pulsetime. But if I say Alexa switch off, no pulse would be generated by the switch...

Thanks a lot already.

s-hadinger, 2020-08-28:
It's not supported out of the box, but maybe you can achieve this with scripting (not Rules). 

https://tasmota.github.io/docs/Scripting-Language/

kamilo558, 2020-08-28:
OK. Is there a more simple way?

Jason2866, 2020-08-28:
No

ascillato2, 2020-08-28:
Closing this issue as it has been answered.

If you need further assistant, please just ask here or in the Tasmota support chat. Thanks. 

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-08-28 by ascillato2