arendst/Tasmota #9183: Add support for infrared thermal sensor array MLX90641 / MLX90640

balajisraghavan, 2020-08-28:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? 
No
**Describe the solution you'd like**  
Add support for the Infrared thermal sensor array MLX90641 & MLX90640. It has a variety of uses - human presence detection, air leaks, a low cost thermal camera & many more.

**Additional context**  
I'd like Tasmota to publish the temperature at each of the sensors in the array over MQTT during every tele-period. Some kind of rule support such as min/max temperature detected exceeding threshold would be a great addition as well. A reliable  open source Arduino driver is available for this sensor.

Staars, 2020-08-29:
I have a MLX90640 and will try write a driver in the next few weeks, which I need too for a personal project.

balajisraghavan, 2020-08-30:
Thanks @Staars. I got the sensor working over the weekend and raised a PR. It's my first contribution to the Tasmota codebase, would be great to receive feedback on what can be done better

ascillato2, 2020-09-13:
Closing this issue as this feature has been added. Thanks.

closed 2020-09-13 by ascillato2

Erickrk, 2023-03-10:
Hello everyone,

I hope you are doing well. I wanted to check if anyone is still actively testing this project. I tried running it myself, but it doesn't seem to be very stable. I am not sure if it's because I am running a script while working with the camera or if there are issues with the driver itself.

During my testing, I noticed that my heap usage goes up to 130 on my ESP32 while running this project with an MLX90640. Additionally, it seems like the WEB UI might be causing some instability. I have noticed that whenever I was able to break the device, it was due to issues with the WEB UI.

If anyone has any suggestions or insights on how to improve the stability of this project, it would be greatly appreciated. Thank you for your time and help.

Staars, 2023-03-11:
Sounds like your system is indeed under heavy load.

The MLX90640 was primarily designed to be used with the MQTT messages, the Web UI should only make it easy to position the POI's (and it looks quite fancy 😄 ).
Working together with scripting was never tested and generally spoken: The more you add, the more you move to undefined behaviour as the ESP is running out of resources.
There is a ton of variables, that influence such edge cases - i.e. modern browsers tend to turn down traffic via AJAX when another window has the focus and so on. Non-perfect WiFi (the router) is the most common problem.

So I never tested the Web-UI for a prolonged time period, but I will try to find some time to reevaluate the sensor on an ESP32 with our latest builds.

Erickrk, 2023-03-15:
Thank you for your response! I would like to mention that although I appreciate your input, I am still encountering negative behavior and am currently working to address it at the hardware level.
And the UI indeed looks fancy! hahaha