arendst/Tasmota #9186: thehackbox org DNS not resolving

mrmikee, 2020-08-28:
### PROBLEM DESCRIPTION
thehackbox.org DNS not resolving

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
$ nslookup thehackbox.org 8.8.8.8
Server:		8.8.8.8
Address:	8.8.8.8#53

** server can't find thehackbox.org: SERVFAIL

$ nslookup thehackbox.org
Server:		192.168.9.1
Address:	192.168.9.1#53

** server can't find thehackbox.org: SERVFAIL

### EXPECTED BEHAVIOUR
DNS to resolve IP address

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-08-29:
The Tasmota OTA server http://hackbox.org/ is temporary down.
Please use http://sidweb.nl/tasmota or http://sidweb.nl/tasmota32

Jason2866, 2020-08-29:
[Tasmotizer 1.1b](https://github.com/tasmota/tasmotizer/tree/sidweb) is using http://sidweb.nl as source

ascillato2, 2020-08-29:
From the announcements in Tasmota Support chat:

The Tasmota OTA server http://thehackbox.org/ is down.
Please, use http://ota.tasmota.com/tasmota and http://ota.tasmota.com/tasmota32

Thanks.

closed 2020-08-29 by ascillato2