arendst/Tasmota #9192: esp32 TLS Secured MQTT

dabos1, 2020-08-29:
Hello,

It would be nice to have mqtt over tls support for the esp32. How it already works with the esp8266. 

Thanks.

**Have you looked for this feature in other issues and in the docs?**  
yes 

**Is your feature request related to a problem? Please describe.**  
no
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  

_A clear and concise description of what you want to happen._

add mqtt tls with fingerprint on esp32

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

balajisraghavan, 2020-08-30:
+1 Tasmota with MQTT over TLS on the ESP32 would be very useful. Porting the existing BearSSL TLS support for ESP8266 would work, if that's possible.

xhess, 2020-09-02:
+1 I would also love to see this.

stefanbode, 2020-09-04:
+1 me too. But I assume there is no TLS version available right now

Elix-g, 2020-09-04:
+1 Would very appreciate it too

> +1 me too. But I assume there is no TLS version available right now

mbedTLS

Jason2866, 2020-09-12:
@Elix-g  no since Tasmota uses a different mqtt library (PubSubClient-EspEasy-2.7.12).
The lib is highly customized for the usage with Tasmota and will NOT be replaced.

closed 2020-10-24 by stale[bot]