arendst/Tasmota #9199: Sonoff ZBBridge as zigbee router?

Alfiegerner, 2020-08-30:
**Have you looked for this feature in other issues and in the docs?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

Would love to use sonoff ZBBridge as a zigbee router, to extend network reliably.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

Firmware to flash to allow the bridge to act as a router / repeater.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-08-31:
Since the firmware that can be flashed via OTA is signed and encrypted from ITEAD,
we have no possibility to provide a other firmware for the ZigbeeBridge.

ascillato2, 2020-08-31:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-08-31 by ascillato2

s-hadinger, 2020-08-31:
That's probably doable with the current firmware but I have no short term plans to explore this possibility.

Lefuneste83, 2020-11-05:
I believe that this would be a very highly desirable feature as :

-The Sonoff ZBbridge is small and easy to hide while keeping it powered on continuously

-Only one coordinator can be integrated within HA rendering Zigbee integration problematic on medium to large size permises as soon as the end devices are not directly in reach of the coordinator. Note that this is a generic topic with the way Zigbee integration work in HA, not specific to this device.

Ideally a common coordinator/router FW with console setup would be perfect but not sure if the MCU has enough memory to store both codes.

s-hadinger, 2020-11-06:
@Lefuneste83 I agree about the importance of having Zigbee routers, but in my experience, almost all bulbs/plugs (IEAK, OSRAM, Philips) are also acting as Zigbee routers. This mitigates the need to have dedicated routers.

Said differently, buy a Zigbee plug, you will have a router and it will be cheaper than a ZBBridge.

Unless you have different needs?

Lefuneste83, 2020-11-06:
Yes I agree this greatly mitigates the need of router functionality. It's just that it would be neat to stick to Tasmota based devices... I'm sure it will eventually be supported at some point...

genebean, 2021-05-24:
I, too, am interested in using the ZBBridge as a router simply because it’s radio is more power than any plug… or so it seems from what I’ve read in reviews.

weitheng, 2021-06-15:
Would love to have this feature too!

cnico, 2021-07-02:
For me too, it would be a great feature to convert a zbbridge in router

s-hadinger, 2021-07-02:
This was already answered above. But a plug and you will have a router. It's too much work to do it on zbbridge.

Fusseldieb, 2021-11-15:
Just bought another 3 ZBBridge devices only to know that it doesn't support router mode. 

Would love to see this implemented tbh

Jason2866, 2021-11-15:
The coder (s-hadinger) of the complete zigbee part in Tasmota answered. He will not do.