arendst/Tasmota #9201: Can't get Tasmota to work with Alexa

XDM525, 2020-08-31:
### PROBLEM DESCRIPTION
I wanted to create my own WIFI relay device using an esp8285 module that I have. I had downloaded the precompiled Tasmota bin file(version 8.4.0.3) and flashed it to my esp8285 module. I then opened it up in the web browser, set the module configuration to Sonoff basic and then selected the Sonoff basic template. I was able to control the relay through the web browser but now I want to control it through the Alexa app on my phone. Next I opened up "Configure Other" and selected the Belkin WeMo emulation however in the Alexa app, no devices are discovered. I then tried it using the Hue Bridge emulation and on the app I selected Hue Bridge V1 but still could not get it to connect to Alexa. Under "Configure Other" at the top I selected Activate by the template parameters and still no luck.  Do I need to setup MQTT in order to get it to work with Alexa? 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _ESP8285____
- [x] Tasmota binary firmware version number used: _8.4.0.3____
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: __esptool___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
00:30:56 CMD: backlog template
00:30:56 CMD: Group 0, Index 1, Command "BACKLOG", Data "template"
00:30:56 SRC: Backlog
00:30:56 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
00:30:56 RSL: RESULT = {"NAME":"Generic","GPIO":[17,255,255,255,255,0,0,0,21,56,255,0,0],"FLAG":0,"BASE":1}

00:32:44 CMD: module
00:32:44 CMD: Group 0, Index 1, Command "MODULE", Data ""
00:32:44 RSL: RESULT = {"Module":{"0":"Generic"}}

0:34:00 CMD: GPIO 255
00:34:00 CMD: Group 0, Index 1, Command "GPIO", Data "255"
00:34:00 RSL: RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
00:05:43 CMD: Status 0

00:05:43 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_BBB63F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMod [0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

00:05:43 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:08:59","StartupUTC":"2020-08-31T22:56:44","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2020-08-31T23:17:53","SaveCount":39,"SaveAddress":"F5000"}}

00:05:43 RSL: STATUS2 = {"StatusFWR":{"Version":"8.4.0.3(tasmota)","BuildDateTime":"2020-08-31T08:03:12","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"382/699"}}

00:05:43 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Summit","Summit"],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}

00:05:43 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}

00:05:43 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_BBB63F-5695","IPAddress":"192.168.1.185","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"68.105.28.16","Mac":"60:01:94:BB:B6:3F","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}

00:05:43 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-08-31T23:05:43","Local":"2020-09-01T00:05:43","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:07","Sunset":"19:31"}}

00:05:43 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-09-01T00:05:43"}}
00:05:43 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-09-01T00:05:43","Uptime":"0T00:08:59","UptimeSec":539,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Summit","BSSId":"88:DC:96:7C:22:66","Channel":1,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:05"}}}
00:06:44 APP: Serial logging disabled
00:06:54 RSL: STATE = {"Time":"2020-09-01T00:06:54","Uptime":"0T00:10:10","UptimeSec":610,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Summit","BSSId":"88:DC:96:7C:22:66","Channel":1,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:05"}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
00:11:20 CMD: weblog 4
00:11:20 RSL: RESULT = {"WebLog":4}
00:11:20 CFG: Saved to flash at F4, Count 40, Bytes 4096
00:11:29 WIF: Checking connection...
00:11:37 UDP: Packet (119)
00:11:37 UDP: Packet (119)
00:11:38 UDP: Packet (119)
00:11:38 UDP: Packet (119)
00:11:39 UDP: Packet (119)
00:11:39 UDP: Packet (119)
00:11:40 UDP: Packet (119)
00:11:40 UDP: Packet (119)
00:11:49 WIF: Checking connection...
00:11:54 RSL: STATE = {"Time":"2020-09-01T00:11:54","Uptime":"0T00:15:10","UptimeSec":910,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Summit","BSSId":"88:DC:96:7C:22:66","Channel":1,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:05"}}
00:11:54 UDP: Packet (119)
00:11:55 UDP: Packet (119)
00:11:56 UDP: Packet (119)
00:11:57 UDP: Packet (119)
00:12:00 HTP: Main Menu
00:12:02 HTP: Information
00:12:07 HTP: Main Menu
00:12:09 HTP: Console
00:12:10 WIF: Checking connection...
00:12:30 WIF: Checking connection...
00:12:33 UDP: Packet (119)
00:12:34 UDP: Packet (119)
00:12:35 UDP: Packet (119)
00:12:36 UDP: Packet (119)
00:12:50 WIF: Checking connection...
00:13:10 WIF: Checking connection...
00:13:30 WIF: Checking connection...
00:13:37 UDP: Packet (119)
00:13:37 UDP: Packet (119)
00:13:38 UDP: Packet (119)
00:13:38 UDP: Packet (119)
00:13:39 UDP: Packet (119)
00:13:39 UDP: Packet (119)
00:13:40 UDP: Packet (119)
00:13:40 UDP: Packet (119)
00:13:50 WIF: Checking connection...

```

### TO REPRODUCE
Use an ESP8285 module, I purchased mine from Adafruit, then flash it with Tasmota version 8.4.0.3. Set the configuration module to sonoff basic and the template the sonoff basic then enable the emulation and try to connect to alexa. 

### EXPECTED BEHAVIOUR
Alexa can't find any new devices 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-09-01:
What Amazon Echo device are you using?
Please set `weblog 4` and paste the logs showing when you do the Alexa discovery.

XDM525, 2020-09-01:
I am just using the phone app

s-hadinger, 2020-09-01:
As mentioned in the documentation, it only works with an actual Echo device.

XDM525, 2020-09-01:
Yep it works. I tried it using echo dot and it functions properly. Thank you.

closed 2020-09-01 by XDM525