arendst/Tasmota #9206: Add a prometheus build

MorrisJobke, 2020-09-01:
**Have you looked for this feature in other issues and in the docs?**  Yes, there is a directive `USE_PROMETHEUS` available but not used in any current build: #9204 

**Is your feature request related to a problem? Please describe.**  
Enabling the USE_PROMETHEUS build flag on either an existing or a new build would help to use the `/metrics` endpoint and lower the barrier. Compiling the new firmware on each update is more work for each of the users if the builds can be done automatically here in the project itself.

**Describe the solution you'd like**  
Enabling the `USE_PROMETHEUS` in a build that suits this. Either the `tasmota` one or a dedicated `tasmota-prometheus` flavor.

**Describe alternatives you've considered**  
Building the release by myself - but this is a higher barrier also for others and joining forces helps maybe.

**Additional context**  
n/a

Thanks for this nice project 🙌

ascillato, 2020-09-01:
Hi,

Prometheus uses few FLASH and RAM, so I don't see that will worth to add a new flavour with just one more feature. So, a new Tasmota Firmware type seems not to be added.

For adding the directive USE_PROMETHEUS to the full **_Tasmota_** firmware type OR to the **_Tasmota Sensors_** firmware type let's see how many people will need this.

Adding the label FEATURE REQUEST to this issue and let's see how many hits it have.

Thanks for sharing your ideas

blakadder, 2020-09-02:
you can submit a new build to https://github.com/tasmota/binaries/tree/circleci-build and it will be automatically built daily

ascillato2, 2020-09-13:
Closing this request as it had no hits from the community and as the flash space is very restricted on the esp8266, it will be used for other features. Sorry. 

Thanks for sharing your ideas.

closed 2020-09-13 by ascillato2

MarkusTeufelberger, 2022-07-03:
https://github.com/tasmota/binaries/tree/circleci-build 404s for me, looks like the repository moved to https://github.com/tasmota/install? There's no circleci-build branch there though, I'm not sure where to add an unofficial build there to get a full firmware image with just `USE_PROMETHEUS` added to the list.

ascillato, 2022-07-03:
Hi, this is a very old issue. Please, check Tasmota's docs 
(https://tasmota.github.io/docs) for full explanation on how to make custom builds. The docs contains lots of examples.

Thanks.

MarkusTeufelberger, 2022-07-08:
Thanks a lot for your help and patience, @Jason2866 was extremely helpful and added `USE_PROMETHEUS` to the unofficial `allsensors` build. I successfully managed to flash and use it on my power plugs. :-)

In case someone else finds this issue in the future: Flash to the `minimal` image first and use it to flash the unofficial `allsensors` build.