arendst/Tasmota #9211: Value assigment to Variable not correct

neubachera, 2020-09-01:
Dear Tasmota Team,

I'm using Tasmota 8.4 George, on a Wemos D1 to count S0 pulses, generated by a energy counter.
The counters are connected via GIPO 0 and GPIO2 and linked via following template:
(note: I'm aware about the restrictions of [usable GPIOs pins](https://tasmota.github.io/docs/Peripherals/#usable-pins)  )

    USER_TEMPLATE "{\"NAME\":\"E_Counter\",\"GPIO\":
    [43,255,42,255,255,255,255,255,255,255,255,255,255],\"FLAG\":15,\"BASE\":18}"

To calulate the Power during the tele Period of 60 sec I'm using the follwing rule:

    Rule1 ON tele-COUNTER#C1 DO Backlog var1=%value%; var2=(var1-mem1)*60; mem1=var1; event  sendpower1 ENDON
	       ON event#sendpower1 DO publish tele/tasmota_ecounter/power1 %var2% ENDON
               ON tele-COUNTER#C2 DO Backlog var3=%value%; var4=(var3-mem2)*60; mem2=var3; event sendpower2 ENDON
	               ON event#sendpower2 DO publish tele/tasmota_ecounter/power2 %var4% ENDON 

While the calculation of "power1" and the assignment of C1 to var1 is alwasy correct, the calculation of "power2" delivers strange values and respectively provides incorrect calculations/results.

One Issue I noticed in the Log is an incorrect assignment of the CounterValue C2 to Var3:

While:
`    21:53:25 MQT: tele/tasmota_ecounter/SENSOR = {"Time":"2020-09-01T21:53:25","COUNTER 
    {"C1":16299195,"C2":35910507}}`

Reports a value of: "C2":35910507 and

`    21:53:25 RUL: TELE-COUNTER#C2 performs "Backlog var3=35910507; var4=(var3-mem2)*60; mem2=var3; event sendpower2"`

indicates a correct assigment of C2 to var3,
the final assignment in:

`21:53:26 CMD: Gruppe 0, Index 3, Befehl "VAR", Daten "=35910507"
21:53:26 MQT: stat/tasmota_ecounter/RESULT = {"Var3":"35910508.000"}
21:53:26 SRC: Backlog
21:53:26 CMD: Gruppe 0, Index 4, Befehl "VAR", Daten "=(var3-mem2)*60"
21:53:26 MQT: stat/tasmota_ecounter/RESULT = {"Var4":"0.000"}
21:53:26 SRC: Backlog
21:53:26 CMD: Gruppe 0, Index 2, Befehl "MEM", Daten "=var3"
21:53:26 MQT: stat/tasmota_ecounter/RESULT = {"Mem2":"35910508.000"}`

suddently reports a value of 35910508.000 ???

Just for completness: C2 doesn't got any impulses during night time, but during the day, once pluses are delivered, a similar behaviour appears, i.e. sometimes the correct values are assinged sometimes not.

Is this a HW problem, any bug or did I made any mistake? 

Any advice?

BR
Andreas

PS: for completeness the complete log over a minute.

	21:52:36 WIF: Prüfe Verbindung...
	21:52:56 WIF: Prüfe Verbindung...
	21:53:16 WIF: Prüfe Verbindung...
	21:53:25 MQT: tele/tasmota_ecounter/STATE = {"Time":"2020-09-01T21:53:25","Uptime":"0T01:11:14","UptimeSec":4274,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"XXXX","BSSId":"BB:BB:BB:BB:BB:BB","Channel":3,"RSSI":100,"Signal":-30,"LinkCount":1,"Downtime":"0T00:00:08"}}
	21:53:25 MQT: tele/tasmota_ecounter/SENSOR = {"Time":"2020-09-01T21:53:25","COUNTER":{"C1":16299195,"C2":35910507}}
	21:53:25 RUL: TELE-COUNTER#C1 performs "Backlog var1=16299195; var2=(var1-mem1)*60; mem1=var1; event sendpower1"
	21:53:25 SRC: Rule
	21:53:25 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "var1=16299195; var2=(var1-mem1)*60; mem1=var1; event sendpower1"
	21:53:25 RUL: TELE-COUNTER#C2 performs "Backlog var3=35910507; var4=(var3-mem2)*60; mem2=var3; event sendpower2"
	21:53:25 SRC: Rule
	21:53:25 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "var3=35910507; var4=(var3-mem2)*60; mem2=var3; event sendpower2"
	21:53:25 SRC: Backlog
	21:53:25 CMD: Gruppe 0, Index 1, Befehl "VAR", Daten "=16299195"
	21:53:25 MQT: stat/tasmota_ecounter/RESULT = {"Var1":"16299195.000"}
	21:53:25 SRC: Backlog
	21:53:25 CMD: Gruppe 0, Index 2, Befehl "VAR", Daten "=(var1-mem1)*60"
	21:53:25 MQT: stat/tasmota_ecounter/RESULT = {"Var2":"420.000"}
	21:53:26 SRC: Backlog
	21:53:26 CMD: Gruppe 0, Index 1, Befehl "MEM", Daten "=var1"
	21:53:26 MQT: stat/tasmota_ecounter/RESULT = {"Mem1":"16299195.000"}
	21:53:26 SRC: Backlog
	21:53:26 CMD: Gruppe 0, Index 1, Befehl "EVENT", Daten "sendpower1"
	21:53:26 MQT: stat/tasmota_ecounter/RESULT = {"Event":"Done"}
	21:53:26 RUL: EVENT#SENDPOWER1 performs "publish tele/tasmota_ecounter/power1 420.000"
	21:53:26 SRC: Rule
	21:53:26 CMD: Gruppe 0, Index 1, Befehl "PUBLISH", Daten "tele/tasmota_ecounter/power1 420.000"
	21:53:26 MQT: tele/tasmota_ecounter/power1 = 420.000
	21:53:26 SRC: Backlog
	21:53:26 CMD: Gruppe 0, Index 3, Befehl "VAR", Daten "=35910507"
	21:53:26 MQT: stat/tasmota_ecounter/RESULT = {"Var3":"35910508.000"}
	21:53:26 SRC: Backlog
	21:53:26 CMD: Gruppe 0, Index 4, Befehl "VAR", Daten "=(var3-mem2)*60"
	21:53:26 MQT: stat/tasmota_ecounter/RESULT = {"Var4":"0.000"}
	21:53:26 SRC: Backlog
	21:53:26 CMD: Gruppe 0, Index 2, Befehl "MEM", Daten "=var3"
	21:53:26 MQT: stat/tasmota_ecounter/RESULT = {"Mem2":"35910508.000"}
	21:53:27 SRC: Backlog
	21:53:27 CMD: Gruppe 0, Index 1, Befehl "EVENT", Daten "sendpower2"
	21:53:27 MQT: stat/tasmota_ecounter/RESULT = {"Event":"Done"}
	21:53:27 RUL: EVENT#SENDPOWER2 performs "publish tele/tasmota_ecounter/power2 0.000"
	21:53:27 SRC: Rule
	21:53:27 CMD: Gruppe 0, Index 1, Befehl "PUBLISH", Daten "tele/tasmota_ecounter/power2 0.000"
	21:53:27 MQT: tele/tasmota_ecounter/power2 = 0.000
	21:53:27 CFG: in Flash gespeichert am F9, zählen 2059, Bytes 4096
	21:53:36 WIF: Prüfe Verbindung...
	21:53:56 WIF: Prüfe Verbindung...
	21:54:16 WIF: Prüfe Verbindung...
	21:54:25 MQT: tele/tasmota_ecounter/STATE = {"Time":"2020-09-01T21:54:25","Uptime":"0T01:12:14","UptimeSec":4334,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"XXXX","BSSId":"BB:BB:BB:BB:BB:BB","Channel":3,"RSSI":100,"Signal":-30,"LinkCount":1,"Downtime":"0T00:00:08"}}
	21:54:25 MQT: tele/tasmota_ecounter/SENSOR = {"Time":"2020-09-01T21:54:25","COUNTER":{"C1":16299201,"C2":35910507}}
	21:54:25 RUL: TELE-COUNTER#C1 performs "Backlog var1=16299201; var2=(var1-mem1)*60; mem1=var1; event sendpower1"
	21:54:25 SRC: Rule
	21:54:25 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "var1=16299201; var2=(var1-mem1)*60; mem1=var1; event sendpower1"
	21:54:25 RUL: TELE-COUNTER#C2 performs "Backlog var3=35910507; var4=(var3-mem2)*60; mem2=var3; event sendpower2"
	21:54:25 SRC: Rule
	21:54:25 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "var3=35910507; var4=(var3-mem2)*60; mem2=var3; event sendpower2"
	21:54:25 SRC: Backlog
	21:54:25 CMD: Gruppe 0, Index 1, Befehl "VAR", Daten "=16299201"
	21:54:25 MQT: stat/tasmota_ecounter/RESULT = {"Var1":"16299201.000"}
	21:54:25 SRC: Backlog
	21:54:25 CMD: Gruppe 0, Index 2, Befehl "VAR", Daten "=(var1-mem1)*60"
	21:54:25 MQT: stat/tasmota_ecounter/RESULT = {"Var2":"360.000"}
	21:54:26 SRC: Backlog
	21:54:26 CMD: Gruppe 0, Index 1, Befehl "MEM", Daten "=var1"
	21:54:26 MQT: stat/tasmota_ecounter/RESULT = {"Mem1":"16299201.000"}
	21:54:26 SRC: Backlog
	21:54:26 CMD: Gruppe 0, Index 1, Befehl "EVENT", Daten "sendpower1"
	21:54:26 MQT: stat/tasmota_ecounter/RESULT = {"Event":"Done"}
	21:54:26 RUL: EVENT#SENDPOWER1 performs "publish tele/tasmota_ecounter/power1 360.000"
	21:54:26 SRC: Rule
	21:54:26 CMD: Gruppe 0, Index 1, Befehl "PUBLISH", Daten "tele/tasmota_ecounter/power1 360.000"
	21:54:26 MQT: tele/tasmota_ecounter/power1 = 360.000
	21:54:26 SRC: Backlog
	21:54:26 CMD: Gruppe 0, Index 3, Befehl "VAR", Daten "=35910507"
	21:54:26 MQT: stat/tasmota_ecounter/RESULT = {"Var3":"35910508.000"}
	21:54:26 SRC: Backlog
	21:54:26 CMD: Gruppe 0, Index 4, Befehl "VAR", Daten "=(var3-mem2)*60"
	21:54:26 MQT: stat/tasmota_ecounter/RESULT = {"Var4":"0.000"}
	21:54:27 SRC: Backlog
	21:54:27 CMD: Gruppe 0, Index 2, Befehl "MEM", Daten "=var3"
	21:54:27 MQT: stat/tasmota_ecounter/RESULT = {"Mem2":"35910508.000"}
	21:54:27 SRC: Backlog
	21:54:27 CMD: Gruppe 0, Index 1, Befehl "EVENT", Daten "sendpower2"
	21:54:27 MQT: stat/tasmota_ecounter/RESULT = {"Event":"Done"}
	21:54:27 RUL: EVENT#SENDPOWER2 performs "publish tele/tasmota_ecounter/power2 0.000"
	21:54:27 SRC: Rule
	21:54:27 CMD: Gruppe 0, Index 1, Befehl "PUBLISH", Daten "tele/tasmota_ecounter/power2 0.000"
	21:54:27 MQT: tele/tasmota_ecounter/power2 = 0.000
	21:54:27 CFG: in Flash gespeichert am F8, zählen 2060, Bytes 4096
	21:54:36 WIF: Prüfe Verbindung...

closed 2020-09-01 by close-issue-app[bot]

lalo-uy, 2020-09-02:
Did you try to put Counter2 stuff on Rule2 ?

neubachera, 2020-09-03:
Yes I tried as well, however I already got the reason/solution please refere to: [https://github.com/arendst/Tasmota/issues/9212](url)