arendst/Tasmota #9212: Wrong assingment of CouterValue to variable

neubachera, 2020-09-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Dear Tasmota Team,

I'm using Tasmota 8.4 George, on a Wemos D1 to count S0 pulses, generated by a energy counter.
The counters are connected via GIPO 0 and GPIO2 and linked via following template:
(note: I'm aware about the restrictions of [usable GPIOs pins](https://tasmota.github.io/docs/Peripherals/#usable-pins)  )

    USER_TEMPLATE "{\"NAME\":\"E_Counter\",\"GPIO\":
    [43,255,42,255,255,255,255,255,255,255,255,255,255],\"FLAG\":15,\"BASE\":18}"

To calulate the Power during the tele Period of 60 sec I'm using the follwing rule:

    Rule1 ON tele-COUNTER#C1 DO Backlog var1=%value%; var2=(var1-mem1)*60; mem1=var1; event  sendpower1 ENDON
	       ON event#sendpower1 DO publish tele/tasmota_ecounter/power1 %var2% ENDON
               ON tele-COUNTER#C2 DO Backlog var3=%value%; var4=(var3-mem2)*60; mem2=var3; event sendpower2 ENDON
	               ON event#sendpower2 DO publish tele/tasmota_ecounter/power2 %var4% ENDON 

While the calculation of "power1" and the assignment of C1 to var1 is alwasy correct, the calculation of "power2" delivers strange values and respectively provides incorrect calculations/results.

One Issue I noticed in the Log is an incorrect assignment of the CounterValue C2 to Var3:

While:
`    21:53:25 MQT: tele/tasmota_ecounter/SENSOR = {"Time":"2020-09-01T21:53:25","COUNTER 
    {"C1":16299195,"C2":35910507}}`

Reports a value of: "C2":35910507 and

`    21:53:25 RUL: TELE-COUNTER#C2 performs "Backlog var3=35910507; var4=(var3-mem2)*60; mem2=var3; event sendpower2"`

indicates a correct assigment of C2 to var3,
the final assignment in:

`21:53:26 CMD: Gruppe 0, Index 3, Befehl "VAR", Daten "=35910507"
21:53:26 MQT: stat/tasmota_ecounter/RESULT = {"Var3":"35910508.000"}
21:53:26 SRC: Backlog
21:53:26 CMD: Gruppe 0, Index 4, Befehl "VAR", Daten "=(var3-mem2)*60"
21:53:26 MQT: stat/tasmota_ecounter/RESULT = {"Var4":"0.000"}
21:53:26 SRC: Backlog
21:53:26 CMD: Gruppe 0, Index 2, Befehl "MEM", Daten "=var3"
21:53:26 MQT: stat/tasmota_ecounter/RESULT = {"Mem2":"35910508.000"}`

suddently reports a value of 35910508.000 ???

Just for completness: C2 doesn't got any impulses during night time, but during the day, once pluses are delivered, a similar behaviour appears, i.e. sometimes the correct values are assinged sometimes not.

Is this a HW problem, any bug or did I made any mistake? 

Any advice?

BR
Andreas

PS: for completeness the complete log over a minute is inserted below.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [ x] Tasmota binary firmware version number used: 8.4.0
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [x ] IDE / Compiler used: VSCode
- [ ] Flashing tools used: _____
[... 139 lines cut ...]
	21:54:27 MQT: stat/tasmota_ecounter/RESULT = {"Mem2":"35910508.000"}
	21:54:27 SRC: Backlog
	21:54:27 CMD: Gruppe 0, Index 1, Befehl "EVENT", Daten "sendpower2"
	21:54:27 MQT: stat/tasmota_ecounter/RESULT = {"Event":"Done"}
	21:54:27 RUL: EVENT#SENDPOWER2 performs "publish tele/tasmota_ecounter/power2 0.000"
	21:54:27 SRC: Rule
	21:54:27 CMD: Gruppe 0, Index 1, Befehl "PUBLISH", Daten "tele/tasmota_ecounter/power2 0.000"
	21:54:27 MQT: tele/tasmota_ecounter/power2 = 0.000
	21:54:27 CFG: in Flash gespeichert am F8, zählen 2060, Bytes 4096
	21:54:36 WIF: Prüfe Verbindung...`
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-09-01:
The most probable cause is that the math in rules is done using single precision floating point, meaning approx 7 decimal digits. Values needing more than that will "drift" to the nearest supportable value. In addition to "just" reducing current mem/counter values to fewer digits, one possible option could be to use `SetOption79` to reset counters at every `TelePeriod`. Then you'd not need to calculate the difference every minute, and you could also avoid the significant flash wear of updating mem variables every minute, providing better lifetime for your device.

neubachera, 2020-09-02:
Thanks for the quick reply. Indeed this was the problem.
The restriction is a bit unfortune, since I want to preserve the countervalue on the same state as the physical counter. However it seems I need to find a workaround. 
Thanks you
Andreas

closed 2020-09-02 by neubachera