arendst/Tasmota #9215: Can't get Tasmota 8.4.0 to work with Alexa in Hue bridge mode

dominoxx, 2020-09-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I wanted to create my own WIFI 3 channel relay device using an esp8266 module. I had downloaded the precompiled Tasmota bin file (version 8.4.0 by Theo Arends) and flashed it to my esp8266 module. I then opened it up in the web browser, set the module configuration to Generic (18) and then selected the Generic (18) template. I was able to control the relays through the web browser but now I want to control it through the Alexa (Generatin 1). Next I opened up "Configure Other" and selected the Hue Bridge emulation, however in the Alexa app, no devices are discovered. Finger is discovering the MCU as "Amazon-Echo-HA-Bridge" I then tried it using the WeMo single device emulation and one new device was discovered and could be controlled through Alexa. Under "Configure Other" at the top I selected Activate by the template parameters and still no luck. What can I do to get it work with Hue Bridge emulation?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP8266 ESP-12F 4M Bytes 
- [x] Tasmota binary firmware version number used: 8.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tasmotizer-1.1a.exe
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
11:19:50 CMD: Backlog Template; Module; GPIO 255
11:19:50 RSL: RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,29,30,255,255,31,255,255,255,255],"FLAG":15,"BASE":18}
11:19:50 RSL: RESULT = {"Module":{"0":"Generic"}}
11:19:51 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"29":"Relay1i"},"GPIO5":{"30":"Relay2i"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"31":"Relay3i"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
11:23:01 CMD: Status 0
11:23:01 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota_ausfahren","Tasmota_stop","Tasmota_einfahren"],"Topic":"tasmota_65E5D6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:23:01 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T18:58:29","StartupUTC":"2020-09-01T15:24:32","Sleep":50,"CfgHolder":4617,"BootCount":10,"BCResetTime":"2020-09-01T15:59:28","SaveCount":24,"SaveAddress":"F8000"}}
11:23:01 RSL: STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-07-29T12:05:28","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"392/699"}}
11:23:01 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UPC21BFEAA",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["0000C001","2805C8000100060000005A00000000000000","00000000","00006000"]}}
11:23:01 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":592,"Free":408,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
11:23:01 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_65E5D6-1494","IPAddress":"192.168.0.243","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"80.69.96.12","Mac":"48:3F:DA:65:E5:D6","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:23:01 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-09-02T10:23:01","Local":"2020-09-02T11:23:01","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:09","Sunset":"19:29"}}
11:23:01 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-09-02T11:23:01"}}
11:23:01 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-09-02T11:23:01","Uptime":"0T18:58:29","UptimeSec":68309,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"UPC21BFEAA","BSSId":"DC:53:7C:C6:23:1D","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-09-02:
The Hue emulation only works on an actual echo device, not in Alexa app as explained in the Tasmota docs.

Check the docs and old issues for more information.

closed 2020-09-02 by ascillato2

dominoxx, 2020-09-02:
I am using an Alexa Gen1, not only software

s-hadinger, 2020-09-02:
You need echo device gen 2 minimum for Hue emulation to work.