arendst/Tasmota #9216: Sonoff Zigbee bridge is not pairing correctly with Sonoff switch

zuhair68, 2020-09-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I flashed Sonoff Zigbee bridge with `Tasmota 8.4.0.3` and `ncp-uart-sw_6.7.6_115200.ota`
I'm not sure if there is a problem when pairing Sonoff devices or I'm doing something wrong. The same problem is happening with Sonoff motion sensor.

I paired [Sonoff Zigbee switch](https://www.itead.cc/smart-home/sonoff-snzb-01-zigbee-wireless-switch.html) with the following steps:

**Command**
`ZbPermitJoin 1`

**Result**
```
tele/tasmota_DC0B35/3330/SENSOR = {"ZbReceived":{"0x3330":{"Device":"0x3330","Manufacturer":"eWeLink","ModelId":"WB01","Endpoint":1,"LinkQuality":150}}}
```

**Command**
```
ZbBind {"Device":"0x3330","ToDevice":"0x0000","Endpoint":1,"ToEndpoint":1,"Cluster":6}
```

**Result**
```
tele/tasmota_DC0B35/RESULT = {"ZbBind":{"Device":"0x3330","Status":0,"StatusMessage":"SUCCESS"}}
```

**Command**
```
ZbSend {"Device":"0x3330","Config":{"Power":{"MinInterval":1,"MaxInterval":600}}}
```

**Result**
```
ZIG: ZbZCLRawReceived: {"0x3330":{"ConfigResponse":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"}}}
tele/tasmota_DC0B35/3330/SENSOR = {"ZbReceived":{"0x3330":{"Device":"0x3330","ConfigResponse":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"},"Endpoint":1,"LinkQuality":129}}}
```

**Command**
```
ZbSend {"Device":"0x3330","ReadConfig":{"Power":true}}
```

**Result**
```
ZIG: ZbZCLRawReceived: {"0x3330":{"ReadConfig":{"0006/1000":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"}}}}
tele/tasmota_DC0B35/3330/SENSOR = {"ZbReceived":{"0x3330":{"Device":"0x3330","ReadConfig":{"0006/1000":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"}},"Endpoint":1,"LinkQuality":139}}}
```

**The switch is paired and appeared in the main menu, the problem is when toggling the switch the power is always off, these are the messages I get on the console on every toggle:**

```
ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x3330","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":124,"securityuse":0,"seqnumber":19,"fc":"0x18","manuf":"0x0000","transact":19,"cmdid":"0x01","payload":"0000001000"}}
ZIG: ZbZCLRawReceived: {"0x3330":{"0006/0000":0}}
RSL: tele/tasmota_DC0B35/3330/SENSOR = {"ZbReceived":{"0x3330":{"Device":"0x3330","Power":0,"Endpoint":1,"LinkQuality":124}}}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge / Sonoff Zigbee switch
- [X ] Tasmota binary firmware version number used: Tasmota 8.4.0.3
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tasmotizer
[... 73 lines cut ...]
11:19:43 ZIG: {"ZbEZSPReceived":"3F0000303304010600010140010000B9010000"}
11:19:43 ZIG: {"ZbEZSPReceived2":"763D9001450000040106000101000100001CFFDD3033FFFF08181601000000100002"}
11:19:43 ZIG: ZbEZSPSentRaw 80
11:19:43 ZIG: {"ZbEZSPReceived":"450000040106000101000100001CFFDD3033FFFF08181601000000100002"}
11:19:43 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x3330","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":137,"securityuse":0,"seqnumber":28,"fc":"0x18","manuf":"0x0000","transact":22,"cmdid":"0x01","payload":"0000001000"}}
11:19:43 ZIG: ZbZCLRawReceived: {"0x3330":{"0006/0000":0}}
11:19:44 RSL: tele/tasmota_DC0B35/3330/SENSOR = {"ZbReceived":{"0x3330":{"Device":"0x3330","Power":0,"Endpoint":1,"LinkQuality":137}}}
11:20:01 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
To turn the switch ON/OFF on every toggle.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Moe-HA-Maker, 2020-09-03:
I had the same problem, it seems that Zigbee2Tasmota is not compatible yet with Sonoff bridge. Use Zigbee2MQTT, it's much straightforward and working like charm. You may also try to flash it with ncp-uart-sw_6.5.5_115200.ota and use Home Assistance.

zuhair68, 2020-09-04:
I think this is a bug in the firmware

s-hadinger, 2020-09-05:
Hi, I'm confused with the logs. When you say that pressing the button generates the following logs:
```
ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x3330","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":124,"securityuse":0,"seqnumber":19,"fc":"0x18","manuf":"0x0000","transact":19,"cmdid":"0x01","payload":"0000001000"}}
ZIG: ZbZCLRawReceived: {"0x3330":{"0006/0000":0}}
RSL: tele/tasmota_DC0B35/3330/SENSOR = {"ZbReceived":{"0x3330":{"Device":"0x3330","Power":0,"Endpoint":1,"LinkQuality":124}}}
```

I suspect that you are missing some other logs above. This logs shows a command `0x01` which is a respond to a read command. Normally a device should never spontaneously send this respond without a corresponding request.

I suspect that the Sonoff device sent a command right before, and Tasmota is automatically requesting the device for its latest status - which I suppose is this message.

You should see above some Power commands. According to Zigbee2MQTT, here are the correspondance:
- 0 = long press
- 1 = double press
- 2 = single press

Can you please confirm? If so I can add the translation to the right type of button press.

zuhair68, 2020-09-05:
Below is the complete response in Tasmota console for a single press:

```
17:54:13 MQT: tele/tasmota_DC0CBC/STATE = {"Time":"2020-09-05T17:54:13","Uptime":"0T00:05:09","UptimeSec":309,"Vcc":3.523,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"MQTT_IoT","BSSId":"F6:9F:C2:2A:23:E9","Channel":6,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}
17:55:07 ZIG: {"ZbEZSPReceived":"4500000401060001010001000046A0C4E2ECFFFF03011D02"}
17:55:07 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":71,"securityuse":0,"seqnumber":70,"fc":"0x01","manuf":"0x0000","transact":29,"cmdid":"0x02","payload":""}}
17:55:07 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006!02":"","Power":2}}
17:55:07 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","0006!02":"","Power":2,"Endpoint":1,"LinkQuality":71}}}
17:55:07 ZIG: ZbEZSPSend 340000E2EC040106000101400100000501050005000000
17:55:08 ZIG: {"ZbEZSPReceived":"34000032"}
17:55:08 ZIG: {"ZbEZSPReceived":"3F0000E2EC0401060001014001000032010000"}
17:55:08 ZIG: {"ZbEZSPReceived":"45000004010600010100010000479CC3E2ECFFFF081805010000001000"}
17:55:08 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":68,"securityuse":0,"seqnumber":71,"fc":"0x18","manuf":"0x0000","transact":5,"cmdid":"0x01","payload":"0000001000"}}
17:55:08 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006/0000":0}}
17:55:08 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","Power":0,"Endpoint":1,"LinkQuality":68}}}
```
On line 5, I can see the correct response:

- 0 = long press
- 1 = double press
- 2 = single press

The last line, which is the last tele message the broker receives, Power is always 0.

This is what the broker receives in the subscribed topic `tele/tasmota_DC0CBC/ECE2/SENSOR`

```
18:07:48 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","0006!02":"","Power":2,"Endpoint":1,"LinkQuality":84}}}
18:07:48 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","Power":0,"Endpoint":1,"LinkQuality":97}}}
```

For example, in Homebridge, the response will be:

```
18:07:48 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","Power":0,"Endpoint":1,"LinkQuality":97}}}
```

This is the console response for the double press:

```
18:14:28 ZIG: {"ZbEZSPReceived":"450000040106000101000100005FA4C5E2ECFFFF03012601"}
18:14:28 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":74,"securityuse":0,"seqnumber":95,"fc":"0x01","manuf":"0x0000","transact":38,"cmdid":"0x01","payload":""}}
18:14:28 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006!01":"","Power":1}}
18:14:28 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":74}}}
18:14:28 ZIG: ZbEZSPSend 340000E2EC040106000101400100000D0105000D000000
18:14:28 ZIG: {"ZbEZSPReceived":"3400003A"}
18:14:28 ZIG: {"ZbEZSPReceived":"3F0000E2EC040106000101400100003A010000"}
18:14:28 ZIG: {"ZbEZSPReceived":"4500000401060001010001000060D0D0E2ECFFFF08180D010000001000"}
18:14:28 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":103,"securityuse":0,"seqnumber":96,"fc":"0x18","manuf":"0x0000","transact":13,"cmdid":"0x01","payload":"0000001000"}}
18:14:28 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006/0000":0}}
18:14:29 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","Power":0,"Endpoint":1,"LinkQuality":103}}}
```

Long Press

```
18:15:13 ZIG: {"ZbEZSPReceived":"4500000401060001010001000062B4C9E2ECFFFF03012700"}
18:15:13 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":84,"securityuse":0,"seqnumber":98,"fc":"0x01","manuf":"0x0000","transact":39,"cmdid":"0x00","payload":""}}
18:15:13 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006!00":"","Power":0}}
18:15:13 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":84}}}
18:15:14 ZIG: ZbEZSPSend 340000E2EC040106000101400100000E0105000E000000
18:15:14 ZIG: {"ZbEZSPReceived":"3400003B"}
18:15:14 ZIG: {"ZbEZSPReceived":"3F0000E2EC040106000101400100003B010000"}
18:15:14 ZIG: {"ZbEZSPReceived":"4500000401060001010001000063B8CAE2ECFFFF08180E010000001000"}
18:15:14 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":87,"securityuse":0,"seqnumber":99,"fc":"0x18","manuf":"0x0000","transact":14,"cmdid":"0x01","payload":"0000001000"}}
18:15:14 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006/0000":0}}
18:15:14 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","Power":0,"Endpoint":1,"LinkQuality":87}}}
```

The last TELE response is always Power:0

s-hadinger, 2020-09-06:
The latest PR doesn't request the switch state anymore.

zuhair68, 2020-09-06:
I might be missing something, but this is the last message received by the broker for a single, double, or long press. Any automation done in Homebridge/Homekit based on the "Power" is not working because "Power" is always 0. Any MQTT plug-in in Homebridge will read "Power": 0

```
{
	"ZbReceived": {
		"0xECE2": {
			"Device": "0xECE2",
			"Power": 0,
			"Endpoint": 1,
			"LinkQuality": 131
		}
	}
} 
```

s-hadinger, 2020-09-08:
Please provide full logs. With v8.5 you shouldn't receive such responses, but I need `weblog 3` logs to double check...

zuhair68, 2020-09-08:
It's already set on `weblog 3` . Do you want the log after the switch press? or the complete log after a reboot?

zuhair68, 2020-09-08:
This is the full log after a restart, the last three for a single, double, and long press.

```
19:03:03 ZIG: Zigbee devices data in Flash (65 bytes)
19:03:19 ZIG: {"ZbEZSPReceived":"45000500001F0000000000000000FF000000FFFF0A81019FC58A1F004B1200"}
19:03:19 MQT: tele/tasmota_DC0CBC/RESULT = {"ZbParent":{"Device":"0x0000","Children":1,"ChildInfo":["0x00124B001F8AC59F"]}}
19:03:20 ZIG: {"ZbEZSPReceived":"3F0006FCFF00001F0000000000000000000000"}
19:07:57 MQT: tele/tasmota_DC0CBC/STATE = {"Time":"2020-09-08T19:07:57","Uptime":"0T00:05:09","UptimeSec":309,"Vcc":3.523,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"MQTT_IoT","BSSId":"F6:9F:C2:2A:23:E9","Channel":6,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:09:47 HTP: Main Menu
19:11:40 HTP: Main Menu
19:11:40 HTP: Main Menu
19:11:42 HTP: Console
19:11:46 CMD: weblog
19:11:46 SRC: WebConsole from 192.168.1.237
19:11:46 CMD: Group 0, Index 1, Command "WEBLOG", Data ""
19:11:46 MQT: stat/tasmota_DC0CBC/RESULT = {"WebLog":3}
19:12:57 MQT: tele/tasmota_DC0CBC/STATE = {"Time":"2020-09-08T19:12:57","Uptime":"0T00:10:09","UptimeSec":609,"Vcc":3.514,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"MQTT_IoT","BSSId":"F6:9F:C2:2A:23:E9","Channel":6,"RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:14:22 ZIG: {"ZbEZSPReceived":"45000004010600010100010000C1F4D9E2ECFFFF03015F02"}
19:14:22 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":126,"securityuse":0,"seqnumber":193,"fc":"0x01","manuf":"0x0000","transact":95,"cmdid":"0x02","payload":""}}
19:14:22 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006!02":"","Power":2}}
19:14:22 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","0006!02":"","Power":2,"Endpoint":1,"LinkQuality":126}}}
19:14:23 ZIG: ZbEZSPSend 340000E2EC040106000101400100000101050001000000
19:14:23 ZIG: {"ZbEZSPReceived":"34000005"}
19:14:23 ZIG: {"ZbEZSPReceived":"3F0000E2EC0401060001014001000005010000"}
19:14:23 ZIG: {"ZbEZSPReceived":"45000004010600010100010000C2F4D9E2ECFFFF081801010000001000"}
19:14:23 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":126,"securityuse":0,"seqnumber":194,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"0000001000"}}
19:14:23 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006/0000":0}}
19:14:23 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","Power":0,"Endpoint":1,"LinkQuality":126}}}
19:14:36 HTP: Main Menu
19:14:37 HTP: Console
19:14:41 ZIG: {"ZbEZSPReceived":"45000004010600010100010000C4ECD7E2ECFFFF03016001"}
19:14:41 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":121,"securityuse":0,"seqnumber":196,"fc":"0x01","manuf":"0x0000","transact":96,"cmdid":"0x01","payload":""}}
19:14:41 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006!01":"","Power":1}}
19:14:41 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":121}}}
19:14:42 ZIG: ZbEZSPSend 340000E2EC040106000101400100000201050002000000
19:14:42 ZIG: {"ZbEZSPReceived":"34000006"}
19:14:42 ZIG: {"ZbEZSPReceived":"3F0000E2EC0401060001014001000006010000"}
19:14:42 ZIG: {"ZbEZSPReceived":"45000004010600010100010000C5FFDCE2ECFFFF081802010000001000"}
19:14:42 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":134,"securityuse":0,"seqnumber":197,"fc":"0x18","manuf":"0x0000","transact":2,"cmdid":"0x01","payload":"0000001000"}}
19:14:42 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006/0000":0}}
19:14:42 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","Power":0,"Endpoint":1,"LinkQuality":134}}}
19:14:51 HTP: Main Menu
19:14:52 HTP: Console
19:14:56 ZIG: {"ZbEZSPReceived":"45000004010600010100010000C7F0D8E2ECFFFF03016100"}
19:14:56 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":124,"securityuse":0,"seqnumber":199,"fc":"0x01","manuf":"0x0000","transact":97,"cmdid":"0x00","payload":""}}
19:14:56 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006!00":"","Power":0}}
19:14:56 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":124}}}
19:14:56 ZIG: ZbEZSPSend 340000E2EC040106000101400100000301050003000000
19:14:56 ZIG: {"ZbEZSPReceived":"34000007"}
19:14:56 ZIG: {"ZbEZSPReceived":"3F0000E2EC0401060001014001000007010000"}
19:14:56 ZIG: {"ZbEZSPReceived":"45000004010600010100010000C8F0D8E2ECFFFF081803010000001000"}
19:14:56 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":124,"securityuse":0,"seqnumber":200,"fc":"0x18","manuf":"0x0000","transact":3,"cmdid":"0x01","payload":"0000001000"}}
19:14:56 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006/0000":0}}
19:14:57 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","Power":0,"Endpoint":1,"LinkQuality":124}}}
```

zuhair68, 2020-09-08:
As you are releasing 8.5 can I be greedy and ask for more?! 

- Sonoff motion sensor SNZB-03 is not working with release 8.4.0.3
- Sonoff temperature sensor SNZB-02 loses the connection wit the coordinator after the restarting the Sonoff Zigbee bridge, I have to re-bind it after each restart. 

Thanks,

s-hadinger, 2020-09-08:
@zuhair68 I tested your the messages from the logs with v8.5, I don't see requests any more, nor "Power":0. Are you sure you updated to v8.5?

What do you mean with 'SNZB-03 is not working with release 8.4.0.3'?

I have a SNZB-02 and it does continue to report temp/humidity after a restart of ZBBridge without re-pairing. What are the symptoms?

zuhair68, 2020-09-08:
Sorry, it's my mistake. I thought 8.5 will be released. Now I'm on 8.5.

The switch is working perfectly even after restarting the coordinator it will re-pair. This is the log for a single, double, and long press.

```
21:09:33 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","0006!01":"","Power":1,"Endpoint":1,"LinkQuality":95}}}
21:09:38 ZIG: {"ZbEZSPReceived":"45000004010600010100010000DBE8D6E2ECFFFF03016B02"}
21:09:38 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":118,"securityuse":0,"seqnumber":219,"fc":"0x01","manuf":"0x0000","transact":107,"cmdid":"0x02","payload":""}}
21:09:38 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006!02":"","Power":2,"Endpoint":1,"LinkQuality":118}}
21:09:38 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","0006!02":"","Power":2,"Endpoint":1,"LinkQuality":118}}}
21:09:49 ZIG: {"ZbEZSPReceived":"45000004010600010100010000DDE0D4E2ECFFFF03016C00"}
21:09:49 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xECE2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":113,"securityuse":0,"seqnumber":221,"fc":"0x01","manuf":"0x0000","transact":108,"cmdid":"0x00","payload":""}}
21:09:49 ZIG: ZbZCLRawReceived: {"0xECE2":{"0006!00":"","Power":0,"Endpoint":1,"LinkQuality":113}}
21:09:49 MQT: tele/tasmota_DC0CBC/ECE2/SENSOR = {"ZbReceived":{"0xECE2":{"Device":"0xECE2","0006!00":"","Power":0,"Endpoint":1,"LinkQuality":113}}}
```

This is how I bind the Temp sensor, it loses the connection with the coordinator after the restart. I have to long-press the button to wake it up.

```
ZbBind {"Device":"0x6BE0","ToDevice":"0x0000","Cluster":"Temperature"}
ZbSend {"Device":"0x6BE0","Config":{"Temperature":{"MinInterval":60,"MaxInterval":600,"ReportableChange":1}}}
ZbBind {"Device":"0x6BE0","Cluster":"Humidity"}
ZbSend {"Device":"0x6BE0","Config":{"Humidity":{"MinInterval":60,"MaxInterval":600,"ReportableChange":5}}}
ZbBind {"Device":"0x6BE0","Cluster":"BatteryVoltage"}
ZbSend {"Device":"0x6BE0","Config":{"BatteryVoltage":{"MinInterval":60,"MaxInterval":600,"ReportableChange":1}}}
```

This is the log for binding the Sonoff motion sensor SNZB-03. There is no feedback from the device when there is a motion. I tried two devices (same model). 

Error: "UNSUPPORTED_ATTRIBUTE".

```
21:03:23 CMD: ZbPermitJoin 1
21:03:23 SRC: WebConsole from 192.168.1.237
21:03:23 CMD: Group 0, Index 1, Command "ZBPERMITJOIN", Data "1"
21:03:23 ZIG: ZbEZSPSend 22003C
21:03:23 ZIG: ZbEZSPSend 3600FCFF00003600000000000000031E0103033C01
21:03:23 MQT: stat/tasmota_DC0CBC/RESULT = {"ZbPermitJoin":"Done"}
21:03:23 ZIG: {"ZbEZSPReceived":"220000"}
21:03:23 MQT: tele/tasmota_DC0CBC/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
21:03:23 ZIG: {"ZbEZSPReceived":"36000042"}
21:03:23 ZIG: {"ZbEZSPReceived":"4500050000360000000000000042FF000000FFFF03033C01"}
21:03:23 ZIG: Internal ZDO message 0x0036 sent from 0x0000 (broadcast)
21:03:24 ZIG: {"ZbEZSPReceived":"3F0006FCFF0000360000000000000042010000"}
21:03:36 ZIG: {"ZbEZSPReceived":"230002006C2505FF711E004B120004"}
21:03:36 ZIG: {"ZbEZSPReceived":"24006C2505FF711E004B12000203FFFF"}
21:03:38 ZIG: {"ZbEZSPReceived":"230002016C2505FF711E004B120004"}
21:03:38 ZIG: {"ZbEZSPReceived":"24006C2505FF711E004B120001000000"}
21:03:38 MQT: tele/tasmota_DC0CBC/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x00124B001E71FF05","ShortAddr":"0x256C","ParentNetwork":"0x0000","Status":1,"Decision":0}}
21:03:40 ZIG: {"ZbEZSPReceived":"4500040000130000000000000001C4CD6C25FFFF0C006C2505FF711E004B12008002"}
21:03:40 MQT: tele/tasmota_DC0CBC/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x00124B001E71FF05","ShortAddr":"0x256C","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
21:03:40 ZIG: ZbEZSPSend 3400006C2500000500000040010000040103046C25
21:03:40 ZIG: {"ZbEZSPReceived":"34000045"}
21:03:40 ZIG: {"ZbEZSPReceived":"4500000401000501FF0001000002B8CA6C25FFFF070901010D00000002"}
21:03:40 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0x256C","srcendpoint":1,"dstendpoint":255,"wasbroadcast":0,"LinkQuality":87,"securityuse":0,"seqnumber":2,"fc":"0x09","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"0D000000"}}
21:03:40 ZIG: ZbZCLRawReceived: {"0x256C":{"0500<01":"0D000000","Endpoint":1,"LinkQuality":87}}
21:03:40 MQT: tele/tasmota_DC0CBC/256C/SENSOR = {"ZbReceived":{"0x256C":{"Device":"0x256C","0500<01":"0D000000","Endpoint":1,"LinkQuality":87}}}
21:03:40 ZIG: {"ZbEZSPReceived":"4500000401000501FF0001000003B8CA6C25FFFF0919020001000000000002"}
21:03:40 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0x256C","srcendpoint":1,"dstendpoint":255,"wasbroadcast":0,"LinkQuality":87,"securityuse":0,"seqnumber":3,"fc":"0x19","manuf":"0x0000","transact":2,"cmdid":"0x00","payload":"010000000000"}}
21:03:40 ZIG: ZbZCLRawReceived: {"0x256C":{"0500<00":"010000000000","ZoneStatusChange":1,"Endpoint":1,"LinkQuality":87}}
21:03:40 MQT: tele/tasmota_DC0CBC/256C/SENSOR = {"ZbReceived":{"0x256C":{"Device":"0x256C","0500<00":"010000000000","ZoneStatusChange":1,"Endpoint":1,"LinkQuality":87}}}
21:03:40 ZIG: {"ZbEZSPReceived":"3F0006FDFF0000130000000000000001000000"}
21:03:40 ZIG: {"ZbEZSPReceived":"3F00006C250000050000004001000045010000"}
21:03:40 ZIG: {"ZbEZSPReceived":"4500000000058000000001000004DCD36C25FFFF0604006C25010102"}
[... cut ...]
21:04:16 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1030,"srcaddr":"0x256C","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":108,"securityuse":0,"seqnumber":8,"fc":"0x18","manuf":"0x0000","transact":4,"cmdid":"0x09","payload":"86000000"}}
21:04:16 ZIG: ZbZCLRawReceived: {"0x256C":{"ReadConfig":{"0406/0000":{"Occupancy":true,"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"}},"Endpoint":1,"LinkQuality":108}}
21:04:16 MQT: tele/tasmota_DC0CBC/256C/SENSOR = {"ZbReceived":{"0x256C":{"Device":"0x256C","ReadConfig":{"0406/0000":{"Occupancy":true,"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"}},"Endpoint":1,"LinkQuality":108}}}
```

s-hadinger, 2020-09-09:
@zuhair68 Actually SNZB-03 does work with Z2T. And it does not need binding.

The useful payload is the following:
```
{"ZbReceived":{"0x256C":{"Device":"0x256C","0500<00":"010000000000","ZoneStatusChange":1,"Endpoint":1,"LinkQuality":87}}}
```

This device does not use the `Occupancy` cluster but instead the `IAS` cluster (Intruder Alarm System).

The attribute you need to look for is `"ZoneStatusChange":1`.
I understand it's not yet very user friendly and I plan to convert it automatically to `Occupancy`. That's in the todo list.

zuhair68, 2020-09-09:
I tried binding the SNZB-03 motion sensor.

**Symptoms:** Not sending any feedback when there is a motion.

Here are the logs:

**Binding**

```
ZbPermitJoin 1
```
**Result**
```
18:05:41 ZIG: {"ZbEZSPReceived":"45000400001300000000000000006CB76C25FFFF0C006C2505FF711E004B12008002"}
18:05:41 MQT: tele/tasmota_DC0CBC/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x00124B001E71FF05","ShortAddr":"0x256C","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
18:05:41 ZIG: ZbEZSPSend 3400006C25000005000000400100000C01030C6C25
18:05:42 ZIG: {"ZbEZSPReceived":"34000098"}
18:05:42 ZIG: {"ZbEZSPReceived":"4500000401000501FF0001000001A0C46C25FFFF070900010D00000002"}
18:05:42 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0x256C","srcendpoint":1,"dstendpoint":255,"wasbroadcast":0,"LinkQuality":71,"securityuse":0,"seqnumber":1,"fc":"0x09","manuf":"0x0000","transact":0,"cmdid":"0x01","payload":"0D000000"}}
18:05:42 ZIG: ZbZCLRawReceived: {"0x256C":{"0500<01":"0D000000","Endpoint":1,"LinkQuality":71}}
18:05:42 MQT: tele/tasmota_DC0CBC/256C/SENSOR = {"ZbReceived":{"0x256C":{"Device":"0x256C","0500<01":"0D000000","Endpoint":1,"LinkQuality":71}}}
18:05:42 ZIG: {"ZbEZSPReceived":"4500000401000501FF0001000002A0C46C25FFFF0919010000000000000002"}
18:05:42 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0x256C","srcendpoint":1,"dstendpoint":255,"wasbroadcast":0,"LinkQuality":71,"securityuse":0,"seqnumber":2,"fc":"0x19","manuf":"0x0000","transact":1,"cmdid":"0x00","payload":"000000000000"}}
18:05:42 ZIG: ZbZCLRawReceived: {"0x256C":{"0500<00":"000000000000","ZoneStatusChange":0,"Endpoint":1,"LinkQuality":71}}
18:05:42 MQT: tele/tasmota_DC0CBC/256C/SENSOR = {"ZbReceived":{"0x256C":{"Device":"0x256C","0500<00":"000000000000","ZoneStatusChange":0,"Endpoint":1,"LinkQuality":71}}}
18:05:42 ZIG: {"ZbEZSPReceived":"3F0006FDFF0000130000000000000000000000"}
18:05:42 ZIG: {"ZbEZSPReceived":"3F00006C250000050000004001000098010000"}
18:05:42 ZIG: {"ZbEZSPReceived":"4500000000058000000001000003A8C66C25FFFF060C006C25010102"}
18:05:42 MQT: tele/tasmota_DC0CBC/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
18:05:42 ZIG: ZbEZSPSend 3400006C250401000001014001000001010700010004000500
18:05:42 ZIG: {"ZbEZSPReceived":"34000099"}
18:05:43 ZIG: {"ZbEZSPReceived":"3F00006C250401000001014001000099010000"}
18:05:43 ZIG: {"ZbEZSPReceived":"4500000401000001010001000004C8CE6C25FFFF1818010104000042076557654C696E6B05000042044D53303102"}
18:05:43 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0x256C","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":97,"securityuse":0,"seqnumber":4,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"04000042076557654C696E6B05000042044D533031"}}
18:05:43 ZIG: ZbZCLRawReceived: {"0x256C":{"0000/0004":"eWeLink","0000/0005":"MS01","Endpoint":1,"LinkQuality":97}}
18:05:43 MQT: tele/tasmota_DC0CBC/256C/SENSOR = {"ZbReceived":{"0x256C":{"Device":"0x256C","Manufacturer":"eWeLink","ModelId":"MS01","Endpoint":1,"LinkQuality":97}}}
18:05:45 ZIG: ZbFlashStore 03206C2505FF711E004B120001010000FFFF4D533031006557654C696E6B0000FF20E2EC9FC58A1F004B120001010000FFFF57423031006557654C696E6B0000FF20E06B4CBF2822004B120001010000FFFF54483031006557654C696E6B0000FF
18:05:45 ZIG: Zigbee Devices Data store in Flash (0x402FF800 - 97 bytes)
18:05:53 ZIG: {"ZbEZSPReceived":"4500000401000501FF0001000005C4CD6C25FFFF0919020001000000000002"}
18:05:53 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0x256C","srcendpoint":1,"dstendpoint":255,"wasbroadcast":0,"LinkQuality":95,"securityuse":0,"seqnumber":5,"fc":"0x19","manuf":"0x0000","transact":2,"cmdid":"0x00","payload":"010000000000"}}
18:05:53 ZIG: ZbZCLRawReceived: {"0x256C":{"0500<00":"010000000000","ZoneStatusChange":1,"Endpoint":1,"LinkQuality":95}}
18:05:53 MQT: tele/tasmota_DC0CBC/256C/SENSOR = {"ZbReceived":{"0x256C":{"Device":"0x256C","0500<00":"010000000000","ZoneStatusChange":1,"Endpoint":1,"LinkQuality":95}}}
```

**Command**
```
Zbbind {"Device":"0x256C","Cluster":"IAS"}
```
**Result:** 

```
18:06:19 CMD: Zbbind {"Device":"0x256C","Cluster":"IAS"}
18:06:19 SRC: WebConsole from 192.168.1.237
18:06:19 CMD: Group 0, Index 1, Command "ZBBIND", Data "{"Device":"0x256C","Cluster":"IAS"}"
18:06:19 ZIG: ZbEZSPSend 3400006C25000021000000400100000D01160D05FF711E004B120001000003AA9D1DFEFF23A46001
18:06:19 MQT: stat/tasmota_DC0CBC/RESULT = {"ZbBind":"Done"}
18:06:19 ZIG: {"ZbEZSPReceived":"3400009A"}
18:06:19 ZIG: {"ZbEZSPReceived":"3F00006C25000021000000400100009A010000"}
18:06:19 ZIG: {"ZbEZSPReceived":"4500000000218000000001000006C4CD6C25FFFF020D0002"}
18:06:19 MQT: tele/tasmota_DC0CBC/RESULT = {"ZbBind":{"Device":"0x256C","Status":0,"StatusMessage":"SUCCESS"}}
```

**Command:**
```
ZbSend {"Device":"0x256C","Config":{"IAS":{"MinInterval":1,"MaxInterval":600}}}
```
**Result:** 
```
17:48:01 CMD: ZbSend {"Device":"0x256C","Config":{"IAS":{"MinInterval":1,"MaxInterval":600}}}
17:48:01 SRC: WebConsole from 192.168.1.237
[... 6 lines cut ...]
```
ZbSend {"Device":"0x256C","ReadConfig":{"IAS":true}}
```
**Result:** 
```
18:06:49 CMD: ZbSend {"Device":"0x256C","ReadConfig":{"IAS":true}}
18:06:49 SRC: WebConsole from 192.168.1.237
18:06:49 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0x256C","ReadConfig":{"IAS":true}}"
18:06:49 ZIG: guessing endpoint 1
18:06:49 ZIG: Unknown attribute name (ignored): IAS
18:06:49 MQT: stat/tasmota_DC0CBC/RESULT = {"ZbSend":"Missing parameters"}
```

**Command**
```
ZbBindState 0x256C
```
**Result:** 
```
18:03:16 MQT: stat/tasmota_DC0CBC/RESULT = {"ZbBindState":"Done"}
18:03:16 ZIG: {"ZbEZSPReceived":"34000094"}
18:03:16 ZIG: {"ZbEZSPReceived":"3F00006C250000330000004001000094010000"}
18:03:16 ZIG: {"ZbEZSPReceived":"4500000000338000000001000006CCCF6C25FFFF050A8B00000002"}
18:03:16 MQT: tele/tasmota_DC0CBC/RESULT = {"ZbBindState":{"Device":"0x256C","Status":139,"StatusMessage":"NOT_FOUND","BindingsTotal":0,"BindingsStart":1,"Bindings":[]}}
```

s-hadinger, 2020-09-11:
Hi, `IAS` is not a valid attribute name. You should use `ZoneStatus` instead.

```
Zbbind {"Device":"0x256C","Cluster":"ZoneStatus"}
ZbSend {"Device":"0x256C","Config":{"ZoneStatus":{"MinInterval":1,"MaxInterval":600}}}
```

Now I have some good news, the 'auto-binding' feature is on track for a release this week-end. After this release, you will not need to do these commands anymore, they will be automatically done at pairing time.

zuhair68, 2020-09-11:
I'm still getting **UNSUPPORTED_ATTRIBUTE** error

```
Zbbind {"Device":"0xB87A","Cluster":"ZoneStatus"}
```
**Result**
```
22:18:43 MQT: stat/tasmota_DC0CBC/RESULT = {"ZbBind":"Done"}
22:18:43 ZIG: {"ZbEZSPReceived":"34000093"}
22:18:43 ZIG: {"ZbEZSPReceived":"3F00007AB80000210000004001000093010000"}
22:18:44 ZIG: {"ZbEZSPReceived":"4500000000218000000001000006C8CE7AB8FFFF02050002"}
22:18:44 MQT: tele/tasmota_DC0CBC/RESULT = {"ZbBind":{"Device":"0xB87A","Status":0,"StatusMessage":"SUCCESS"}}
```
```
ZbSend {"Device":"0xB87A","Config":{"ZoneStatus":{"MinInterval":1,"MaxInterval":600}}}
```
**Result**
```
22:18:54 CMD: ZbSend {"Device":"0xB87A","Config":{"ZoneStatus":{"MinInterval":1,"MaxInterval":600}}}
22:18:54 SRC: WebConsole from 192.168.1.237
22:18:54 CMD: Group 0, Index 1, Command "ZBSEND", Data "{"Device":"0xB87A","Config":{"ZoneStatus":{"MinInterval":1,"MaxInterval":600}}}"
22:18:54 ZIG: guessing endpoint 1
22:18:54 ZIG: ZbEZSPSend 3400007AB80401000501014001000003010B1003060002001901005802
22:18:54 MQT: stat/tasmota_DC0CBC/RESULT = {"ZbSend":"Done"}
22:18:54 ZIG: {"ZbEZSPReceived":"34000094"}
22:18:55 ZIG: {"ZbEZSPReceived":"3F00007AB80401000501014001000094010000"}
22:18:55 ZIG: {"ZbEZSPReceived":"4500000401000501010001000007D0D07AB8FFFF071803078600020002"}
22:18:55 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0xB87A","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":103,"securityuse":0,"seqnumber":7,"fc":"0x18","manuf":"0x0000","transact":3,"cmdid":"0x07","payload":"86000200"}}
22:18:55 ZIG: ZbZCLRawReceived: {"0xB87A":{"ConfigResponse":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"},"Endpoint":1,"LinkQuality":103}}
22:18:55 MQT: tele/tasmota_DC0CBC/B87A/SENSOR = {"ZbReceived":{"0xB87A":{"Device":"0xB87A","ConfigResponse":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"},"Endpoint":1,"LinkQuality":103}}}
```

s-hadinger, 2020-09-11:
This should not be a problem. The binding is maybe enough.

zuhair68, 2020-09-11:
The binding will give a success response, but no feedback from the device when there is a motion.

s-hadinger, 2020-09-12:
Please try again with the latest development branch. Re-pair the device with the new version.

zuhair68, 2020-09-12:
I reflashed the bridge to start fresh with 8.5.0.1

I'm still getting `UNSUPPORTED_ATTRIBUTE` error. See line `12:30:49 ZIG:` nothing shows in the console from the sensor after line `12:34:58 RSL:`

Symptoms: the sensor is not sending any feedback when motion occurs. 

```
ZbPermitJoin 1
```

```
12:30:21 CMD: ZbPermitJoin 1
12:30:21 SRC: WebConsole from 192.168.1.237
12:30:21 CMD: Group 0, Index 1, Command "ZBPERMITJOIN", Data "1"
12:30:21 ZIG: ZbEZSPSend 22003C
12:30:21 ZIG: ZbEZSPSend 3600FCFF00003600000000000000011E0103013C01
12:30:21 RSL: stat/tasmota_DC0CBC/RESULT = {"ZbPermitJoin":"Done"}
12:30:21 ZIG: {"ZbEZSPReceived":"220000"}
12:30:21 RSL: tele/tasmota_DC0CBC/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
12:30:21 ZIG: {"ZbEZSPReceived":"360000A3"}
12:30:21 ZIG: {"ZbEZSPReceived":"45000500003600000000000000A3FF000000FFFF03013C01"}
12:30:21 ZIG: Internal ZDO message 0x0036 sent from 0x0000 (broadcast)
12:30:22 ZIG: {"ZbEZSPReceived":"230000017AB805FF711E004B120004"}
12:30:22 ZIG: {"ZbEZSPReceived":"3F0006FCFF00003600000000000000A3010000"}
12:30:23 ZIG: {"ZbEZSPReceived":"24007AB805FF711E004B120001000000"}
12:30:23 RSL: tele/tasmota_DC0CBC/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x00124B001E71FF05","ShortAddr":"0xB87A","ParentNetwork":"0x0000","Status":1,"Decision":0}}
12:30:23 ZIG: {"ZbEZSPReceived":"45000500001F0000000000000000FF000000FFFF0A810105FF711E004B1200"}
12:30:23 RSL: tele/tasmota_DC0CBC/RESULT = {"ZbParent":{"Device":"0x0000","Children":1,"ChildInfo":["0x00124B001E71FF05"]}}
12:30:24 ZIG: {"ZbEZSPReceived":"3F0006FCFF00001F0000000000000000010000"}
12:30:25 ZIG: ZbFlashStore 01107AB805FF711E004B120000000000FF
12:30:25 ZIG: Zigbee Devices Data store in Flash (0x402FF800 - 17 bytes)
12:30:32 ZIG: {"ZbEZSPReceived":"230000007AB805FF711E004B120004"}
12:30:32 ZIG: {"ZbEZSPReceived":"24007AB805FF711E004B12000203FFFF"}
12:30:32 ZIG: {"ZbEZSPReceived":"24007AB805FF711E004B12000203FFFF"}
12:30:32 ZIG: {"ZbEZSPReceived":"24007AB805FF711E004B12000203FFFF"}
12:30:35 ZIG: {"ZbEZSPReceived":"230000017AB805FF711E004B120004"}
12:30:35 ZIG: {"ZbEZSPReceived":"24007AB805FF711E004B120001000000"}
12:30:35 RSL: tele/tasmota_DC0CBC/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x00124B001E71FF05","ShortAddr":"0xB87A","ParentNetwork":"0x0000","Status":1,"Decision":0}}
12:30:36 ZIG: {"ZbEZSPReceived":"4500040000130000000000000000F8DA7AB8FFFF0C007AB805FF711E004B12008002"}
12:30:36 RSL: tele/tasmota_DC0CBC/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x00124B001E71FF05","ShortAddr":"0xB87A","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
12:30:36 ZIG: ZbEZSPSend 3400007AB800000500000040010000020103027AB8
12:30:36 ZIG: {"ZbEZSPReceived":"340000A8"}
12:30:36 ZIG: {"ZbEZSPReceived":"4500000401000501FF0001000001F8DA7AB8FFFF070900010D00000002"}
12:30:36 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0xB87A","srcendpoint":1,"dstendpoint":255,"wasbroadcast":0,"LinkQuality":129,"securityuse":0,"seqnumber":1,"fc":"0x09","manuf":"0x0000","transact":0,"cmdid":"0x01","payload":"0D000000"}}
12:30:36 ZIG: ZbZCLRawReceived: {"0xB87A":{"0500<01":"0D000000","Endpoint":1,"LinkQuality":129}}
12:30:36 RSL: tele/tasmota_DC0CBC/B87A/SENSOR = {"ZbReceived":{"0xB87A":{"Device":"0xB87A","0500<01":"0D000000","Endpoint":1,"LinkQuality":129}}}
12:30:36 ZIG: {"ZbEZSPReceived":"4500000401000501FF0001000002FCDB7AB8FFFF0919010000000000000002"}
12:30:36 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":1280,"srcaddr":"0xB87A","srcendpoint":1,"dstendpoint":255,"wasbroadcast":0,"LinkQuality":131,"securityuse":0,"seqnumber":2,"fc":"0x19","manuf":"0x0000","transact":1,"cmdid":"0x00","payload":"000000000000"}}
12:30:36 ZIG: ZbZCLRawReceived: {"0xB87A":{"0500<00":"000000000000","ZoneStatusChange":0,"0406/0000":0,"Endpoint":1,"LinkQuality":131}}
12:30:36 RSL: tele/tasmota_DC0CBC/B87A/SENSOR = {"ZbReceived":{"0xB87A":{"Device":"0xB87A","0500<00":"000000000000","ZoneStatusChange":0,"Occupancy":0,"Endpoint":1,"LinkQuality":131}}}
12:30:36 ZIG: {"ZbEZSPReceived":"3F0006FDFF0000130000000000000000000000"}
12:30:36 ZIG: {"ZbEZSPReceived":"3F00007AB800000500000040010000A8010000"}
12:30:36 ZIG: {"ZbEZSPReceived":"4500000000058000000001000003F4D97AB8FFFF0602007AB8010102"}
12:30:36 RSL: tele/tasmota_DC0CBC/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
12:30:36 ZIG: ZbEZSPSend 3400007AB80401000001014001000001010700010004000500
12:30:36 ZIG: {"ZbEZSPReceived":"340000A9"}
12:30:37 ZIG: {"ZbEZSPReceived":"3F00007AB804010000010140010000A9010000"}
12:30:37 ZIG: {"ZbEZSPReceived":"4500000401000001010001000004F0D87AB8FFFF1818010104000042076557654C696E6B05000042044D53303102"}
12:30:37 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0xB87A","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":124,"securityuse":0,"seqnumber":4,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"04000042076557654C696E6B05000042044D533031"}}
12:30:37 ZIG: ZbZCLRawReceived: {"0xB87A":{"0000/0004":"eWeLink","0000/0005":"MS01","Endpoint":1,"LinkQuality":124}}
12:30:37 RSL: tele/tasmota_DC0CBC/B87A/SENSOR = {"ZbReceived":{"0xB87A":{"Device":"0xB87A","Manufacturer":"eWeLink","ModelId":"MS01","Endpoint":1,"LinkQuality":124}}}
12:30:38 ZIG: ZbEZSPSend 3400007AB800000400000040010000030104037AB801
12:30:38 ZIG: {"ZbEZSPReceived":"340000AA"}
12:30:38 ZIG: {"ZbEZSPReceived":"3F00007AB800000400000040010000AA010000"}
[... cut ...]
12:34:58 RSL: tele/tasmota_DC0CBC/STATE = {"Time":"2020-09-12T12:34:58","Uptime":"0T00:05:09","UptimeSec":309,"Vcc":3.523,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"MQTT_IoT","BSSId":"F6:9F:C2:2A:23:E9","Channel":6,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}
12:39:58 RSL: tele/tasmota_DC0CBC/STATE = {"Time":"2020-09-12T12:39:58","Uptime":"0T00:10:09","UptimeSec":609,"Vcc":3.523,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"MQTT_IoT","BSSId":"F6:9F:C2:2A:23:E9","Channel":6,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}
```

s-hadinger, 2020-09-12:
@zuhair68 Thanks for the logs, this is useful to me since I don't have this device.

I need to see why I'm not getting BatteryLevels either.

From the logs of SNZB-03, I extracted the following:
```
12:30:36 RSL: tele/tasmota_DC0CBC/B87A/SENSOR = {"ZbReceived":{"0xB87A":{"Device":"0xB87A","0500<01":"0D000000","Endpoint":1,"LinkQuality":129}}}
12:30:36 RSL: tele/tasmota_DC0CBC/B87A/SENSOR = {"ZbReceived":{"0xB87A":{"Device":"0xB87A","0500<00":"000000000000","ZoneStatusChange":0,"Occupancy":0,"Endpoint":1,"LinkQuality":131}}}
2:31:11 RSL: tele/tasmota_DC0CBC/B87A/SENSOR = {"ZbReceived":{"0xB87A":{"Device":"0xB87A","0500<00":"010000000000","ZoneStatusChange":1,"Occupancy":1,"Endpoint":1,"LinkQuality":139}}}
```

The first line is an enrollment request, I don't think we need to enroll the device in a zone.
On all other lines, Z2T now automatically converts `"ZoneStatusChange":1` to `"Occupancy":1`, same for 0.

The `UNSUPPORTED_ATTRIBUTE` error will remain, but has no negative effect. It's just in case the device uses attribute reporting instead of Zone Status change notification.

I'm not sure I can change the speed at which the device sends Occupancy data. I will check.

zuhair68, 2020-09-12:
Shouldn't the motion sensor send an immediate notification message when it senses a motion? Please don't hesitate to ask me for more testing or logs.

s-hadinger, 2020-09-13:
There’s nothing we can do to shorten this delay. What delay do you observe ?

coldfire84, 2020-09-13:
> There’s nothing we can do to shorten this delay. What delay do you observe ?

Also have a Sonoff Zigbee Bridge and SNZB-03, working as expected on 8.5.0.1: instant/ immediate notification on motion detected. In my testing to date, the shortest time to a no motion/ occupancy event, following a motion event, is 60 seconds. This delay is of course longer if motion continues.

zuhair68, 2020-09-13:
It's not a delay, the sensor is not sending any notification if there is a motion. 
For example, if you program an action in the home automation based on motion it won't fire up because the sensor is not sending any messages back to the bridge.

zuhair68, 2020-09-13:
> > There’s nothing we can do to shorten this delay. What delay do you observe ?
> 
> 
> 
> May be unrelated...
> 
> 
> 
> Also have a Sonoff Zigbee Bridge and SNZB-03, working as expected on 8.5.0.1: instant/ immediate notification on motion detected. In my testing to date, the shortest time to a no motion/ occupancy event, following a motion event, is 60 seconds. This delay is of course longer if motion continues.

That's good to know! Can you please share how did you pair the motion sensor? I have the same sensor and it is not sending any notification when motion occurred. I tried two devices with the same problem. 
Sonoff temperature sensor and Sonoff switch are working flawlessly after the last release.

coldfire84, 2020-09-14:
> Can you please share how did you pair the motion sensor?

Honestly, it was as simple as running the ```ZbPermitJoin 1``` command on the bridge and then holding the pairing button for ~5 seconds on the sensor itself. With an earlier firmware version I struggled to get it to pair, but re-pairing on latest development branch Tasmota ZBBridge firmware (via http://ota.tasmota.com/tasmota/) solved that.

There is a niggle around there being no battery information displayed in the Tasmota UI (and thus that I can consume for notification on low battery), but the device otherwise functions as expected.

s-hadinger, 2020-09-14:
I'm still struggling to have the device report its battery status. I believe I'm doing exactly the same configuration as Zigbee2MQTT.

Can anyone confirm that Sonoff Zigbee devices report battery with Zigbee2MQTT?

I will work on a work-around with Rules. The devices still respond to an explicit read of the Battery attributes.

coldfire84, 2020-09-14:
> Can anyone confirm that Sonoff Zigbee devices report battery with Zigbee2MQTT?

Not tested with Zigbee2MQTT, but this would suggest it is indeed supported: https://github.com/Koenkk/zigbee2mqtt/issues/4079#issuecomment-672221389

zuhair68, 2020-09-14:
The battery status is working with the switch.

```
ZbSend {"Device":"0xECE2","Config":{"BatteryPercentage":{"MinInterval":60,"MaxInterval":600,"ReportableChange":1}}}
```

[image]

s-hadinger, 2020-09-14:
I tried a rule triggered by data received by the device and it works. I'll craft a work around rule.

zuhair68, 2020-09-14:
I'm still struggling to get the SNZB-03 motion sensor to send instant notifications when motion occurs, it's successfully paired with LQI 100.
I was suspecting that something wrong in the flashing process, I did a fresh firmware flash with no luck. 
(Switch)SNZB-01 and (Temp sensor)SNZB-02 are working perfectly.

zuhair68, 2020-09-14:
The battery percentage is also working with motion sensor

[image]

coldfire84, 2020-09-15:
> The battery percentage is also working with motion sensor

@zuhair68 did you have to do anything to get this working?

zuhair68, 2020-09-15:
> > The battery percentage is also working with motion sensor
> 
> @zuhair68 did you have to do anything to get this working?

@coldfire84 Yes, I used this command while it was pairing:

For the switch SNZB-01 and the motion sensor SNZB-03, it didn't work with the temperature sensor SNZB-03
```
ZbSend {"Device":"0xECE2","Config":{"BatteryPercentage":{"MinInterval":60,"MaxInterval":600,"ReportableChange":1}}}
```

coldfire84, 2020-09-15:
Thanks @zuhair68 , I noticed that in the earlier comments, but get a weird error message when I try and run the command against the sensor in question.

Sensor info:
```
14:41:33 CMD: ZbStatus2
14:41:33 MQT: stat/zbbridge-1/RESULT = {"ZbStatus2":[{"Device":"0xF011","Name":"DS_Bathroom_Motion","IEEEAddr":"0x00124B001F90D19D","ModelId":"MS01","Manufacturer":"eWeLink","Endpoints":["0x01"]}]}
```

Command as listed:
```
14:41:46 CMD: ZbSend {"Device":"0xF011","Config":{"BatteryPercentage":{"MinInterval":60,"MaxInterval":600,"ReportableChange":1}}}
14:41:46 MQT: stat/zbbridge-1/RESULT = {"ZbSend":"Done"}
14:41:53 ZIG: {"ZbEZSPReceived":"80004211F0"}
14:41:53 MQT: tele/zbbridge-1/RESULT = {"ZbRouteError":{"ShortAddr":"0xF011","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
14:42:03 ZIG: {"ZbEZSPReceived":"80004211F0"}
14:42:03 MQT: tele/zbbridge-1/RESULT = {"ZbRouteError":{"ShortAddr":"0xF011","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
14:42:12 ZIG: {"ZbEZSPReceived":"80004211F0"}
14:42:12 MQT: tele/zbbridge-1/RESULT = {"ZbRouteError":{"ShortAddr":"0xF011","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
```

The sensor is working, as in I get the immediate notification on movement.

coldfire84, 2020-09-15:
> The error MAC_INDIRECT_TIMEOUT means that the coordinator couldn't reach the device. You need to wake up the device when sending these commands.

Thanks, re-tried sending the command after trigger a motion event and still got the timeout. So figured I'd re-pair the device. I updated the bridge yesterday and could of sworn I re-paired the sensor, but here is the output from the pairing (you can see the zbsend command in the output):
```
14:58:54 CMD: ZbPermitJoin 1
14:58:54 MQT: stat/zbbridge-1/RESULT = {"ZbPermitJoin":"Done"}
14:58:54 MQT: tele/zbbridge-1/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
14:58:54 ZIG: {"ZbEZSPReceived":"4500050000360000000000000047FF000000FFFF03013C01"}
14:58:54 ZIG: Internal ZDO message 0x0036 sent from 0x0000 (broadcast)
14:59:02 ZIG: {"ZbEZSPReceived":"2300000011F09DD1901F004B120004"}
14:59:02 ZIG: {"ZbEZSPReceived":"240011F09DD1901F004B12000203FFFF"}
14:59:04 ZIG: {"ZbEZSPReceived":"2300000111F09DD1901F004B120004"}
14:59:05 ZIG: {"ZbEZSPReceived":"240011F09DD1901F004B120001000000"}
14:59:05 MQT: tele/zbbridge-1/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x00124B001F90D19D","ShortAddr":"0xF011","ParentNetwork":"0x0000","Status":1,"Decision":0}}
14:59:05 ZIG: {"ZbEZSPReceived":"45000400001300000000000000017CBB11F0FFFF0C0011F09DD1901F004B12008002"}
14:59:05 MQT: tele/zbbridge-1/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x00124B001F90D19D","ShortAddr":"0xF011","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
14:59:05 ZIG: {"ZbEZSPReceived":"4500000401000501FF000100000278BA11F0FFFF070901010D00000002"}
14:59:05 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"DS_Bathroom_Motion":{"Device":"0xF011","Name":"DS_Bathroom_Motion","0500<01":"0D000000","Endpoint":1,"LinkQuality":45}}}
14:59:06 ZIG: {"ZbEZSPReceived":"4500000401000501FF000100000378BA11F0FFFF0919020001000000000002"}
14:59:06 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"DS_Bathroom_Motion":{"Device":"0xF011","Name":"DS_Bathroom_Motion","0500<00":"010000000000","ZoneStatusChange":1,"Occupancy":1,"Endpoint":1,"LinkQuality":45}}}
14:59:06 ZIG: {"ZbEZSPReceived":"45000000000580000000010000047CBB11F0FFFF06020011F0010102"}
14:59:06 MQT: tele/zbbridge-1/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
14:59:07 ZIG: {"ZbEZSPReceived":"45000004010000010100010000057CBB11F0FFFF1818020104000042076557654C696E6B05000042044D53303102"}
14:59:07 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"DS_Bathroom_Motion":{"Device":"0xF011","Name":"DS_Bathroom_Motion","Manufacturer":"eWeLink","ModelId":"MS01","Endpoint":1,"LinkQuality":47}}}
14:59:08 ZIG: {"ZbEZSPReceived":"450000000004800000000100000678BA11F0FFFF17030011F01201040102040004000003000005010001030002"}
14:59:08 MQT: tele/zbbridge-1/RESULT = {"ZbState":{"Status":33,"Endpoint":"0x01","ProfileId":"0x0104","DeviceId":"0x0402","DeviceVersion":0"InClusters":["0x0000","0x0003","0x0500","0x0001"],"OutClusters":["0x0003"]}}
14:59:11 ZIG: {"ZbEZSPReceived":"45000004010005010100010000077CBB11F0FFFF09180301010000310D0002"}
14:59:11 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"DS_Bathroom_Motion":{"Device":"0xF011","Name":"DS_Bathroom_Motion","ZoneType":13,"Endpoint":1,"LinkQuality":47}}}
14:59:12 ZIG: auto-bind `ZbBind {"Device":"0xF011","Endpoint":1,"Cluster":"0x0001"}`
14:59:13 ZIG: {"ZbEZSPReceived":"45000000002180000000010000087CBB11F0FFFF02040002"}
14:59:13 MQT: tele/zbbridge-1/RESULT = {"ZbBind":{"Device":"0xF011","Name":"DS_Bathroom_Motion","Status":0,"StatusMessage":"SUCCESS"}}
14:59:14 ZIG: auto-bind `ZbBind {"Device":"0xF011","Endpoint":1,"Cluster":"0x0500"}`
14:59:15 ZIG: {"ZbEZSPReceived":"45000000002180000000010000097CBB11F0FFFF02050002"}
14:59:15 MQT: tele/zbbridge-1/RESULT = {"ZbBind":{"Device":"0xF011","Name":"DS_Bathroom_Motion","Status":0,"StatusMessage":"SUCCESS"}}
14:59:16 ZIG: auto-bind `ZbSend {"Device":"0xF011","Config":{"BatteryVoltage":{"MinInterval":900,"MaxInterval":900,"ReportableChange":0.1},"BatteryPercentage":{"MinInterval":900,"MaxInterval":900,"ReportableChange":1}}}`
14:59:17 ZIG: {"ZbEZSPReceived":"450000040101000101000100000A7CBB11F0FFFF041804070002"}
14:59:17 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"DS_Bathroom_Motion":{"Device":"0xF011","Name":"DS_Bathroom_Motion","ConfigResponse":{},"Endpoint":1,"LinkQuality":47}}}
14:59:18 ZIG: auto-bind `ZbSend {"Device":"0xF011","Config":{"ZoneStatus":{"MinInterval":1,"MaxInterval":3600}}}`
14:59:19 ZIG: {"ZbEZSPReceived":"450000040100050101000100000B7CBB11F0FFFF071805078600020002"}
14:59:19 MQT: tele/zbbridge-1/SENSOR = {"ZbReceived":{"DS_Bathroom_Motion":{"Device":"0xF011","Name":"DS_Bathroom_Motion","ConfigResponse":{"ZoneStatus":{"Status":134,"StatusMsg":"UNSUPPORTED_ATTRIBUTE"}},"Endpoint":1,"LinkQuality":47}}}
```

Still no battery status :(

BerniApple, 2020-09-17:
> Zigbee2MQTT

how can you

> I had the same problem, it seems that Zigbee2Tasmota is not compatible yet with Sonoff bridge. Use Zigbee2MQTT, it's much straightforward and working like charm. You may also try to flash it with ncp-uart-sw_6.5.5_115200.ota and use Home Assistance.

how can you flash Zigbee2MQTT to sonoff zigbee bridge?

zuhair68, 2020-09-19:
@s-hadinger All devices (SNZB-01, SNZB-02, and SNZB-03) are working perfectly with Sonoff Zigbee bridge! The only thing that is not working is the BatteryPercentage for SNZB-02 (Temp & humidity sensor).

This is the Homebridge conf.json with MQTTTHING plugin.

```
"accessories": [
        {
            "accessory": "mqttthing",
            "type": "statelessProgrammableSwitch",
            "name": "Office Switch",
            "url": "MQTT://192.168.3.100:1883",
            "logMqtt": true,
            "topics": {
                "getSwitch": {
                    "topic": "tele/tasmota_DC0CBC/ECE2/SENSOR",
                    "apply": "return JSON.parse(message).ZbReceived['0xECE2'].Power;"
                },
                "getBatteryLevel": {
                    "topic": "tele/tasmota_DC0CBC/ECE2/SENSOR",
                    "apply": "return JSON.parse(message).ZbReceived['0xECE2'].BatteryPercentage;"
                }
            },
            "switchValues": [
                "2",
                "1",
                "0"
            ]
        },
        {
            "accessory": "mqttthing",
            "type": "occupancySensor",
            "name": "Office Occupancy",
            "url": "MQTT://192.168.3.100:1883",
            "caption": "",
            "topics": {
                "getOccupancyDetected": {
                    "topic": "tele/tasmota_DC0CBC/23E4/SENSOR",
                    "apply": "return JSON.parse(message).ZbReceived['0x23E4'].Occupancy;"
                },
                "getStatusLowBattery": {
                    "topic": "tele/tasmota_DC0CBC/23E4/SENSOR",
                    "apply": "return JSON.parse(message).ZbReceived['0x23E4'].BatteryPercentage;"
                }
            },
            "integerValue": true,
            "logMqtt": true
        },
        {
            "accessory": "mqttthing",
            "type": "temperatureSensor",
            "name": "Office Temperature",
            "url": "MQTT://192.168.3.100:1883",
            "caption": "",
            "topics": {
                "getCurrentTemperature": {
                    "topic": "tele/tasmota_DC0CBC/531D/SENSOR",
                    "apply": "return JSON.parse(message).ZbReceived['0x531D'].Temperature;"
                }
            },
            "logMqtt": true
        },
        {
            "accessory": "mqttthing",
            "type": "humiditySensor",
            "name": "Office Humidity",
            "url": "MQTT://192.168.3.100:1883",
            "caption": "",
            "topics": {
                "getCurrentRelativeHumidity": {
                    "topic": "tele/tasmota_DC0CBC/531D/SENSOR",
                    "apply": "return JSON.parse(message).ZbReceived['0x531D'].Humidity;"
                }
            },
            "logMqtt": true
        }
]
```

BerniApple, 2020-09-20:
@zuhair68 

During a range test my aqara zigbee sensor lost the connection and it didn't  reconnect to the sonoffbridge anymore. i had to pair it again from scratch, are you having the same problem?

s-hadinger, 2020-09-20:
I'm surprised that the device does not reconnect automatically. It should initiate a rejoin. 

Indeed try a new pairing.

coldfire84, 2020-09-22:
@s-hadinger , have tied with the latest development firmware (updated today) and whilst the Tasmota GUI is now showing last event for the sensor I'm still missing the battery status.

[image]

I've tried removing/ re-pairing and get the same results each time - am I missing something (obvious, or not!)?

Thanks!

ascillato2, 2020-09-24:
> All devices (SNZB-01, SNZB-02, and SNZB-03) are working perfectly with Sonoff Zigbee bridge! 

Great! 

>The only thing that is not working is the BatteryPercentage for SNZB-02 (Temp & humidity sensor).

#9332

closed 2020-09-24 by ascillato2