arendst/Tasmota #9217: Missing value on Xiaomi temp/hum sensor

pixelmagic66, 2020-09-02:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

I have some Xiaomi Aqara temperature and humidity sensors (the round ones) en with some of the devices I receive values with the text "Xiaomi_64"  and "Xiaomi_65". I suspect that these are not yet configured in tasmota at the moment. Tasmota version is development 8.4.0.3(zbbridge) with a Sonoff Zigbee bridge version ncp-uart-sw_6.7.6_115200.ota

These are the raw messages:

`19:46:52 ZIG: {"ZbEZSPReceived":"45000004010000010100010000E1C4CD5359FFFF2618480A01FF421F0121D10B0421A81305215E000624020000000064296708652130160A21000002"}
19:46:52 MQT: tasmota/zigbee_bridge/SENSOR = {"ZbReceived":{"0x5953":{"Device":"0x5953","Name":"Temp_Trapkast","BatteryVoltage":3.025,"BatteryPercentage":100,"Voltage":3.025,"Battery":100,"Xiaomi_64":2151,"Xiaomi_65":5680,"Endpoint":1,"LinkQuality":95}}}
`

The values 2151 seems to be the temperature 21.51 Celsius and the 5680 the Humidity of 56.80 percent.

I hope this helps to improve tasmota zigbee further

s-hadinger, 2020-09-02:
Thanks. This is very useful. I will add it a little later.

pixelmagic66, 2020-09-03:
You are very welcome, glad I can help. Will look into other devices.

Thank you for the amazing work so far in such a short time, the software is very stable here and reception seems better then with the original Xiaomi bridge, so the end result is very reliable.

unscel, 2020-09-03:
Same device (WSDCGQ01LM) but the messages seem perfectly fine with me. {"ZbReceived":{"hall_temp_hum":{"Device":"0x8287","Name":"hall_temp_hum","BatteryVoltage":3.025,"BatteryPercentage":100,"Voltage":3.025,"Battery":100,"Temperature":29.93,"Humidity":59.47,"Endpoint":1,"LinkQuality":108}}}

pixelmagic66, 2020-09-03:
@unscel Yes, I have 4 of these devices, 2 work like you say and the other 2 work like in the first post. Maybe they are of some other revision or something ? The 2 new ones are from a recent order, the other two are over 2 years old if I recall correct.

s-hadinger, 2020-09-05:
Thanks, now that the Zigbee refactoring of the internal flow is done, I'm on it.

The problem is that Xiaomi uses the same attribute id 0x64 for different metrics depending on the device. Unfortunately it means that we can't have a universal decoder, and I need to add device specific code.

s-hadinger, 2020-09-05:
@pixelmagic66 Can you please type `ZbStatus2 0x5953` and paste the result?

pixelmagic66, 2020-09-05:
@s-hadinger I have 4 of these sensors at the moment, some more on the way. I made a dump of all of them, hope this helps.

16:07:45 CMD: ZbStatus2 0x5953
16:07:45 MQT: tasmota/zigbee_bridge/RESULT = {"ZbStatus2":[{"Device":"0x5953","Name":"Temp_Trapkast","IEEEAddr":"0x00158D0001FA3C35","ModelId":"lumi.sens","Manufacturer":"LUMI","Endpoints":["0x01","0x02","0x03"]}]}

16:08:27 CMD: ZbStatus2 0x33D0
16:08:27 MQT: tasmota/zigbee_bridge/RESULT = {"ZbStatus2":[{"Device":"0x33D0","Name":"Temp_Kelder","IEEEAddr":"0x00158D0004132D35","ModelId":"lumi.sensor_ht","Manufacturer":"LUMI","Endpoints":["0x01","0x02","0x03"]}]}

16:09:01 CMD: zbstatus2 0x9F02
16:09:01 MQT: tasmota/zigbee_bridge/RESULT = {"ZbStatus2":[{"Device":"0x9F02","Name":"Temp_Badkamer","IEEEAddr":"0x00158D00042502AB","ModelId":"lumi.sens","Manufacturer":"LUMI","Endpoints":["0x01","0x02","0x03"]}]}

16:09:20 CMD: zbstatus2 0x3A56
16:09:20 MQT: tasmota/zigbee_bridge/RESULT = {"ZbStatus2":[{"Device":"0x3A56","Name":"Temp_Werkkamer","IEEEAddr":"0x00158D0001B90C22","ModelId":"lumi.sens","Manufacturer":"LUMI","Endpoints":["0x01","0x02","0x03"]}]}

s-hadinger, 2020-09-05:
Thanks, it should be solved now.  You don't need to re-pair the devices, just upgrade to the latest version in the development branch. Or wait for Tasmota 8.5.

closed 2020-09-05 by s-hadinger

webfrank, 2020-09-07:
Hi,
   I have just upgraded tasmota-zbbridge 8.5.0 but the issue is still present:

16:09:39 MQT: zbgateway/tele/SENSOR = {"ZbReceived":{"0x800A":{"Device":"0x800A","BatteryVoltage":3.13,"BatteryPercentage":100,"Xiaomi_64":2632,"Xiaomi_65":6549,"Endpoint":1,"LinkQuality":42}}}
16:09:39 ZIG: {"ZbEZSPReceived":"450000040100000101000100004574B90A80FFFF1518270A0500420E6C756D692E73656E736F725F6874"}
16:09:39 MQT: zbgateway/tele/SENSOR = {"ZbReceived":{"0x800A":{"Device":"0x800A","ModelId":"lumi.sensor_ht","Endpoint":1,"LinkQuality":42}}}

s-hadinger, 2020-09-07:
Sorry I can't reproduce. I created a fake device with ModelId `lumi.sensor_ht` and it works. If the log was just after pairing this would explain it. The ModelId was received after the first frame. Subsequent frames should decode correctly.

pixelmagic66, 2020-09-07:
I am sorry to report that here also with the 8.5.0 release also no change in the workings:

20:58:43 ZIG: {"ZbEZSPReceived":"4500000401000001010001000077C8CE5359FFFF2618D90A01FF421F0121C70B0421A81305215E00062403000100006429C30865218C160A210000"}
20:58:43 MQT: tasmota/zigbee_bridge/tele/SENSOR = {"ZbReceived":{"0x5953":{"Device":"0x5953","BatteryVoltage":3.01,"BatteryPercentage":100,"Xiaomi_64":2243,"Xiaomi_65":5772,"Endpoint":1,"LinkQuality":97}}}
20:58:45 ZIG: Zigbee Devices Data store in Flash (0x402FF800 - 181 bytes)

21:00:31 ZIG: {"ZbEZSPReceived":"450000040100000101000100004DC0CC4AEFFFFF261C5F11080A01FF421D0121C70B03281C0421A81305216900062402000100000A210000641000"}
21:00:31 MQT: tasmota/zigbee_bridge/tele/SENSOR = {"ZbReceived":{"0xEF4A":{"Device":"0xEF4A","BatteryVoltage":3.01,"BatteryPercentage":100,"Xiaomi_64":0,"Endpoint":1,"LinkQuality":92}}}

pixelmagic66, 2020-09-07:
> 
> 
> Sorry I can't reproduce. I created a fake device with ModelId `lumi.sensor_ht` and it works. If the log was just after pairing this would explain it. The ModelId was received after the first frame. Subsequent frames should decode correctly.

Should that not be lumi.sens ?? That is the ID of the first sensor I reported.

webfrank, 2020-09-08:
Hi, just tested from scratch and the issue is only for the first msg when ModelId is not reported, afterwards it works.

I found a similar issue with Xiaomi Smoke sensor where Xiaomi_64 has been decoded as SmokeDensity but I think it is incorrect, I think it is SecurityZoneChange. Should I open a new issue for this?

pixelmagic66, 2020-09-08:
> Hi, just tested from scratch and the issue is only for the first msg when ModelId is not reported, afterwards it works.
>
> I found a similar issue with Xiaomi Smoke sensor where Xiaomi_64 has been [...]

Sadly that is not the case over here, 12 hours later still giving the Xiaomi_ values here:

08:57:19 ZIG: {"ZbEZSPReceived":"450000040100000101000100005370B8D033FFFF2618310A01FF421F0121A90B0421A81305210800062402000000006429E1076521FB1C0A210000"}
08:57:19 MQT: tasmota/zigbee_bridge/tele/SENSOR = {"ZbReceived":{"0x33D0":{"Device":"0x33D0","BatteryVoltage":2.99,"BatteryPercentage":92,"Xiaomi_64":2017,"Xiaomi_65":7419,"Endpoint":1,"LinkQuality":39}}}

pixelmagic66, 2020-09-08:
> @s-hadinger I have 4 of these sensors at the moment, some more on the way. I made a dump of all of them, hope this helps.
>
> 16:07:45 CMD: ZbStatus2 0x5953
> 16:07:45 MQT: tasmota/zigbee_bridge/RESULT = [...]

I made with firmware 8.5.0 another dump like you requested on sunday and getting some odd results:

09:14:17 CMD: ZbStatus2 0x5953
09:14:17 MQT: tasmota/zigbee_bridge/stat/RESULT = {"ZbStatus2":[{"Device":"0x5953","Name":"Temp_Trapkast","IEEEAddr":"0x0000000000000000","Endpoints":[]}]}

09:14:31 CMD: ZbStatus2 0x33D0
09:14:31 MQT: tasmota/zigbee_bridge/stat/RESULT = {"ZbStatus2":[{"Device":"0x33D0","IEEEAddr":"0x0000000000000000","Endpoints":[]}]}

09:14:45 CMD: zbstatus2 0x9F02
09:14:45 MQT: tasmota/zigbee_bridge/stat/RESULT = {"ZbStatus2":[{"Device":"0x9F02","IEEEAddr":"0x0000000000000000","Endpoints":[]}]}

09:14:57 CMD: zbstatus2 0x3A56
09:14:57 MQT: tasmota/zigbee_bridge/stat/RESULT = {"ZbStatus2":[{"Device":"0x3A56","Name":"Temp_Werkkamer","IEEEAddr":"0x00158D0001B90C22","ModelId":"lumi.sens","Manufacturer":"LUMI","Endpoints":["0x01","0x02","0x03"]}]}

I looks like on some devices the IEEEAddr is not showing correctly ?

s-hadinger, 2020-09-08:
I did the following to show that the bug is fixed:
```
zbezspreceive2 45000004010000010100010000E1C4CD5359FFFF2618480A01FF421F0121D10B0421A81305215E000624020000000064296708652130160A21000002
```
Response is
```
{"Device":"0x5953","BatteryVoltage":3.03,"BatteryPercentage":100,"Xiaomi_64":2151,"Xiaomi_65":5680,"Endpoint":1,"LinkQuality":95}
```

Then I forced the ModelId to `lumi.sens` with `zbmodelid 0x5953,lumi.sens` and the result is now:
```
{"Device":"0x5953","BatteryVoltage":3.03,"BatteryPercentage":100,"Temperature":21.51,"Humidity":56.8,"Endpoint":1,"LinkQuality":95}
```

The problem is fixed. @pixelmagic66 the `Zbstatus2` for 0x5953 shows there is no `modelId` so Tasmota can't know which device it is. Please trigger a re-pairing.

pixelmagic66, 2020-09-08:
@s-hadinger It looks after re-paring you are absolutely right. I seems to be related to a bad OTA flashing yesterday, I had to rewire the FTDI to correctly flash the bridge again. Then when reloading a config backup all seemed ok but it looks like it was not 100%.

Thanks for the effort !

bibikiwi, 2021-06-13:
The same devices intermittently give one format or the other. Strange. (I have the square ones with the slightly rounded corners)

Running on Sonoff Zigbee Bridge : Tasmota 9.4.0(zbbridge) 2021-04-21T12:49:14 2_7_4_9/2.2.2-dev(38a443e)

```
root@console:/var/log/remote# tail -n1000 pvh-zb1.log | grep aqara_salon | tail -n10
Jun 13 08:00:19 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_salon": 
{"Device":"0xCB93","Name":"aqara_salon","Temperature":21.13,"Humidity":51.21,"Pressure":1021,"PressureScale":-1,"PressureScaledValue":10210,"SeaPressure":1021,"Endpoint":1,"LinkQuality":100}}}
Jun 13 08:45:56 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_salon":{"Device":"0xCB93","Name":"aqara_salon","Temperature":21.26,"Humidity":52.42,"Pressure":1021,"PressureScale":-1,"PressureScaledValue":10213,"SeaPressure":1021,"Endpoint":1,"LinkQuality":94}}}
Jun 13 08:47:47 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_salon":{"Device":"0xCB93","Name":"aqara_salon","BatteryVoltage":2.99,"BatteryPercentage":92,"Xiaomi_64":2133,"Xiaomi_65":5256,"Xiaomi_66":102139,"Endpoint":1,"LinkQuality":100}}}
Jun 13 09:26:04 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_salon":{"Device":"0xCB93","Name":"aqara_salon","Temperature":21.79,"Humidity":52.27,"Pressure":1021,"PressureScale":-1,"PressureScaledValue":10215,"SeaPressure":1021,"Endpoint":1,"LinkQuality":94}}}
Jun 13 09:41:26 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_salon":{"Device":"0xCB93","Name":"aqara_salon","BatteryVoltage":2.97,"BatteryPercentage":86,"Xiaomi_64":2192,"Xiaomi_65":5104,"Xiaomi_66":102161,"Endpoint":1,"LinkQuality":94}}}
Jun 13 10:34:06 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_salon":{"Device":"0xCB93","Name":"aqara_salon","Temperature":22.3,"Humidity":50.2,"Pressure":1021,"PressureScale":-1,"PressureScaledValue":10216,"SeaPressure":1021,"Endpoint":1,"LinkQuality":94}}}
Jun 13 10:39:57 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_salon":{"Device":"0xCB93","Name":"aqara_salon","BatteryVoltage":2.97,"BatteryPercentage":86,"Xiaomi_64":2228,"Xiaomi_65":4962,"Xiaomi_66":102169,"Endpoint":1,"LinkQuality":100}}}
Jun 13 11:35:26 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_salon":{"Device":"0xCB93","Name":"aqara_salon","BatteryVoltage":2.97,"BatteryPercentage":86,"Xiaomi_64":2260,"Xiaomi_65":5111,"Xiaomi_66":102155,"Endpoint":1,"LinkQuality":94}}}
Jun 13 12:03:52 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_salon":{"Device":"0xCB93","Name":"aqara_salon","Temperature":22.81,"Humidity":51.59,"Pressure":1021,"PressureScale":-1,"PressureScaledValue":10216,"SeaPressure":1021,"Endpoint":1,"LinkQuality":100}}}
Jun 13 12:29:26 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_salon":{"Device":"0xCB93","Name":"aqara_salon","BatteryVoltage":2.97,"BatteryPercentage":86,"Xiaomi_64":2283,"Xiaomi_65":5240,"Xiaomi_66":102155,"Endpoint":1,"LinkQuality":100}}}
root@console:/var/log/remote# tail -n1000 pvh-zb1.log | grep aqara_master | tail -n10
Jun 13 08:04:02 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_masterbedroom":{"Device":"0xE2B2","Name":"aqara_masterbedroom","Temperature":22.83,"Humidity":54.8,"Pressure":1021,"PressureScale":-1,"PressureScaledValue":10213,"SeaPressure":1021,"Endpoint":1,"LinkQuality":58}}}
Jun 13 08:28:27 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_masterbedroom":{"Device":"0xE2B2","Name":"aqara_masterbedroom","BatteryVoltage":3,"BatteryPercentage":98,"Xiaomi_64":2280,"Xiaomi_65":5492,"Xiaomi_66":102142,"Endpoint":1,"LinkQuality":60}}}
Jun 13 08:50:03 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_masterbedroom":{"Device":"0xE2B2","Name":"aqara_masterbedroom","Temperature":22.83,"Humidity":56.02,"Pressure":1021,"PressureScale":-1,"PressureScaledValue":10216,"SeaPressure":1021,"Endpoint":1,"LinkQuality":73}}}
Jun 13 09:28:11 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_masterbedroom":{"Device":"0xE2B2","Name":"aqara_masterbedroom","BatteryVoltage":3,"BatteryPercentage":98,"Xiaomi_64":2290,"Xiaomi_65":5597,"Xiaomi_66":102177,"Endpoint":1,"LinkQuality":55}}}
Jun 13 09:35:44 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_masterbedroom":{"Device":"0xE2B2","Name":"aqara_masterbedroom","Temperature":22.95,"Humidity":55.92,"Pressure":1021,"PressureScale":-1,"PressureScaledValue":10218,"SeaPressure":1021,"Endpoint":1,"LinkQuality":68}}}
Jun 13 10:23:55 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_masterbedroom":{"Device":"0xE2B2","Name":"aqara_masterbedroom","BatteryVoltage":3,"BatteryPercentage":98,"Xiaomi_64":2317,"Xiaomi_65":5564,"Xiaomi_66":102186,"Endpoint":1,"LinkQuality":47}}}
Jun 13 11:00:54 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_masterbedroom":{"Device":"0xE2B2","Name":"aqara_masterbedroom","Temperature":23.46,"Humidity":55.72,"Pressure":1021,"PressureScale":-1,"PressureScaledValue":10218,"SeaPressure":1021,"Endpoint":1,"LinkQuality":73}}}
[... cut ...]
Jun 13 12:15:41 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_masterbedroom":{"Device":"0xE2B2","Name":"aqara_masterbedroom","BatteryVoltage":3,"BatteryPercentage":98,"Xiaomi_64":2378,"Xiaomi_65":5175,"Xiaomi_66":102172,"Endpoint":1,"LinkQuality":45}}}
Jun 13 12:42:58 pvh-zb1 ESP-MQT: homeassistant/tele/zbbridge_DBFBED/SENSOR = {"ZbReceived":{"aqara_masterbedroom":{"Device":"0xE2B2","Name":"aqara_masterbedroom","Temperature":23.86,"Humidity":52,"Pressure":1021,"PressureScale":-1,"PressureScaledValue":10216,"SeaPressure":1021,"Endpoint":1,"LinkQuality":55}}}
root@console:/var/log/remote# 

```

PR #9240 (2020-09-05): Zigbee better support for WSDCGQ01LM variants
## Description:

Support decoding for Xiaomi specific field for WSDCGQ01LM with `ModelId==lumi.sens.

**Related issue (if applicable):** fixes #9217 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.1
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.m [... cut]