arendst/Tasmota #9218: TasmotaClient Firmware update does not work with esp32 cam

j0seph1387, 2020-09-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): __esp32 cam and nodemcu esp8266___
- [x ] Tasmota binary firmware version number used: ___8.4.0 master  / 8.4.0.3 dev__
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [ x] IDE / Compiler used: __VS Code___
- [x ] Flashing tools used: ___VS Code__
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```

  Configuration output here:

20:17:36 CMD: Backlog Template; Module; GPIO 255
20:17:36 RSL: stat/tasmota_018BD4/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
20:17:36 RSL: stat/tasmota_018BD4/RESULT = {"Module":{"2":"ESP32 Cam AiThinker"}}
20:17:36 RSL: stat/tasmota_018BD4/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"4192":"Client RX"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"4224":"Client RST"},"GPIO15":{"4160":"Client TX"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```

  STATUS 0 output here:

20:19:44 CMD: Status 0
20:19:44 RSL: stat/tasmota_018BD4/STATUS = {"Status":{"Module":2,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_018BD4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:19:44 RSL: stat/tasmota_018BD4/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"RTC Watch dog reset digital core and rtc module","Uptime":"0T00:42:49","StartupUTC":"2020-09-02T18:36:55","Sleep":50,"CfgHolder":4617,"BootCount":39,"BCResetTime":"2020-08-29T20:27:25","SaveCount":138}}
20:19:44 RSL: stat/tasmota_018BD4/STATUS2 = {"StatusFWR":{"Version":"8.4.0(webcam)","BuildDateTime":"2020.09.02 18:29:36","Core":"1_0_4","SDK":"v3.2.3-14-gd3e562907","CpuFrequency":240,"Hardware":"ESP32","CR":"363/699"}}
20:19:44 RSL: stat/tasmota_018BD4/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box 6490 Cable",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000001","00006000"]}}
20:19:44 RSL: stat/tasmota_018BD4/STATUS4 = {"StatusMEM":{"ProgramSize":1196,"Free":1984,"Heap":100,"PsrMax":4095,"PsrFree":3345,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC78F","00140001","00000CCD","010013C0","40010181","80004000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,31,81","Sensors":"1,2,3,5,6,9"}}
20:19:44 RSL: stat/tasmota_018BD4/STATUS5 = {"StatusNET":{"Hostname":"tasmota_018BD4-3028","IPAddress":"192.168.178.53","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"FC:F5:C4:01:8B:D4","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:19:44 RSL: stat/tasmota_018BD4/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_018BD4","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
20:19:44 RSL: stat/tasmota_018BD4/STATUS7 = {"StatusTIM":{"UTC":"2020-09-02T19:19:44","Local":"2020-09-02T20:19:44","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:09","Sunset":"19:29"}}
20:19:44 RSL: stat/tasmota_018BD4/STATUS10 = {"StatusSNS":{"Time":"2020-09-02T20:19:44","TasmotaClient":{"TempI": 24.94, "huI": 45.92, "PrIn": 961.12}}}
20:19:44 RSL: stat/tasmota_018BD4/STATUS11 = {"StatusSTS":{"Time":"2020-09-02T20:19:44","Uptime":"0T00:42:49","UptimeSec":2569,"Heap":100,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":52,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"FRITZ!Box 6490 Cable","BSSId":"C8:0E:14:A2:AC:49","Channel":1,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

20:22:16 CMD: weblog 4
20:22:16 RSL: stat/tasmota_018BD4/RESULT = {"WebLog":4}
20:22:17 CFG: Saved, zählen 139, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-09-02:
Have you enabled `#define USE_TASMOTA_CLIENT`  in _my_user_config.h_ ?

By default, it is disabled.

j0seph1387, 2020-09-02:
hi, thank you very much for the quick response!
of course I used a self-compiled version with #define USE_TASMOTA_CLIENT compiled (VS Code). all other TasmotaClient functions also work without problems! I can use all commands of the TasmotaClient lib without problems with both esp8266 and esp32 cam! CliendSend, no problem and I also receive all replies from my arduino.
the firmware update of the arduino should be recognized automatically via the .hex upload instead of the tasmota.bin.
The firmware upload to the arduino works with the esp8266, with the esp32 cam I get the error message "Magic Byte is not 0xE9".
I hope the google translated English is reasonably understandable :)

arendst, 2020-09-03:
Let me know if this solved your issue.

j0seph1387, 2020-09-03:
Hello big master :) thank you and your helpers for your great work! i have 6 sockets and some other esp's with tasmota running in my network :)
I have to tell you that your changes have unfortunately not fixed the problem but made it a little worse. i downloaded new tasmota dev from github and recompiled them with vs code, flashed an esp8266 and an esp32 cam. both communicate wonderfully with the arduino. the .hex upload works with the esp8266, with the esp32 I unfortunately got the error "Magic Byte is not 0xE9" again on the first attempt, the next 3 attempts ended a few seconds after the upload started with a "The website is not accessible ", the fifth attempt again with" Magic Byte is not 0xE9 ".

j0seph1387, 2020-09-03:
and should I maybe briefly whether the Tasmota client component will be continued at all? I saw that the creator of the TasmotaSlave lib has not yet switched his lib to client.
I also miss the documentation for the "ClientSent" command on the doc page. I was looking for it after I noticed that the "ClientSent RST" command no longer works.

arendst, 2020-09-04:
I'm just wondering which three gpios do you use for the tasmota client connection?

As the esp32 supports way more gpios tasmota does NOT support any serial connection other than console connections to gpio1/3 pair.
Also as you mention specific the esp32-cam you will be running out of usable gpios soon as the webcam software uses a lot of dedicated gpios for both camera, flash light and cf slot.

You might want to try an esp32-devkit first to see if that works with tasmota client. Also because I can imagine the webcam background code could interfere with the serial code.

In the meantime I revisit the upload code.

j0seph1387, 2020-09-04:
Hello, I am using the esp32 cam module.
the full name should be "Ai-Thinker ESP32-CAM Module".
my config is:
GPIO 04: PWM 1 (Flash light)
GPIO 02: Client TX
GPIO 14: Client RST
GPIO 15: Client RX

GPIO 02, 14 and 15 should not be used by the camera.
the communication between the ESP32 and the Arduino pro mini also works without problems, only the .hex upload unfortunately does not work. i receive all my data from the arduino in tasmota and can send all commands with "ClientSend ..." to the arduino, which then also receives and executes them. the webcam stream on the main page works without jerking or interference.
everything does what it should, only the .hex upload not working.

I will test the configuration with rx / tx on GPIO 1/3 tonight

i can order an esp32 devkit and test it but i think it won't help with my problem.

please tell me how I can help you, I'll be happy to send you everything you need, log files, .bin files, ...

thank you very much for your efforts and your work!

arendst, 2020-09-04:
Thx for the info.

I see no need to test gpio1/3 as they are supposed to not work. You already confirmed the current gpios communicate just fine.

I'll focus on the .hex upload malfunction.

arendst, 2020-09-04:
Just tried an Arduino hex upload on a esp32-dev device. See below logging:
```
11:35:57 HTP: Firmware Upgrade
11:36:07 CFG: Saved, Count 66, Bytes 4096
11:36:07 UPL: File BlinkSendCommand.ino.hex ...
11:36:09 UPL: File type 3
11:36:09 UPL: Successful 7966 bytes. Restarting
11:36:09 HTP: Upload done
11:36:22 TCL: Bootloader could not be found
11:36:22 TCL: Flashing aborted!
```
It succeeds in the upload without error message. (File type message is not in current dev release). It fails on TCL: Flashing as I don't have an Arduino connected.

Trying the same on an esp32-cam device with syslog 3 enabled also works fine albeit not all the time. At least I never see your error message regarding wrong file type. 

Are you sure your Arduino hex file looks like this:
```
:100000000C945D000C9485000C9485000C94850084
:100010000C9485000C9485000C9485000C9485004C
:100020000C9485000C9485000C9485000C9485003C
...
```
where it starts with a colon which in turn is being checked by the upload code for .hex validity.

j0seph1387, 2020-09-04:
the start of the .hex:

:100000000C94E6000C940E010C940E010C940E015D
:100010000C940E010C940E010C940E010C940E0124
:100020000C940E010C940E010C940E010C940E0114
:100030000C940E010C940E010C940E010C940E0104
:100040000C94FE0F0C940E010C94CC0F0C94A60F84
:100050000C940E010C940E010C940E010C940E01E4
:100060000C9448100C940E0100407A10F35A00A032

i add a .zip with the arduio hex, the tasmo.bin and my user_config_override

j0seph1387, 2020-09-04:
the same .hex uploads with an esp8266 nodemcu and a wemos d1 mini uploads without any problems to the arduino

arendst, 2020-09-04:
The upload works fine on my esp32-cam too:
```
Sep  4 12:43:53 esp32cam2 ESP-CFG: Saved, Count 432, Bytes 4096
Sep  4 12:43:53 esp32cam2 ESP-UPL: File TasmotaClient_PCBox_V2.ino.eightanaloginputs.hex ...
Sep  4 12:43:55 esp32cam2 ESP-UPL: File type 3
Sep  4 12:43:59 esp32cam2 ESP-UPL: Successful 49371 bytes. Restarting
Sep  4 12:43:59 esp32cam2 ESP-HTP: Upload done
Sep  4 12:44:11 esp32cam2 ESP-TCL: Bootloader could not be found
Sep  4 12:44:11 esp32cam2 ESP-TCL: Flashing aborted!
```

j0seph1387, 2020-09-04:
:( ok, thank u! i will try i tonight again and report to u.

j0seph1387, 2020-09-04:
my output when I upload the .hex file:

18:11:20 RSL: tele/tasmota_2B5654/RESULT = {"TasmotaClient":{"CF": 75, "VF": 125, "AP" 150, "AUTO": on, "MODE": 1}}
18:11:25 UPL: Datei TasmotaClient_PCBox_V2.ino.eightanaloginputs.hex ...
abort() was called at PC 0x4017c5dd on core 1

Backtrace: 0x40093090:0x3ffb1950 0x400932c1:0x3ffb1970 0x4017c5dd:0x3ffb1990 0x4008c20b:0x3ffb19b0 0x40113345:0x3ffb19d0 0x400df6a9:0x3ffb19f0 0x400ed768:0x3ffb1a10 0x401913b1:0x3ffb1a30 0x40109d92:0x3ffb1a50 0x40109dd9:0x3ffb1a70 0x4010a97a:0x3ffb1aa0 0x4010b61d:0x3ffb1ad0 0x4010c4f5:0x3ffb1cc0 0x4010a00d:0x3ffb1ee0 0x400da083:0x3ffb1f30 0x400ebf2c:0x3ffb1f50 0x400dfdab:0x3ffb1f70 0x4010447e:0x3ffb1f90 0x40117ded:0x3ffb1fb0 0x4008f619:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
⸮00:00:00 CFG: Loaded, zählen 22
00:00:00 QPC: Count 1
00:00:00 Projekt tasmota Tasmota Version 8.4.0.3(webcam)-1_0_4

I will compile a new tasmota.bin and try my luck again.

j0seph1387, 2020-09-04:
I compiled a new firmware with gitpod (platformio run -e tasmota32-webcam), flashed tasmota32-minimal, then the new firmware.bin, camara and client work fine!  .hex file upload ended with:

19:33:54 RSL: tele/tasmota_2B5654/RESULT = {"TasmotaClient":{"TempIn": 0.00, "HumIn": 0.00, "PrIn": 0.00}}
19:33:54 RSL: tele/tasmota_2B5654/RESULT = {"TasmotaClient":{"CF": 50, "VF": 48, "AP" 150, "AUTO": on, "MODE": 1}}
19:33:57 UPL: Datei TasmotaClient_PCBox_V2.ino.eightanaloginputs.hex ...
abort() was called at PC 0x4017a085 on core 1

Backtrace: 0x40093090:0x3ffb1950 0x400932c1:0x3ffb1970 0x4017a085:0x3ffb1990 0x4008c20b:0x3ffb19b0 0x4010fb15:0x3ffb19d0 0x400df5f1:0x3ffb19f0 0x400ec556:0x3ffb1a10 0x4018ee45:0x3ffb1a30 0x40108416:0x3ffb1a50 0x4010845d:0x3ffb1a70 0x40108ffe:0x3ffb1aa0 0x40109ca1:0x3ffb1ad0 0x4010ab79:0x3ffb1cc0 0x40108691:0x3ffb1ee0 0x400d9fd3:0x3ffb1f30 0x400ead18:0x3ffb1f50 0x400dfcf3:0x3ffb1f70 0x40102e02:0x3ffb1f90 0x401158b5:0x3ffb1fb0 0x4008f619:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
00:00:00 CFG: Loaded, zählen 64
00:00:00 QPC: Count 1
00:00:00 Projekt tasmota Tasmota Version 8.4.0.3(webcam)-1_0_4

j0seph1387, 2020-09-04:
next attempt ... new firmware on gitpod with "platformio run -e tasmota32" compiled (without "-webcam"), flashed, everything works fine. .hex upload started and it works!
then i checked for the new messages from the arduino but they were the old ones. after that i tried it 5 more times, always the same result, serial output tells me:

20:00:26 RSL: tele / tasmota_2B5654 / RESULT = {"TasmotaClient": {"TempIn": 0.00, "HumIn": 0.00, "PrIn": 0.00}}
20:00:27 RSL: tele / tasmota_2B5654 / RESULT = {"TasmotaClient": {"CF": 57, "VF": 107, "AP" 150, "AUTO": on, "MODE": 1}}
20:00:30 UPL: TasmotaClient_PCBox_V2.ino.eightanaloginputs.hex file ...
20:00:35 UPL: successful 49379 bytes
20:00:35 TCL: Found bootloader
20:00:37 TCL: Flash done!
20:00:37 RSL: tele / tasmota_2B5654 / RESULT = {"TasmotaClient": {"TempIn": 0.00, "HumIn": 0.00, "PrIn": 0.00}}
20:00:37 RSL: tele / tasmota_2B5654 / RESULT = {"TasmotaClient": {"CF": 100, "VF": 150, "AP" 150, "AUTO": on, "MODE": 1}}
20:00:38 APP: restart
ets Jun 8 2016 00:22:57

rst: 0xc (SW_CPU_RESET), boot: 0x1a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP: 0xee
clk_drv: 0x00, q_drv: 0x00, d_drv: 0x00, cs0_drv: 0x00, hd_drv: 0x00, wp_drv: 0x00
mode: DOUT, clock div: 2
load: 0x3fff0018, len: 4
load: 0x3fff001c, len: 1044
load: 0x40078000, len: 8896
load: 0x40080400, len: 5828
entry 0x400806ac
00:00:00 CFG: Loaded, count 82
00:00:00 QPC: Count 1
00:00:00 Project tasmota Tasmota version 8.4.0.3 (tasmota) -1_0_4

the firmware on the arduino remains the old one.
tested on a second ep32 cam that is permanently installed in a circuit tested, unfortunately same result, but i can't read the serial output. the upload works, but the arduino is sending the old messages ...

j0seph1387, 2020-09-05:
ok, back locally. i downloaded the latest tasmota dev, a fresh vs code portable and the tasmota compile package. compiled, activated with tasmota32, tasmota32-webacm and tasmota32-sensors in platform_override.ini . I flashed every version and tested the .hex upload several times ...

tasmota32: website feedback a bit slow, camera jerks and has stripes, hex upload says Flash done but the arduino answers with old firmware

tasmota32-webacm: camera has stripes but does not jerk as much, hex upload says Flash done, Flash is done but the arduino no longer answers, arduino flashed with the same hex file, client speaks again, after the second hex upload the arduino needed a new bootloader: D

tasmota32 sensors: best camera picture, no stripes, smooth picture, hex upload says Flash done but the arduino replies with old firmware

tomorrow i will compile all versions again with gitpod and start and test the same game again
[image]

j0seph1387, 2020-09-05:
hello, i have to report to you that i have the same problems with the gitpod compiled tasmota.bin data!

tasmota32: tells me flash successful, arduino replies with old messages
tasmota32-webcam: tells me flash successful, arduino no longer responds, code broken
tasmota32-sensors: tells me flash successful, arduino replies with old messages

i have tried the .hex upload at least three times with every tasmota version! I loaded the tasmota32-minimal between every tasmota.bin upload!

Before every .hex upload, I checked whether the arduino is working and sending its messages!

I'm sorry, but could you tell me what else to do or what information you still need from me?

j0seph1387, 2020-09-06:
hello, today i started a new round with gitpod, new tasmota.bin compiled for tasmota32 and tasmota32-webcam with the same result.

tasmota32: tells me Flash done, but arduino answers with old firmware

tasmota32-webcam: tells me Flash done, the arduino no longer responds, code broken

I also checked the "ClientReset" command, it works without any problems.

could you please tell me what else i could check or what information you still need from me

arendst, 2020-09-06:
> tasmota32: tells me Flash done, but arduino answers with old firmware

Do you see the flash done a minute after the upload succeeded ro earlier. To me it seems the real flash starts almost 30 seconds after the upload succeeds.

You might want to look at either the serial console and/or the weblog (options 3 set) to see what really goes on.

For me this issue is on hold. Other using TasmotaCLient may chime in.

j0seph1387, 2020-09-06:
my serial output with the tasmota32.bin after the .hex upload:

17:09:37 RSL: tele / tasmota_2B5654 / RESULT = {"TasmotaClient": {"CFan": 100, "VFan": 135, "AP" 150, "AUTO": on, "MODE": 255}}
17:09:40 RSL: tele / tasmota_2B5654 / STATE = {"Time": "2020-09-06T17: 09: 40", "Uptime": "0T00: 05: 11", "UptimeSec": 311, "Heap ": 73," SleepMode ":" Dynamic "," Sleep ": 50," LoadAvg ": 21," MqttCount ": 0," POWER ":" OFF "," Dimmer ": 10," Fade ":" OFF "," Speed ​​": 1," LedTable ":" ON "," Wifi ": {" AP ": 1," SSId ":" FRITZ! Box 6490 Cable "," BSSId ":" C8: 0E: 14: A2: AC: 49 "," Channel ": 1," RSSI ": 100," Signal ": - 40," LinkCount ": 1," Downtime ":" 0T00: 00: 04 "}}
17:09:40 RSL: tele / tasmota_2B5654 / SENSOR = {"Time": "2020-09-06T17: 09: 40", "TasmotaClient": {"TempI": 0.00, "huI": 0.00, "PrIn" : 0.00}}
17:09:41 RSL: tele / tasmota_2B5654 / RESULT = {"TasmotaClient": {"TempIn": 0.00, "HumIn": 0.00, "PresIn": 0.00}}
17:09:42 RSL: tele / tasmota_2B5654 / RESULT = {"TasmotaClient": {"CFan": 100, "VFan": 134, "AP" 150, "AUTO": on, "MODE": 255}}
17:09:42 UPL: TasmotaClient_PCBox_V2.ino.eightanaloginputs.hex file ...
17:09:47 UPL: successful 49452 bytes
17:09:48 TCL: Found bootloader
17:09:51 TCL: Flash done!
17:09:51 RSL: tele / tasmota_2B5654 / RESULT = {"TasmotaClient": {"TempIn": 0.00, "HumIn": 0.00, "PresIn": 0.00}}
17:09:52 RSL: tele / tasmota_2B5654 / RESULT = {"TasmotaClient": {"CFan": 100, "VFan": 150, "AP" 150, "AUTO": on, "MODE": 255}}
17:09:53 APP: restart
ets Jun 8 2016 00:22:57

rst: 0xc (SW_CPU_RESET), boot: 0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP: 0xee
clk_drv: 0x00, q_drv: 0x00, d_drv: 0x00, cs0_drv: 0x00, hd_drv: 0x00, wp_drv: 0x00
mode: DOUT, clock div: 2
load: 0x3fff0018, len: 4
load: 0x3fff001c, len: 1044
load: 0x40078000, len: 8896
load: 0x40080400, len: 5828
entry 0x400806ac
00:00:00 CFG: Loaded, count 311
00:00:00 QPC: Count 1
00:00:00 Project tasmota Tasmota version 8.4.0.3 (tasmota) -1_0_4
00:00:00 WIF: connect with AP1 FRITZ! Box 6490 Cable Channel 1 BSSId C8: 0E: 14: A2: AC: 49 in mode 11 as tasmota_2B5654-5716 ...
00:00:03 WIF: connected

yes, i see the successful upload in the output, but the arduino answers with the old output after the restart

j0seph1387, 2020-09-08:
Hello Mr. Arendst, may I ask you something else about TasmotaClient? So I don't really want to open up a new issue.
I miss the "ClientSend" and "CientReset" commands on the Tasmota docs page and the more important question, what will happen to the TasmotaClient because andrethomas is not updating his TasmotaSlave lib to TasmotaClient?

arendst, 2020-09-09:
> I miss the "ClientSend" and "CientReset" commands on the Tasmota docs page 

You can add them yourselves. See docs howto.

> what will happen to the TasmotaClient because andrethomas is not updating his TasmotaSlave lib to TasmotaClient

Open an issue at his github TasmotaSlave lib and request updates/changes if they are needed.

closed 2020-11-15 by stale[bot]