arendst/Tasmota #9227: I2cDriver16 0 command crashes the i2cbus (and i2cscans result in error) t

kevinbaluha, 2020-09-03:
### PROBLEM DESCRIPTION

running on NodeMCU

built with USE_VL53L0X defined. TSL2561 has address collision so I ran I2cDriver16 0 in the console. now i2cscan has an error. issuing I2cDriver16 1 does not fix it. To fix it I had to reset and erase flash.

first did it on master branch fresh build then failed on 8.0.4 precompiled, then failed on custom build.

build environment mac os using atom.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Generic
- [ x] Tasmota binary firmware version number used: 8.0.4 (inititially)
  - [x] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: atom
- [x] Flashing tools used: nodeMCU pyflasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
19:46:50 CMD: Backlog Template; Module; GPIO 255`:
19:46:51 RSL: stat/tasmota_1A993C/RESULT = {"NAME":"NodeMCU I2c + temp","GPIO":[255,255,255,255,6,5,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:46:51 RSL: stat/tasmota_1A993C/RESULT = {"Module":{"1":"Sonoff Basic"}}
19:46:51 RSL: stat/tasmota_1A993C/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
  Configuration output here:

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
00:00:00 CFG: Loaded from flash at F5, Count 23
00:00:00 Project tasmota Tasmota Version 8.4.0(sensors)-2_7_2_1
00:00:00 WIF: Connecting to AP1 IOT Channel 6 BSSId 46:D9:E7:23:58:65 in mode 11N as tasmota_1A993C-6460...
00:00:01 WIF: Connected
19:50:36 HTP: Web server active on tasmota_1A993C-6460 with IP address 192.168.102.120
19:50:37 RSL: tele/tasmota_1A993C/INFO1 = {"Module":"Sonoff Basic","Version":"8.4.0(sensors)","FallbackTopic":"cmnd/DVES_1A993C_fb/","GroupTopic":"cmnd/tasmotas/"}
19:50:37 RSL: tele/tasmota_1A993C/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_1A993C-6460","IPAddress":"192.168.102.120"}
19:50:37 RSL: tele/tasmota_1A993C/INFO3 = {"RestartReason":"Software/System restart"}
19:50:37 RSL: stat/tasmota_1A993C/RESULT = {"POWER":"ON"}
19:50:37 RSL: stat/tasmota_1A993C/POWER = ON
19:50:41 RSL: tele/tasmota_1A993C/STATE = {"Time":"2020-09-03T19:50:41","Uptime":"0T00:00:09","UptimeSec":9,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"IOT","BSSId":"46:D9:E7:23:58:65","Channel":6,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:50:42 CMD: Status 0
19:50:42 RSL: stat/tasmota_1A993C/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_1A993C","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:50:42 RSL: stat/tasmota_1A993C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:10","StartupUTC":"2020-09-03T18:50:32","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2020-09-03T19:03:55","SaveCount":23,"SaveAddress":"F5000"}}
19:50:42 RSL: stat/tasmota_1A993C/STATUS2 = {"StatusFWR":{"Version":"8.4.0(sensors)","BuildDateTime":"2020-09-03T10:53:34","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"382/699"}}
19:50:42 RSL: stat/tasmota_1A993C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOT","IOT"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000"]}}
19:50:42 RSL: stat/tasmota_1A993C/STATUS4 = {"StatusMEM":{"ProgramSize":625,"Free":376,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":2,"Features":["00000809","8FDA8787","0415A005","B7FFBFCD","05DA9BC4","64367CC7","00084052"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
19:50:42 RSL: stat/tasmota_1A993C/STATUS5 = {"StatusNET":{"Hostname":"tasmota_1A993C-6460","IPAddress":"192.168.102.120","Gateway":"192.168.102.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.102.1","Mac":"84:F3:EB:1A:99:3C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:50:42 RSL: stat/tasmota_1A993C/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1A993C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:50:42 RSL: stat/tasmota_1A993C/STATUS7 = {"StatusTIM":{"UTC":"2020-09-03T18:50:42","Local":"2020-09-03T19:50:42","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:10","Sunset":"19:27"}}
19:50:42 RSL: stat/tasmota_1A993C/STATUS10 = {"StatusSNS":{"Time":"2020-09-03T19:50:42"}}
19:50:42 RSL: stat/tasmota_1A993C/STATUS11 = {"StatusSTS":{"Time":"2020-09-03T19:50:42","Uptime":"0T00:00:10","UptimeSec":10,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"IOT","BSSId":"46:D9:E7:23:58:65","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```

```
  Console output here:
19:53:01 CMD: i2cscan
19:53:01 SRC: WebConsole from 192.168.102.53
19:53:01 CMD: Group 0, Index 1, Command "I2CSCAN", Data ""
19:53:01 RSL: stat/tasmota_1A993C/RESULT = {"Command":"Error"}

```

### TO REPRODUCE
Configure I2c. have something running.
issue console command:
i2cDriver16 0
when it restarts type i2cscan.

### EXPECTED BEHAVIOUR
i2cDriver16 (TSL2561) should be disabled and other drivers with the same address will probe and intitialize.

### SCREENSHOTS

### ADDITIONAL CONTEXT
trying to get TOF sensor working.

**(Please, remember to close the issue when the problem has been addressed)**

device111, 2020-09-04:
you compile yourself. deactivate the TSL2561 driver, compile and try it again.

kevinbaluha, 2020-09-04:
can you help - how do you disable the TSL2561 driver for a particular build. (tasmota-sensors).

joba-1, 2020-09-04:
https://github.com/arendst/Tasmota/blob/146eb6654d421c82a14b6ec2e0e3357be199eaf5/tasmota/my_user_config.h#L510

make sure this is not #defined when you compile.

kevinbaluha, 2020-09-04:
not exactly what i was looking for. (what controls what's built for a given build tasmota-sensors, tasmota etc.

device111, 2020-09-04:
If you change the my_user_config.h like @joba-1 joba wrote , take the "tasmota.bin" to flash.

device111, 2020-09-04:
..and you must activate Your Sensor with
`#define USE_VL53L0X`

kevinbaluha, 2020-09-04:
ok thanks for you 'limited' feedback. I will test the way you suggested but it still doesn't address the issue of the console command I2cDriver16 0 creating an error that the i2c bus does not recover from. (I will test, isolate and push a change). also looking for the way to control what's built for the tasmota-sensors build.

device111, 2020-09-04:
a little self initiative can be expected in a free project. All information can be found here.
We try to find a Solution for you...
So take the Infos or leave it.

closed 2020-09-05 by ascillato2