arendst/Tasmota #9228: PMS5003/PMS7003 with Tasmota sensors 8.3, 8.3.1 and 8.4 "kills" sonoff CPU

sirfalo, 2020-09-03:
Hi,

I found out that after upgrade to 8.4, my sonoff was available for like 1 minute and then dissapeared from network.
After some investigation with different firmwares and checking how fast it reacts, I noticed that at start sonoff works, but second after second it responds slower and slower. From MQTT I can switch it maybe 2-3 times, but even when using just button on WWW, it reacts slower and slower, eventualy is not accessible.
When I unplug PMS sensor, all start to works normally.

PMS uses only 1 signal to sonoff.

On 8.2 release all works OK.

closed 2020-09-03 by close-issue-app[bot]

ascillato, 2020-09-04:
Seems not to be a software problem. Seems to be a hardware problem. Your sensor is draining too much current from your sonoff device. It is a sonoff basic? If it is, those are known to have a very weak power supply. A sonoff basic power supply was designed ONLY to power itself

sirfalo, 2020-09-04:
Well... But why it worked very well for more than a year on 6.5? And now works perfectly fine on 8.2. So doesn't sound like hardware issue...
Unless, new release somehow drains more power from CPU?

arendst, 2020-09-06:
From 8.3.0 release notes:

> - Add command ``Sensor18 0..32000`` to control PMS5003 sensor interval to extend lifetime by Gene Ruebsamen (#8128)

The PMS5003 now optionally needs both Rx and Tx signals connected. Pls try to connect Tx too and see what happens.

Jason2866, 2020-09-07:
Mhh, the change should not affect the behaviour if TX is not connected.
My PMS7003 is working without a problem with latest version.
I added the TX connection, so it still can be a bug in the driver maybe not behaving like before.

sirfalo, 2020-09-07:
> From 8.3.0 release notes:
> 
> > * Add command `Sensor18 0..32000` to control PMS5003 sensor interval to extend lifetime by Gene Ruebsamen (#8128)
> 
> The PMS5003 now optionally needs both Rx and Tx signals connected. Pls try to connect Tx too and see what happens.

Hi Theo,
Thanks for answer :) I need to make some wirings there, but I will let you know results :)