arendst/Tasmota #9235: Random temperatures spikes

JanickGers85, 2020-09-04:
### PROBLEM DESCRIPTION
I have many Shelly in my house, all with Tasmota firmware.
I experience unexpected synchronized temperature spike in all of them except a Shelly Plug for no apparent reason, moreover one of the goes randomly up to 90-95°C with a relay always on but without any active load over it (I use a rule to switch on a smart bulb connected to the always on relay).
All the devices are in the same house with the same room temperatures and are not exposed to the sun.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 1, Shelly 2.5
- [x] Tasmota binary firmware version number used: 8.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: espeasy
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
Shelly 2.5 #1 (temperature spikes)
00:12:28 MQT: stat/shelly-cucina/RESULT = {"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
00:12:28 MQT: stat/shelly-cucina/RESULT = {"Module":{"0":"Shelly 2.5"}}
00:12:28 MQT: stat/shelly-cucina/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"Nessuno"},"GPIO2":{"17":"Pulsante1"},"GPIO3":{"0":"Nessuno"},"GPIO4":{"21":"Relè1"},"GPIO5":{"83":"Switch2n"},"GPIO9":{"0":"Nessuno"},"GPIO10":{"0":"Nessuno"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"82":"Switch1n"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"22":"Relè2"},"GPIO16":{"156":"ADE7953 - IRQ"}}

Shelly 2.5 #2 (temperature spikes)
00:14:00 MQT: stat/shelly-giardino/RESULT = {"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
00:14:00 MQT: stat/shelly-giardino/RESULT = {"Module":{"0":"Shelly 2.5"}}
00:14:00 MQT: stat/shelly-giardino/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"Nessuno"},"GPIO2":{"17":"Pulsante1"},"GPIO3":{"0":"Nessuno"},"GPIO4":{"21":"Relè1"},"GPIO5":{"83":"Switch2n"},"GPIO9":{"0":"Nessuno"},"GPIO10":{"0":"Nessuno"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"82":"Switch1n"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"22":"Relè2"},"GPIO16":{"156":"ADE7953 - IRQ"}}

Shelly 2.5 #3 (temperature spikes)
00:15:15 MQT: stat/shelly-bagno/RESULT = {"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
00:15:15 MQT: stat/shelly-bagno/RESULT = {"Module":{"0":"Shelly 2.5"}}
00:15:15 MQT: stat/shelly-bagno/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"Nessuno"},"GPIO2":{"17":"Pulsante1"},"GPIO3":{"0":"Nessuno"},"GPIO4":{"21":"Relè1"},"GPIO5":{"83":"Switch2n"},"GPIO9":{"0":"Nessuno"},"GPIO10":{"0":"Nessuno"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"82":"Switch1n"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"22":"Relè2"},"GPIO16":{"156":"ADE7953 - IRQ"}}

Shelly 2.5 #4 (temperature spikes + reaching 90/95°)
00:16:29 MQT: stat/shelly-salotto/RESULT = {"NAME":"Shelly 2.5","GPIO":[56,0,17,0,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
00:16:29 MQT: stat/shelly-salotto/RESULT = {"Module":{"0":"Shelly 2.5"}}
00:16:29 MQT: stat/shelly-salotto/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"0":"Nessuno"},"GPIO2":{"17":"Pulsante1"},"GPIO3":{"0":"Nessuno"},"GPIO4":{"21":"Relè1"},"GPIO5":{"83":"Switch2n"},"GPIO9":{"0":"Nessuno"},"GPIO10":{"0":"Nessuno"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"82":"Switch1n"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"22":"Relè2"},"GPIO16":{"156":"ADE7953 - IRQ"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
Rules only on Shelly 2.5 #4 (temperature spikes + reaching 90/95°)
00:18:40 MQT: stat/shelly-salotto/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":72,"Free":439,"Rules":"ON switch1#state DO WebSend [192.168.1.33] /cm?cmnd=Power%20TOGGLE ENDON"}
00:18:41 MQT: stat/shelly-salotto/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
00:18:41 MQT: stat/shelly-salotto/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
Shelly 2.5 #1 (temperature spikes)
00:20:05 MQT: stat/shelly-cucina/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"shelly-cucina","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:20:05 MQT: stat/shelly-cucina/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-IT.bin","RestartReason":"External System","Uptime":"9T05:01:07","StartupUTC":"2020-08-26T17:18:58","Sleep":50,"CfgHolder":4617,"BootCount":43,"BCResetTime":"2020-07-04T13:26:14","SaveCount":947,"SaveAddress":"FB000"}}
00:20:05 MQT: stat/shelly-cucina/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020-07-29T12:26:48","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"412/699"}}
00:20:05 MQT: stat/shelly-cucina/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Dino-HOME",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000"]}}
00:20:05 MQT: stat/shelly-cucina/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000410","8FDAE797","04368001","000000CD","010013C0","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
00:20:05 MQT: stat/shelly-cucina/STATUS5 = {"StatusNET":{"Hostname":"shelly-cucina-7819","IPAddress":"192.168.1.31","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"98:F4:AB:B8:9E:8B","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:20:05 MQT: stat/shelly-cucina/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.217","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B89E8B","MqttUser":"mosquitto","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:20:05 MQT: stat/shelly-cucina/STATUS7 = {"StatusTIM":{"UTC":"2020-09-04T22:20:05","Local":"2020-09-05T00:20:05","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":99,"Sunrise":"06:40","Sunset":"19:40"}}
00:20:05 MQT: stat/shelly-cucina/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:20:05 MQT: stat/shelly-cucina/STATUS10 = {"StatusSNS":{"Time":"2020-09-05T00:20:05","Switch1":"ON","Switch2":"OFF","ANALOG":{"Temperature":58.3},"ENERGY":{"TotalStartTime":"2020-07-04T13:26:14","Total":0.194,"Yesterday":0.004,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Frequency":0,"Voltage":0,"Current":[0.000,0.000]},"TempUnit":"C"}}
00:20:05 MQT: stat/shelly-cucina/STATUS11 = {"StatusSTS":{"Time":"2020-09-05T00:20:05","Uptime":"9T05:01:07","UptimeSec":795667,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Dino-HOME","BSSId":"DC:39:6F:EF:E0:B5","Channel":6,"RSSI":100,"Signal":-29,"LinkCount":2,"Downtime":"0T00:02:12"}}}
00:20:06 MQT: tele/shelly-cucina/STATE = {"Time":"2020-09-05T00:20:06","Uptime":"9T05:01:08","UptimeSec":795668,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"MqttCount":3,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Dino-HOME","BSSId":"DC:39:6F:EF:E0:B5","Channel":6,"RSSI":100,"Signal":-30,"LinkCount":2,"Downtime":"0T00:02:12"}}
[... cut ...]
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
Not applicable

```

### TO REPRODUCE
It's completely random, or at least it seems so.

### EXPECTED BEHAVIOUR
No temperature spikes when no load is used, and surely no synchronization of the spikes

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
I use Home Assistant with MQTT integration.

**(Please, remember to close the issue when the problem has been addressed)**

lalo-uy, 2020-09-05:
DS18B20 sensors go to 85C if read after resest without previus start
conversion. In a network ( I have several of 6 to 20 devices ) that
indicate noise in the Vcc power line. If the spike is synchronized on
several device I suspect of power noise. 100nF cap between Vcc and Gnd as
near to de sensor as possible correct most of thes error for me.

ascillato2, 2020-09-13:
Hi, any news on this?

ascillato2, 2020-09-15:
Please, when you have time, tell us if the proposal from @lalo-uy solved your problem. Thanks.

closed 2020-09-15 by ascillato2