arendst/Tasmota #9238: Alternative PWM implementation for ESP8266 devices

bobmarley2021, 2020-09-05:
**Have you looked for this feature in other issues and in the docs?**  

Yes

**Is your feature request related to a problem? Please describe.**  

It's not an issue with Tasmota, but a limitation of the 8266s software PWM implementation. 

**Describe the solution you'd like**  

There is a project that claims to have greatly improved the software PWM for 8266 devices here: https://lurchi.wordpress.com/2016/06/29/esp8266-pwm-revisited-and-reimplemented/

Features include up to 100% duty cycle, and up to 25KHz frequency. This would greatly improve usage on 8266s, especially boards like the Sinilink XY-WFMS, where the fets are included and the 8266 cannot be swapped out for a 32.

Adding a `setoption` to switch to the alternative implementation would be ideal :)

**Describe alternatives you've considered**  

Alternative firmware? But they all use the same limited PWM implementation. Ie. No more than 90% duty cycle, no more than ~2KHz etc.

**Additional context**  

The project's git is here: https://github.com/StefanBruens/ESP8266_new_pwm

These forks appear to have more recent activity:

https://github.com/UweHeinritz/ESP8266_new_pwm
https://github.com/emeggiolaro/ESP8266_new_pwm
https://github.com/nochkin/new_pwm

Jason2866, 2020-09-05:
We already use different PWM code for Tasmota which performs very nice.
See https://github.com/tasmota/Arduino/tree/2.7.4.1-tasmota => `platform_packages = jason2866/framework-arduinoespressif8266` 
uses https://github.com/esp8266/Arduino/pull/7022

s-hadinger, 2020-09-05:
@bobmarley2021 The code you are referring to is 4 years old. What makes you think it is better than the latest from Arduino core that Tasmota is currently using?

bobmarley2021, 2020-09-05:
Hi both,

Forgive me, but based on the symptoms I experienced (the slider in the UI only being useful in the last 10%, and the fact that I didn't seem to be able to adjust the pwm range (I assume this would affect the slider) I thought that the implementation was not improved. 

Plus: the Tasmota documentation clearly states that: `100Hz to 4kHz` is the minimum/maximum of the PWM frequency, which is closer to the manufacturer implementation rather than the improved implementation from the blog. 

Can you tell me how I set the pwmfrequency to 20000Hz and adjust the range of the slider?

s-hadinger, 2020-09-05:
Tasmota limits on purpose to 4kHz because for leds and servos it doesn't make sense to go beyond. 

Also please not that the pwm résolution is 10 bits and the more you increase the frequency the more you lose resolution. 

20 KHz at 10 bits would mean a granularity of 48ns I.e. less than 4 CPU cycles. Sorry but that's not possible. 

What are you trying to achieve?

bobmarley2021, 2020-09-05:
Ah ok. Thanks for the explanation. 

I have a 5V 2 wire fan. Those cheapo USB powered desk fans you can find on the usual sites. I have removed its PCB and have replaced it with a Sinilink XY-WFMS. 

The aim is to have a HA controllable fan. The problem I am experiencing is that I am struggling to find the correct / set the correct pwmfreq and pwmrange. When I read the docs limitation of 4K and the blog post, I've incorrectly assumed this is a firmware limitation. 

All I seem to be able to get is a slider that works of any use in the last 10%, ie all the way to the right. When the slider is to the left, the fan is whiny and slow. I also don't think the maximum speed is very good. Clearly there's something that needs changing to do with the frequency and range, but everything I've tried via the console doesn't seem to produce favourable results. 

Perhaps I am not typing the console commands correctly?

s-hadinger, 2020-09-05:
What template are you using? If it's a fan you definitely need to turn off gamma correction with `LedTable 0`

Jason2866, 2020-09-05:
I would even go further `SetOption15 0`

s-hadinger, 2020-09-05:
I did quick research. Indeed for fans they advise frequencies above 20khz to avoid noise, I.e. being higher than audible frequencies. 

The internal lib allows you to do that but you will need to recompile a version without the 4khz limit. Bear in mind that you will have fewer effective steps, probably less than 100. But for a fan I believe it's enough.

bobmarley2021, 2020-09-05:
I am using this: https://templates.blakadder.com/sinilink_XY-WFMS.html

But then change GPIO04 from relay to PWM1(37). 

I will try with `LedTable 0` and `SetOption15 0` and see if it improves. 

I wouldn't have a foggiest idea how to compile this and what modification to make... any chance a setoption to toggle this limit might be considered for future versions?

Jason2866, 2020-09-05:
Change this line https://github.com/s-hadinger/Tasmota/blob/development/tasmota/tasmota.h#L110
to 20000

bobmarley2021, 2020-09-05:
Great thanks. 

Looks like I might be able to muddle through with Platform IO for compiling. 

I'll give this a go tomorrow morning and see if it makes a difference then report back. 👍

bobmarley2021, 2020-09-06:
I have compiled a version of Tasmota altering the line you mentioned @Jason2866, but I am still having issues setting the PWM frequency to anything but 977! I even set a minimum of 2000 in the same file...

```09:44:01 CMD: pwmfrequency20000
09:44:01 RSL: stat/tasmota_857FFE/RESULT = {"PWMFrequency":977}
```

I also cannot set LED table to zero:

```
09:48:01 CMD: Ledtable 0
09:48:01 RSL: stat/tasmota_857FFE/RESULT = {"Command":"Unknown"}
```

Any ideas?

Jason2866, 2020-09-06:
If SetOption 15 is set ledtable is not active anymore (i think). Not tested...
Your command for PwmFrequency is wrong. `PwmFrequency 20000`
There has to be a ` ` !

bobmarley2021, 2020-09-06:
Gotcha, that's set now, but I don't appear to have any fan control at all and the slider in the UI is not present. The button on the board that also acted as a toggle no longer functions, despite being present in the template. I'm not sure what has happened...

s-hadinger, 2020-09-06:
Honestly nobody tested it. The Arduino lib supports up to 60khz.

bobmarley2021, 2020-09-06:
Ok, I'll have to have a play. Perhaps 20 is too high for this fan, I will need to decrease the value gradually.

What command would affect the range of the slider? Is that literally 'pwmrange'? I see it only goes up to 1023.. is this also limited by the build? It seems like a range of 2000+ would be more suitable!

s-hadinger, 2020-09-06:
You don't need to change the range. It only affects resolution. It defines the number of steps between 0% and 100%

bobmarley2021, 2020-09-06:
Ok, thanks for all the info and help :+1:

closed 2020-09-06 by bobmarley2021

rogerlew, 2020-12-22:
Hi All,

I also know this is an old issue. This thread came up when I was looking to use Tasmota to control the speed of a 4-pin 12 VDC fan.

I have a 12 VDC Noctua 4-pin PC fan that I use as a ventiliation fan for a small shed office. I was able to set PWM_Max ([tasmota/tasmota.h](https://github.com/s-hadinger/Tasmota/blob/845b89d260ab5260588526ab7556907a02a9d6a0/tasmota/tasmota.h#L115)) to 25000 and it works great!

### Wiring
Tasmota is on a Sonoff Basic. The relay powers a 24 VAC transformer which opens a motorized damper and a 12VDC power supply. The 12VDC supply is wired to the ground and VCC of the fan. The PWM is on GPIO14 and goes to the speed control pin. The ground from the sonoff is tied to the ground of the 12VDC power supply.

### Home Assistant

I have it configured in home assistant as a mqtt fan
```
fan:
  - platform: mqtt
    name: "Shed Vent"
    state_topic: "stat/sonoff03/POWER"
    command_topic: "cmnd/sonoff03/POWER"
    availability_topic: "tele/sonoff03/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    speed_command_topic: "cmnd/sonoff03/DIMMER"
    payload_low_speed: "65"
    payload_medium_speed: "75"
    payload_high_speed: "100"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: false
    speeds:
      - off
      - low
      - medium
      - high
```

Thanks!

Asarria081, 2022-05-23:
Hi.

I want to control the speed of two 12 volts fans using tasmota on a esp 8266, then I would try to control them as mqtt fans using @rogerlew  configuration in HA.
My esp8266 is configurated in Tasmota as a Generic Module.
What I need to configurate in tasmota to do it work? It´s enough choosing PWM option for any GPIO of my ESP8266 or I need to do anything else.