arendst/Tasmota #9241: Wakeup light in 256 steps

jesseruiter, 2020-09-05:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
When the wakeup command is called, the light is still quite bright at the start and the fading isn't very smooth.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
When the wakeup command uses 256 (instead of 100) steps the fading will be smoother and I don't wake up directly when the value is 1%😁

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I can't come up with an alternative solution.

**Additional context**  
_Add any other context or screenshots about the feature request here._
Not needed.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-09-05:
Please give more details about your setup. What kind of light are you using (pwm?) and did you enable gamma correction with `LedTable 1`?

jesseruiter, 2020-09-05:
I am using a 5050 SMD LED Strip in combination with a Magic Home controller. I don't use gamma correction, as LedTable == 0.

jesseruiter, 2020-09-16:
However, I don't think it depends on my setup, but on the way the wakeup function is implemented. Right now the least bright is 1 percent (so 1/100) instead of the possible 1/256. The latter would result in a smoother wakeup pattern.

s-hadinger, 2020-09-16:
Then I believe that you should enable Gamma Correction, otherwise the brightness goes very quickly to a high perceived value.

jesseruiter, 2020-09-18:
I tried that function, but then my brightness of my led strip isn't lineair anymore and the initial brightness is still 1/100 instead of the possible 1/256. I think this issue is therefore a feature request, instead of a question.

ascillato, 2020-09-18:
Hi,

Have you tried `LedTable 1`?

jesseruiter, 2020-09-22:
Yes I did, but that made the smoothness of the LED strip's brightness worse.

jesseruiter, 2020-09-22:
> I tried that function, but then my brightness of my led strip isn't lineair anymore and the initial brightness is still 1/100 instead of the possible 1/256. I think this issue is therefore a feature request, instead of a question.

But as said here, it is not so much a problem, but more a suggestion to smoothen the wake up pattern. Because I use the LED strip with the wakeup function as a wakeup light divided over 20 minutes, the experience would be better if the brightness is incremented 1/256 every ~5 seconds, instead of an increase of 1/100 every 12 seconds.

ascillato2, 2020-09-22:
> LED strip with the wakeup function as a wakeup light divided over 20 minutes

Ok. With that time frame you need a higher resolution. With smaller time frames the 100 steps are smooth enough. 

Changing to feature request. Thanks for explaining.

s-hadinger, 2020-09-22:
@jesseruiter I completely agree with you. When you use Gamma Correction `LedTable 1` it doesn't really make a difference because the first steps are very distant from each other and splitting in 256 wouldn't make a difference.

I don't have time in the short term to change this, but it shouldn't be a big change to the code, although we need to adjust the duration if it's 256 steps instead of 100.

jesseruiter, 2020-09-22:
Thanks for your reaction! I look forward to the update sometime in the future.

s-hadinger, 2020-09-26:
Oops, I observed a crash... Not ready

PR #9391 (2020-09-26): Change ``WakeUp`` uses 256 steps instead of 100 (#9241)
## Description:

**Related issue (if applicable):** fixes #9241 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.1
  - [x] The code change is tested and works on core ESP32 V.1.12.2
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE: The code change must pass CI  [... cut]
closed 2020-09-26 by s-hadinger