arendst/Tasmota #9245: Rule Break Not Working as Expected

PipeDeveloper, 2020-09-05:
### PROBLEM DESCRIPTION

When there is "BREAK" on rules, they doesnt work as expected when you try to trigger a value between a maximun or minimum.

Please this is just an example but this applies on most of configurations on this type. For example turn on a fan (maximum power, switch 2) when the temperature is above 20°C, turn on the same fan (minimum power, switch 1) when the temperature is between 10 and 20°C and turn off the fan when the temperature is lower 10°C.

Values are very floating so its necesary to set to `Rule1 5` option, to just read it once, otherwise if will be trigger rules all time like crazy.
So reading the rules cookbook is just a simple rule, something like this.
```
Rule1
on DS18B20#Temperature>20 do Action "A" break
on DS18B20#Temperature>10 do Action "B" endon
on DS18B20#Temperature<=10 do Action "C" endon
```
just to simplify
Action "A" should be a **backlog power1 0; power2 1** (maximum power)
Action "B" should be a **backlog power1 1; power2 0** (minimum power)
Action "C" should be a **backlog power1 0; power2 0** (all off)

**THE ISSUE ITSELF**
In the same example lets study what happen to the output if we input some values. Please follow the order, this is very important for the description of the issue.

Input: Sensor = 9°C
Output: Action "C" (thats fine)

Input: Sensor = 14°C
Output: Action "B" (thats fine)

Input: Sensor = 21°C
Output: Action "A" (thats fine)

Input: Sensor = 17°C
Output: **Nothing happens** (here is the issue, it should trigger Action "B")

Input: Sensor = 8°C
Output: Action "C"

In summary there are 6 combinations from one to another action. With that configuration
From "C" to "B" works
From "C" to "A" works
From "B" to "A" works
From "A" to "B" (this is not trigger)
From "A" to "C" works

**EXPERIMENTING**
I didnt stopped and tried to figure what happen if i try another order or combine endon and break seems interesting what happen with this configuration.

```
Rule1
on DS18B20#Temperature>20 do Action "A" break
on DS18B20#Temperature>10 do Action "B" break
on DS18B20#Temperature<=10 do Action "C" endon
```
To make it short this happens

From "C" to "B" works
From "C" to "A" works
From "B" to "A" (this is not trigger)
From "A" to "B" works
From "A" to "C" works

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used: Wemos D1 mini PRO
- [ ] Tasmota binary firmware version number used: 8.4.0(sensors) (Release branch) Core/SDK Version | 2_7_2_1/2.2.2-dev(38a443e)
  - [ ] Pre-compiled 8.4.0(sensors) (Release branch)
  - [ ] Self-compiled
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE

### EXPECTED BEHAVIOUR

The expectation is to have a fluid trigger depending the value.
In the same example is Rule1 should trigger the condition in both directions.

From A to B // from B to C // from C to B // from B to C
A -> B -> C -> B -> A

And exeptional cases
From A to C // from C to A
A -> C
C -> A

### SCREENSHOTS

### ADDITIONAL CONTEXT
This is maybe a bug that have the function BREAK, it seems that it doesnt refresh values to trigger because at a big change of values, it should trigger if the value reach to the field that correspond.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-09-07:
Do you have set to trigger ONCE in the rule?

PipeDeveloper, 2020-10-12:
Updated at latest version, issue still remaining

schup011, 2020-10-24:
I can confirm the same behavior, I wondered the whole time what I was doing wrong.

PipeDeveloper, 2020-10-24:
Yes! the break function is bugged when you try with three or more fields to trigger, no matter the order or the direcion.

arendst, 2020-10-25:
I'll try to understand the issue...

arendst, 2020-10-25:
The good thing is I can reproduce with the following rule:
```
on var1#state>20 do dimmer 50 break on var1#state>10 do dimmer 30 endon on var1#state<=10 do dimmer 10 endon
```
The bad thing is I now have to find a cure...

PipeDeveloper, 2020-10-26:
@arendst Thank you very much!

I'll update to release version 9.0.0.2
http://ota.tasmota.com/tasmota/

Let's see what happen

schup011, 2020-10-26:
I am not so sure, I will test, but in my case, it was not related with ONCE. I had the rule running on normally.

PipeDeveloper, 2020-10-26:
I'll wait till the 9.0.0.3 release is available to test the changes

PipeDeveloper, 2020-11-12:
Tested with http://ota.tasmota.com/tasmota/
tasmota-sensors.bin | http://ota.tasmota.com/tasmota/tasmota-sensors.bin | 631k | 9.1.0.1 | 20201112 15:17

The issue persists, but different.
My Rule
```
Rule1
on SR04#distance>133 do backlog LedPower1 0; LedPower2 0; LedPower3 1 break
on SR04#distance>74 do backlog LedPower1 0; LedPower2 1; LedPower3 0 endon
on SR04#distance<=74 do backlog LedPower1 1; LedPower2 0; LedPower3 0 endon
```
When a value above 133cm it triggers the first action, but then inmidiatly triggers the second action.

Example change from 85cm to 167cm and stay on 167cm
Output:
```
12:34:01 RUL: SR04#DISTANCE>133 performs "backlog LedPower1 0; LedPower2 0; LedPower3 1"
12:34:01 MQT: stat/estanque/RESULT = {"LedPower1":"OFF"}
12:34:02 MQT: stat/estanque/RESULT = {"LedPower2":"OFF"}
12:34:02 MQT: stat/estanque/RESULT = {"LedPower3":"ON"}
12:34:06 RUL: SR04#DISTANCE>74 performs "backlog LedPower1 0; LedPower2 1; LedPower3 0"
12:34:06 MQT: stat/estanque/RESULT = {"LedPower1":"OFF"}
12:34:06 MQT: stat/estanque/RESULT = {"LedPower2":"ON"}
12:34:06 MQT: stat/estanque/RESULT = {"LedPower3":"OFF"}
```

arendst, 2020-11-17:
Cannot reproduce. Using the shwon before:
```
on var1#state>20 do dimmer 50 break on var1#state>10 do dimmer 30 endon on var1#state<=10 do dimmer 10 endon
```
it still works as designed.

What you may encounter is a retrigger from the rule within 4 seconds which reports several distances due to "not so stable" sensor.

PipeDeveloper, 2020-11-17:
You may emulate by entering different values on `var1` on this case the most high value from break. (remember set rule1 5, to ONCE)
Try to enter random values above 20 in your case, 25-31-28-23, you should trigger `dimmer 50` then a few seconds after that it triggers `dimmer 30`.

For stability its ok for integer values, but for decimal, it is very floating on a range +/- 0.3
32.372
32.563
32.428

j0seph1387, 2020-11-17:
for me "break" doesn't work with this rule too

Rule1
   ON system # boot DO Backlog RuleTimer1 10; Var1 255 ENDON
   ON rules # timer! = 1 BREAK
   ON BME280 # Humidity <65 DO Backlog SUB1 1 ENDON
   ON BME280 # Humidity> 70 DO Backlog ADD1 1 ENDON
   ON Var1 # State <0 DO Var1 0 ENDON
   ON Var1 # State> 1023 DO Var1 1023 ENDON
   ON rules # timer = 1 DO Backlog PWM5% Var1%; Ruletimer1 5 ENDON

the humidity is triggered 2 times a second ...

22:27:58 CMD: Rule1   ON system#boot DO Backlog RuleTimer1 10; Var1 255 ENDON   ON rules#timer!=1  BREAK      ON BME280#Humidity<65 DO Backlog SUB1 1 ENDON     ON BME280#Humidity>70 DO Backlog ADD1 1 ENDON     ON Var1#State < 0 DO Var1 0 ENDON   ON Var1#State > 1023 DO Var1 1023 ENDON     ON rules#timer=1 DO Backlog PWM5 %Var1%; Ruletimer1 5 ENDON 
22:27:58 RUL: Stored uncompressed, would compress from 328 to 197 (-40%)
22:27:58 MQT: stat/tasmota_GB1_FCDDF0/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":328,"Free":183,"Rules":"ON system#boot DO Backlog RuleTimer1 10; Var1 255 ENDON   ON rules#timer!=1  BREAK      ON BME280#Humidity<65 DO Backlog SUB1 1 ENDON     ON BME280#Humidity>70 DO Backlog ADD1 1 ENDON     ON Var1#State < 0 DO Var1 0 ENDON   ON Var1#State > 1023 DO Var1 1023 ENDON     ON rules#timer=1 DO Backlog PWM5 %Var1%; Ruletimer1 5 ENDON "}
22:27:58 RUL: BME280#HUMIDITY<65 performs "Backlog SUB1 1"
22:27:58 MQT: stat/tasmota_GB1_FCDDF0/RESULT = {"Sub1":"122.000"}
22:27:59 RUL: BME280#HUMIDITY<65 performs "Backlog SUB1 1"
22:27:59 MQT: stat/tasmota_GB1_FCDDF0/RESULT = {"Sub1":"121.000"}
22:27:59 RUL: BME280#HUMIDITY<65 performs "Backlog SUB1 1"
22:27:59 MQT: stat/tasmota_GB1_FCDDF0/RESULT = {"Sub1":"120.000"}
22:27:59 RUL: RULES#TIMER=1 performs "Backlog PWM5 120.000; Ruletimer1 5"
22:28:00 MQT: stat/tasmota_GB1_FCDDF0/RESULT = {"PWM":{"PWM1":1023,"PWM2":1023,"PWM3":1023,"PWM4":1023,"PWM5":120}}
22:28:00 RUL: PWM#PWM5 performs "VAR5 120"
22:28:00 MQT: stat/tasmota_GB1_FCDDF0/RESULT = {"Var5":"120"}
22:28:00 MQT: stat/tasmota_GB1_FCDDF0/RESULT = {"T1":5,"T2":0,"T3":0,"T4":0,"T5":0,"T6":0,"T7":0,"T8":0}
22:28:00 RUL: BME280#HUMIDITY<65 performs "Backlog SUB1 1"
22:28:00 MQT: stat/tasmota_GB1_FCDDF0/RESULT = {"Sub1":"119.000"}
22:28:00 RUL: BME280#HUMIDITY<65 performs "Backlog SUB1 1"
22:28:01 MQT: stat/tasmota_GB1_FCDDF0/RESULT = {"Sub1":"118.000"}
22:28:01 RUL: BME280#HUMIDITY<65 performs "Backlog SUB1 1"
22:28:01 MQT: stat/tasmota_GB1_FCDDF0/RESULT = {"Sub1":"117.000"}
22:28:02 RUL: BME280#HUMIDITY<65 performs "Backlog SUB1 1"
22:28:02 MQT: stat/tasmota_GB1_FCDDF0/RESULT = {"Sub1":"116.000"}
22:28:02 RUL: BME280#HUMIDITY<65 performs "Backlog SUB1 1"

83/5000
or do i have a mistake in my rule or do i understand the break function incorrectly?

1008/5000
Before I wanted to send this I thought to myself, I'd rather test again with the current develop version, so off to gitpod :) but I can't compile the current dev version with my config for esp32cam, the master version compiles without problems.

Indexing .pio/build/tasmota32/libFrameworkArduino.a
Linking .pio/build/tasmota32/firmware.elf
/home/gitpod/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/tasmota32/firmware.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/home/gitpod/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/home/gitpod/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 18212 bytes
collect2: error: ld returned 1 exit status
*** [.pio/build/tasmota32/firmware.elf] Error 1

ascillato2, 2020-11-24:
@j0seph1387

your rules have syntax errors, that is why they don't work. You have spaces where they shouldn't be and you have missing some DO.

for example `ON system # boot DO Backlog RuleTimer1 10; Var1 255 ENDON`, should be `ON system#boot DO Backlog RuleTimer1 10; Var1 255 ENDON`

then, `ON rules # timer! = 1 BREAK` should be `ON rules#timer=1 do [some command] BREAK`

Please, re check the docs for syntax. Thanks.

ascillato2, 2020-11-24:
Sorry, I can not reproduce it either using the rule shown before. Seems that your sensor value is fluctuating.

closed 2020-11-24 by ascillato2