arendst/Tasmota #9248: Command MQTT send of Telemetry data

RogerHill, 2020-09-06:
**Have you looked for this feature in other issues and in the docs?**  
Yes, I have searched the forum for a solution with out success. I have also posted to the forum but have not received a workable solution.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I need to be able to command the Tasmota device to transmit immediate telemetry data via the MQTT telemetry topic.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I would like the addition of a status command to send immediate telementry data via the normal MQTT topic for telementry. I would like this command to be available from the console, Rules and MQTT. This does not need to reset the telemetry period, but that could be an option. The status command should not be included in the Status 0 to avoid conflict with management software.
 
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

I have tried using the Status 8 command to issue current sensor data but unfortunately this sends the sensor data to a different MQTT Topic and in a different JSON Format. I have Node-red flows watching the telemetry data and it is difficult to watch and decode seperate MQTT topics. I have also found that a number of the management software very frequently poll status 8 for unpdateing tasmota details.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2020-09-06:
Try to send command "teleperiod" (without parameters).
I think it answers with the status you are asking for.
Joerg

Jason2866, 2020-09-06:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support](https://github.com/arendst/Tasmota#support-information).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-06 by ascillato2

RogerHill, 2020-09-07:
Thank you Joerg.

That did indeed achieve what I wanted.

Thanks again Roger

PS Applogies if this was the wrong place to pose this, I had tried the forum without luck.