arendst/Tasmota #9250: HUE emulation - No devices detected, only Hub found in BOSCH Smarthome App

erichabg, 2020-09-06:
### PROBLEM DESCRIPTION
With Tasmota V8.2 the BOSCH Smarthome App finds a Hue Bridge and a Hue lamp for each Tasmota device and therefore the device can be switched on and off.
With Tasmota V8.4.0 and also the development V8.4.0.3 there is only the Hue bridge found but no Hue lamp behind the bridge. Therefore switching the lamp is not possible.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Wemos Mini D1, Sonoff Basic and also ESP32
- [x ] Tasmota binary firmware version number used: 8.4.0 and 8.4.0.3
  - [x] Pre-compiled
  - [x ] Self-compiled
    - [x] IDE / Compiler used: platformio
- [ x] Flashing tools used: esptool.py and OTA
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
For ESP32:
00:25:18 MQT: EFHome/clients/Garage2/stat/Garage2/1/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
00:25:19 MQT: EFHome/clients/Garage2/stat/Garage2/1/RESULT = {"Module":{"0":"ESP32-DevKit"}}
00:25:19 MQT: EFHome/clients/Garage2/stat/Garage2/1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"261":"Relay_i6"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"263":"Relay_i8"},"GPIO17":{"262":"Relay_i7"},"GPIO18":{"260":"Relay_i5"},"GPIO19":{"259":"Relay_i4"},"GPIO21":{"258":"Relay_i3"},"GPIO22":{"0":"None"},"GPIO23":{"256":"Relay_i1"},"GPIO25":{"257":"Relay_i2"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"1216":"AM2301"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"G ..

Wemos D1 mini:
12:20:39 MQT: EFHome/clients/Garage1/status/Garage1/1/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
12:20:39 MQT: EFHome/clients/Garage1/status/Garage1/1/RESULT = {"Module":{"18":"Generic"}}
12:20:39 MQT: EFHome/clients/Garage1/status/Garage1/1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"2":"AM2301"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}.

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
<no rules used>

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
12:22:27 MQT: EFHome/clients/Garage1/status/Garage1/1/STATUS = {"Status":{"Module":18,"DeviceName":"Garage1","FriendlyName":["Garage1"],"Topic":"Garage1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:22:27 MQT: EFHome/clients/Garage1/status/Garage1/1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://192.168.178.114/updatefw/sonoff.bin","RestartReason":"Exception","Uptime":"0T12:23:33","StartupUTC":"2020-09-05T22:58:54","Sleep":50,"CfgHolder":4617,"BootCount":124,"BCResetTime":"2020-08-09T19:27:00","SaveCount":296,"SaveAddress":"F7000"}}
12:22:27 MQT: EFHome/clients/Garage1/status/Garage1/1/STATUS2 = {"StatusFWR":{"Version":"8.4.0(tasmota)","BuildDateTime":"2020.09.05 12:33:49","Boot":31,"Core":"2_7_2_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"406/699"}}
12:22:27 MQT: EFHome/clients/Garage1/status/Garage1/1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["EFHOME2",""],"TelePeriod":60,"Resolution":"55C180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00000000","00006000"]}}
12:22:27 MQT: EFHome/clients/Garage1/status/Garage1/1/STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAE797","04368001","000000CD","010013F1","C000F981","00004004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6,29"}}
12:22:27 MQT: EFHome/clients/Garage1/status/Garage1/1/STATUS5 = {"StatusNET":{"Hostname":"Garage1-3415","IPAddress":"192.168.178.177","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"2C:F4:32:12:8D:57","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:22:27 MQT: EFHome/clients/Garage1/status/Garage1/1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.137","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_128D57","MqttUser":"mosquitto","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
12:22:27 MQT: EFHome/clients/Garage1/status/Garage1/1/STATUS7 = {"StatusTIM":{"UTC":"2020-09-06T11:22:27","Local":"2020-09-06T12:22:27","StartDST":"2020-03-29T01:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"05:41","Sunset":"18:46"}}
12:22:27 MQT: EFHome/clients/Garage1/status/Garage1/1/STATUS10 = {"StatusSNS":{"Time":"2020-09-06T12:22:27","AM2301":{"Temperature":26.9,"Humidity":43.4,"DewPoint":13.4},"MCP230XX":{"D0":1,"D1":0,"D2":1,"D3":1,"D4":1,"D5":1,"D6":0,"D7":1,"D8":1,"D9":1,"D10":1,"D11":1,"D12":1,"D13":1,"D14":1,"D15":1,"MCP230_OUT":{"OUT_D0":"OFF","OUT_D1":"ON","OUT_D2":"OFF","OUT_D3":"OFF","OUT_D4":"OFF","OUT_D5":"OFF","OUT_D6":"ON","OUT_D7":"OFF","END":1}},"TempUnit":"C"}}
12:22:27 MQT: EFHome/clients/Garage1/status/Garage1/1/STATUS11 = {"StatusSTS":{"Time":"2020-09-06T12:22:27","Uptime":"0T12:23:33","UptimeSec":44613,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"EFHOME2","BSSId":"xx:xx:xx:xx:18:1D","Channel":6,"RSSI":28,"Signal":-86,"LinkCount":1,"Downtime":"0T00:15:16"}}}
12:22:27 MQT: EFHome/clients/Garage1/status/Garage1/1/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40101674","4026418f","40264124","402640cb","40263228","40263251","40260cbc","4010396e","4025d025","40252071","402619b3","40000f58","40261426","4026ab77","4026a437","40248ce0","40000f49","40000f49","40000e19","40105aed","40105af3","4010000d","40269320","402692d1","40101160","40102738","401059e7","401032cb","401034ac","4010396e","4000050c"]}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
12:24:36 CMD: weblog 4
12:24:36 MQT: EFHome/clients/Garage1/status/Garage1/1/RESULT = {"WebLog":4}
12:24:36 CFG: in Flash gespeichert am F6, zählen 297, Bytes 4096
12:24:38 UDP: Packet (101)
12:24:39 HTP: Hue-API (/128d57)
12:24:39 WIF: Prüfe Verbindung...
12:24:41 UDP: Packet (101)
12:24:44 HTP: Hue-API (/128d57)
12:24:44 UDP: Packet (101)
12:24:49 UDP: Packet (101)
12:24:49 HTP: Hue-API (/128d57)
12:24:52 UDP: Packet (101)
12:24:54 HTP: Hue-API (/128d57)
12:24:55 UDP: Packet (101)
[... 52 lines cut ...]
12:26:12 HTP: Hue-API (/128d57)
12:26:15 UDP: Packet (101)
12:26:16 MQT: EFHome/clients/Garage1/status/Garage1/1/STATE = {"Time":"2020-09-06T12:26:16","Uptime":"0T12:27:22","UptimeSec":44842,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"Wifi":{"AP":1,"SSId":"EFHOME2","BSSId":"2C:3A:FD:B1:18:1D","Channel":6,"RSSI":28,"Signal":-86,"LinkCount":1,"Downtime":"0T00:15:16"}}
12:26:16 MQT: EFHome/clients/Garage1/status/Garage1/1/SENSOR = {"Time":"2020-09-06T12:26:16","AM2301":{"Temperature":26.8,"Humidity":43.3,"DewPoint":13.3},"MCP230XX":{"D0":1,"D1":0,"D2":1,"D3":1,"D4":1,"D5":1,"D6":0,"D7":1,"D8":1,"D9":1,"D10":1,"D11":1,"D12":1,"D1 [... cut ...]
-Use Bosch Smarthome App to detect new devices - Philipps Hue Bridge  -> detects Hue bridge and Hue lamp -> OK
-Delete the Hue bridge in the Bosch App (for testing)
-Update to Tasmota V8.4.x 
-Use Bosch Smarthome App to detect new devices - Philipps Hue Bridge  -> detects Hue bridge but no Hue lamp -> Error

### EXPECTED BEHAVIOUR
Hue bridge and at least one Hue lamp is detected

### SCREENSHOTS
--

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-09-06:
Sorry, the HueEmulation from Tasmota is only tested and supported with Alexa hardware.
For all other(s) there is no support. It looks like there is no valid UDP request for discovery.
There is no UDP "Hue" traffic

erichabg, 2020-09-06:
Tthere must be a change between V8.2. and V8.4.  But I did not found something in the release notes. Maybe you have an idea what was changed here?

s-hadinger, 2020-09-06:
There was no significant difference in Hue responses from v8.2 to v8.4.

The request `/128d57` sent by your app is wrong, it should start with `/api/`.

Can you send the equivalent logs with v8.2 where it shows it's successful?

erichabg, 2020-09-06:
Here is the log with V8.2.
13:43:51 CMD: weblog 4
13:43:51 MQT: EFHome/clients/Laser/status/Laser/1/RESULT = {"WebLog":4}
13:43:51 UDP: Packet (101)
13:43:51 HTP: Hue-Setup
13:43:51 CFG: in Flash gespeichert am FB, zählen 363, Bytes 4096
13:43:53 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
13:43:54 UDP: Packet (101)
13:43:56 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
13:43:57 UDP: Packet (101)
13:43:57 UDP: Packet (174)
13:43:57 HTP: Hue-API (/config)
13:43:58 UDP: Packet (174)
13:43:59 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
13:43:59 UDP: Packet (174)
13:44:00 UDP: Packet (174)
13:44:01 UDP: Packet (101)
13:44:01 HTP: Hue-Setup
13:44:02 HTP: Hue-API (/config)
13:44:02 HTP: Hue-API ()
13:44:02 HTP: Hue POST-Argumente ({"devicetype":"BoschSmartHome#SmartHomeController"})
13:44:02 HTP: Hue Authentication Result ([{"success":{"username":"79d0ba"}}])
13:44:02 HTP: Hue-API (/79d0ba)
13:44:02 HTP: Hue User-Agent: BOSCH-SMART-HOME-HUE-BRIDGE-CLIENT, gen=2
13:44:03 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
13:44:04 WIF: Prüfe Verbindung...
13:44:04 UDP: Packet (101)
13:44:06 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
13:44:07 HTP: Hue-API (/79d0ba)
13:44:07 HTP: Hue User-Agent: BOSCH-SMART-HOME-HUE-BRIDGE-CLIENT, gen=2
13:44:07 UDP: Packet (101)
13:44:10 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
13:44:11 UDP: Packet (123)
13:44:11 UDP: Packet (101)
13:44:11 HTP: Hue-Setup
13:44:12 HTP: Hue-API (/79d0ba)
13:44:12 HTP: Hue User-Agent: BOSCH-SMART-HOME-HUE-BRIDGE-CLIENT, gen=2
13:44:13 HTP: Hue-API (/79d0ba/groups/)
13:44:13 HTP: Hue POST-Argumente ({"name":"BOSCH_SMART_HOME_INTRUSION","lights":["127732641"]})
13:44:13 HTP: Hue Authentication Result ([{"success":{"username":"79d0ba"}}])
13:44:14 HTP: Hue-API (/79d0ba/lights/127732641/state)
13:44:14 HTP: Hue POST-Argumente ({"alert":"lselect"})
13:44:14 HTP: Hue Result ([{"error":{"type":901,"address":"/","description":"Internal Error"}}])
13:44:14 UDP: Packet (101)
13:44:16 UDP: Packet (123)
13:44:16 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
13:44:17 HTP: Hue-API (/79d0ba)
13:44:17 HTP: Hue User-Agent: BOSCH-SMART-HOME-HUE-BRIDGE-CLIENT, gen=2
13:44:17 UDP: Packet (101)
13:44:19 HTP: Hue-API (/79d0ba/lights/127732641/state)
13:44:19 HTP: Hue POST-Argumente ({"alert":"none"})
13:44:19 HTP: Hue Result ([{"error":{"type":901,"address":"/","description":"Internal Error"}}])
13:44:19 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
13:44:21 UDP: Packet (123)
13:44:21 UDP: Packet (101)
13:44:22 HTP: Hue-Setup
13:44:22 HTP: Hue-API (/79d0ba)
13:44:22 HTP: Hue User-Agent: BOSCH-SMART-HOME-HUE-BRIDGE-CLIENT, gen=2
13:44:23 UDP: Packet (167)
13:44:24 WIF: Prüfe Verbindung...
13:44:24 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
13:44:24 UDP: Packet (167)
13:44:25 UDP: Packet (101)
13:44:25 UDP: Packet (167)
13:44:26 UDP: Packet (167)
13:44:26 UDP: Packet (123)
13:44:27 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
13:44:27 HTP: Hue-API (/79d0ba)
13:44:27 HTP: Hue User-Agent: BOSCH-SMART-HOME-HUE-BRIDGE-CLIENT, gen=2
13:44:28 UDP: Packet (101)

s-hadinger, 2020-09-06:
The UDP listener was changed in v8.3 and packets were limited to the following:

```
#define UDP_BUFFER_SIZE         120      // Max UDP buffer size needed for M-SEARCH message
#define UDP_MAX_PACKETS   3             // we support x more packets than the current one
```

From your logs it looks that v8.2 did received the UPNP packets but 8.4 didn't (or discarded it).

Are you able to compile your own version in increase UDP_BUFFER_SIZE to 200 and UDP_MAX_PACKETS to 6?

erichabg, 2020-09-06:
Yes, I can. I give it a try!

erichabg, 2020-09-06:
I did it, but unfortunately without success!
15:49:03 UDP: Packet (101)
15:49:03 HTP: Hue-Setup
15:49:04 UDP: Packet (123)
15:49:05 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:06 UDP: Packet (101)
15:49:08 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:08 UDP: Packet (123)
15:49:09 UDP: Packet (101)
15:49:11 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:11 WIF: Prüfe Verbindung...
15:49:13 UDP: Packet (101)
15:49:13 HTP: Hue-Setup
15:49:15 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:16 UDP: Packet (101)
15:49:18 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:19 UDP: Packet (101)
15:49:21 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:23 UDP: Packet (101)
15:49:23 HTP: Hue-Setup
15:49:25 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:26 UDP: Packet (101)
15:49:28 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:29 UDP: Packet (101)
15:49:30 HTP: Hue-API (/config)
15:49:30 HTP: Hue-API (/79d0ba)
15:49:30 HTP: Hue User-Agent: BOSCH-SMART-HOME-HUE-BRIDGE-CLIENT, gen=2
15:49:31 WIF: Prüfe Verbindung...
15:49:31 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:33 UDP: Packet (101)
15:49:33 HTP: Hue-Setup
15:49:35 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:36 UDP: Packet (101)
15:49:38 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:39 UDP: Packet (101)
15:49:41 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:43 UDP: Packet (101)
15:49:43 HTP: Hue-Setup
15:49:45 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:46 UDP: Packet (101)
15:49:48 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:49 UDP: Packet (101)
15:49:51 WIF: Prüfe Verbindung...
15:49:51 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:53 UDP: Packet (101)
15:49:53 HTP: Hue-Setup
15:49:55 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:56 UDP: Packet (101)
15:49:56 MQT: EFHome/clients/Laser/status/Laser/1/STATE = {"Time":"2020-09-06T15:49:56","Uptime":"0T00:01:12","UptimeSec":72,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"0","Wifi":{"AP":1,"SSId":"EFHOME2","BSSId":"08:96:D7:66:97:A5","Channel":1,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:06"}}
15:49:57 UDP: Packet (174)
15:49:58 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:49:58 UDP: Packet (174)
15:49:59 UDP: Packet (101)
15:49:59 UDP: Packet (174)
15:50:00 UDP: Packet (174)
15:50:01 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:50:03 UDP: Packet (101)
15:50:03 HTP: Hue-Setup
15:50:05 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:50:06 UDP: Packet (101)
15:50:08 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:50:09 UDP: Packet (101)
15:50:11 WIF: Prüfe Verbindung...
15:50:12 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:50:13 UDP: Packet (101)
15:50:13 HTP: Hue-Setup
15:50:15 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:50:16 UDP: Packet (101)
15:50:19 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:50:19 UDP: Packet (101)
[... 78 lines cut ...]
15:51:10 UDP: Packet (101)
15:51:11 WIF: Prüfe Verbindung...
15:51:12 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:51:14 UDP: Packet (101)
15:51:14 HTP: Hue-Setup
15:51:16 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:51:17 UDP: Packet (101)
15:51:19 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:51:20 UDP: Packet (101)
15:51:22 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:51:24 UDP: Packet (101)
15:51:24 HTP: Hue-Setup
15:51:26 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:51:27 UDP: Packet (101)
15:51:29 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:51:30 UDP: Packet (101)
15:51:31 WIF: Prüfe Verbindung...
15:51:32 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:51:34 UDP: Packet (101)
15:51:34 HTP: Hue-Setup
15:51:36 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:51:37 UDP: Packet (101)
15:51:39 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851
15:51:40 UDP: Packet (101)
15:51:42 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.128:55851

s-hadinger, 2020-09-06:
Thanks, actually I have another idea. You currently have 8 relays declared, can you please reduce to 6 relays and check if it works?

The response in v8.4 contains two additional fields `name` and `modelid` and it can make the response too big and truncated.

erichabg, 2020-09-06:
No, that doesn't solve the problem. And I have also tested with Sonoff S2x (no additional GPIOs configured) and there is the same behaviour.

erichabg, 2020-09-06:
I have now removed the name and the modelid from the source code
const char HUE_LIGHTS_STATUS_JSON2[] PROGMEM =
  ",\"type\":\"Extended color light\","
 // E.F. "\"name\":\"%s\","
 // E.F. "\"modelid\":\"%s\","
  "\"manufacturername\":\"%s\","
  "\"uniqueid\":\"%s\"}";

and have also changed the places in the source where this struct was used, but the behaviour is the same.

(I'm now out for some days, will be back next weekend)

s-hadinger, 2020-09-06:
I'm out of ideas.

ascillato2, 2020-09-13:
Closing this issue until the user has time for more testings.

Thanks.

closed 2020-09-13 by ascillato2