arendst/Tasmota #9254: Compile error framework not found

kugelkopf123, 2020-09-06:
### PROBLEM DESCRIPTION
PlatformIO does not compile because it uses "https://github.com/Jason2866/platform-espressif8266/releases/download/2.9.0/framework-arduinoespressif8266-3.20900.0.tar.gz"

does not find.

Error in Platformio Console:
```
Processing tasmota (platform: espressif8266@2.6.2; framework: arduino; board: esp01_1m)
----------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing framework-arduinoespressif8266 @ *
git version 2.17.2 (Apple Git-113)
Cloning into '/Volumes/Macintosh SD/Users/christopherfleuth/.platformio/packages/_tmp_installing-qcFuLd-package'...
remote: Repository not found.
fatal: repository 'https://github.com/jason2866/framework-arduinoespressif8266/' not found
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/jason2866/framework-arduinoespressif8266', '/Volumes/Macintosh SD/User/christopherfleuth/.platformio/packages/_tmp_installing-qcFuLd-package']]
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Platformio_override.ini:

```
; ***  Example PlatformIO Project Configuration Override File   ***
; ***  Changes done here override settings in platformio.ini    ***
;
; *****************************************************************
; ***  to activate rename this file to platformio_override.ini  ***
; *****************************************************************
;
; Please visit documentation for the options and examples
; http://docs.platformio.org/en/stable/projectconf.html

[platformio]
extra_configs = platformio_tasmota_cenv.ini

; *** Build/upload environment
default_envs =
; *** Uncomment the line(s) below to select version(s)
                tasmota
;                tasmota-debug
;                tasmota-ircustom
;                tasmota-minimal
;                tasmota-lite
;                tasmota-knx
;                tasmota-sensors
;                tasmota-display
;                tasmota-zbbridge
;                tasmota-ir
;                tasmota32
;                tasmota32-webcam
;                tasmota32-minimal
;                tasmota32-lite
;                tasmota32-knx
;                tasmota32-sensors
;                tasmota32-display
;                tasmota32-ir
;                tasmota32-ircustom

[common]
platform                  = ${core.platform}
platform_packages         = ${core.platform_packages}
build_unflags             = ${core.build_unflags}
build_flags               = ${core.build_flags}
; *** Optional Debug messages
;                            -DDEBUG_TASMOTA_CORE
;                            -DDEBUG_TASMOTA_DRIVER
;                            -DDEBUG_TASMOTA_SENSOR

; set CPU frequency to 80MHz (default) or 160MHz
[... 146 lines cut ...]
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Download Tasmota Dev Branch. Rename user_config_override_sample.h to user_config_override.h and rename platformio_override_sample.h to platformio_override.h

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

eku, 2020-09-06:
Yes, I can confirm that.

Jason2866, 2020-09-06:
Update to latest Platformio Release v.5.0

kugelkopf123, 2020-09-06:
OK. For ESP8266 does it work. As far as I see. But for ESP32 not. 

```
*** use provided user_config_override.h as planned ***
*** [tasmota/StackThunk_light.cpp] /Volumes/Macintosh SD/Benutzer/christopherfleuth/Downloads/Github/Tasmota/.pioenvs/tasmota32/.sconsign27.dblite: No such file or directory
===================================================== [FAILED] Took 26.15 seconds =====================================================

Environment    Status    Duration
-------------  --------  ------------
tasmota32      FAILED    00:00:26.147
================================================ 1 failed, 0 succeeded in 00:00:26.147 ================================================
The terminal process "platformio 'run', '--environment', 'tasmota32'" terminated with exit code: 1.
```

eku, 2020-09-07:
> Update to latest Platformio Release v.5.0
How do you know if the error message is not appropriate?

kugelkopf123, 2020-09-07:
Tried the ESP8266 with a fresh install of vsCode, Platformio core 5 and TasmotaDevBranch Download on another MAC.
I havent changed anything(nothing in user_config_override.h or platformio_override.ini). Just downloaded it, en tried to compile. 
Instantly get this Error while building for esp8266:

```
> Executing task: platformio run --environment tasmota-DE <

Processing tasmota-DE (platform: espressif8266@2.6.2; framework: arduino; board: esp01_1m)
-------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 (2.6.2) > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 2.7.4+1 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting tasmota.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 83 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <BearSSL> 0.6
|-- <Adafruit CCS811 Library> 1.0.0
|   |-- <Wire> 1.0
|-- <Adafruit MAX31865 library> 1.0.1
|   |-- <SPI> 1.0
|-- <Adafruit MCP9808 Library> 1.1.2
|   |-- <Wire> 1.0
|-- <Adafruit SGP30 Sensor> 1.2.0
|   |-- <Wire> 1.0
|-- <Adafruit VEML7700 Library> 1.0.0
|   |-- <Adafruit BusIO> 1.0.10
|   |   |-- <Wire> 1.0
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <ESP8266Audio> 1.5.0
|   |-- <SPI> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <LittleFS(esp8266)> 0.1.0
|   |-- <SD(esp8266)> 2.0.0
|   |   |-- <SDFS> 0.1.0
|   |   |   |-- <SPI> 1.0
|   |   |   |-- <ESP8266SdFat> 1.1.0
|   |   |   |   |-- <SPI> 1.0
|   |-- <SDFS> 0.1.0
|   |   |-- <SPI> 1.0
|   |   |-- <ESP8266SdFat> 1.1.0
|   |   |   |-- <SPI> 1.0
|   |-- <ESP8266SdFat> 1.1.0
|   |   |-- <SPI> 1.0
|-- <I2Cdevlib-Core>
|   |-- <Wire> 1.0
|-- <LibTeleinfo> 1.1.3
|-- <I2Cdevlib-MPU6050>
|   |-- <I2Cdevlib-Core>
|   |   |-- <Wire> 1.0
|-- <Mutichannel_Gas_Sensor> 0.0.1
|   |-- <Wire> 1.0
|-- <ArduinoNTPd>
|   |-- <ESP8266WiFi> 1.0
|-- <OpenTherm Library> 0.9.0
|-- <SD(esp8266)> 2.0.0
|   |-- <SDFS> 0.1.0
|   |   |-- <SPI> 1.0
[... 189 lines cut ...]
  File "/Users/christopher/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/Main.py", line 1326:
    SCons.SConsign.write()
  File "/Users/christopher/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/SConsign.py", line 114:
    syncmethod()
  File "/Users/christopher/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/dblite.py", line 130:
    f = self._open(self._tmp_name, "wb", self._mode)
Exception ignored in: <function dblite.__del__ at 0x100ec9200>
Traceback (most recent call last):
  File "/Users/christopher/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/dblite.py", line 126, in __del__
  File "/Users/christopher/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/dblite.py", line 123, in close
  File "/Users/christopher/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/dblite.py", line 130, in sync
FileNotFoundError: [Errno 2] No such file or directory: '/Users/christopher/Downloads/Tasmota-development-2/.pioenvs/tasmota-DE/.sconsign37.tmp'
==================================================== [FAILED] Took 12.14 seconds ====================================================

Environment    Status    Duration
-------------  --------  ------------
tasmota-DE     FAILED    00:00:12.139
=============================================== 1 failed, 0 succeeded in 00:00:12.139 ===============================================
The terminal process "platformio 'run', '--environment', 'tasmota-DE'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
 
```

BTW on this Mac same problem with the ESP32.

Jason2866, 2020-09-07:
The Install and Setup from your PlatformIO is complete and successfull
[image]
It does run on MAC. Our zigbee developer @s-hadinger  does use a MAC too. He has no issue

Jason2866, 2020-09-07:
Delete folder `tool-scons`. You have a old version. The actual is `tool-scons @ 4.40001.0`
If this doesnt help delete the complete folder `.platformio` with ALL subfolders.

eku, 2020-09-07:
The update to version 5.0 has indeed fixed the problems, at least for me.

Jason2866, 2020-09-07:
Anyway. The issue is NOT Tasmota related. Please close this issue.
If the issue is still there for you please open a issue in platformio github.

closed 2020-09-07 by ascillato2

kugelkopf123, 2020-09-07:
If I delete the .platformio folder in my home directory, the next time I compile (ESP32), the outdated version will be installed again. 
Only the packages/tools-scons folder needs to be deleted from the .platformio folder. The next time you compile, the latest version will be loaded. 
Thanks for your help!

kugelkopf123, 2020-09-08:
@ascillato2 
There is still this problem left:
`*** [tasmota/StackThunk_light.cpp] /Volumes/Macintosh SD/Benutzer/christopher/Downloads/Github/Tasmota/.pioenvs/tasmota32-DE/.sconsign37.dblite: No such file or directory
`
And by the way, its not possible to "update" to 4.0.4.
Perhaps you could have a look into your .platformio Folder. In my i find tools-scons/scons-local-4.0.1 and i cant do anything on this. Everytime i try to build for esp32 it gets overwritten with this version. 
Btw. Tried latest Dev this morning,too. Same result.

Jason2866, 2020-09-08:
@kugelkopf123  the issue is closed correctly. The issue you have is not caused from Tasmota.
The Problem you have is PlatformIO related. You can try latest Tasmota development version. I have updated the espressif32 platformio framework to latest 2.0.0. version. Maybe it helps.
If not the ONLY way is opening a issue in platformio github

kugelkopf123, 2020-09-08:
@Jason2866 
Ok. Ive tried that. The previous Post was the result of the new esp32 v. 2.0.0 frameworks.
It's just strange that this error occured right after an update to the current version of the Tasmota Dev.
So nothing else was installed or changed. Only the Tasmota project folder was updated via Github Desktop.