arendst/Tasmota #9256: Sonoff ZigBee Join issue

webfrank, 2020-09-07:
### PROBLEM DESCRIPTION
After sending "ZbPermitJoin 1" and pairing device I expect a "Disable Pairing" message.

If I send "ZbPermitJoin 0" which should disable joining I receive a "Pairing mode enabled" message.

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff ZigBee Gateway
- [X] Tasmota binary firmware version number used: 8.4.0.3
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: esptool and OTA (as per documentation)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

07:03:45 CMD: Backlog Template; Module; GPIO 255
07:03:45 MQT: zbgateway/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
07:03:46 MQT: zbgateway/stat/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
07:03:46 MQT: zbgateway/stat/RESULT = {"GPIO0":{"56":"Led1i"},"GPIO1":{"165":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"166":"Zigbee Rx"},"GPIO4":{"215":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"158":"LedLinki"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"17":"Button1"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

07:04:36 CMD: status 0
07:04:36 MQT: zbgateway/stat/STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"zbgateway","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:04:36 MQT: zbgateway/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T13:10:38","StartupUTC":"2020-09-06T16:53:58","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2020-09-03T18:56:39","SaveCount":28,"SaveAddress":"F7000"}}
07:04:36 MQT: zbgateway/stat/STATUS2 = {"StatusFWR":{"Version":"8.4.0.3(zbbridge)","BuildDateTime":"2020-09-03T13:31:06","Boot":31,"Core":"2_7_4_1","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"341/699"}}
07:04:36 MQT: zbgateway/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["CASA",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
07:04:36 MQT: zbgateway/stat/STATUS4 = {"StatusMEM":{"ProgramSize":530,"Free":472,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F100786","04500001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,21,23,41","Sensors":""}}
07:04:36 MQT: zbgateway/stat/STATUS5 = {"StatusNET":{"Hostname":"zbgateway","IPAddress":"192.168.1.109","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"C8:2B:96:05:E7:35","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
07:04:36 MQT: zbgateway/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_05E735","MqttUser":"DVES_USER","MqttCount":17,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
07:04:36 MQT: zbgateway/stat/STATUS7 = {"StatusTIM":{"UTC":"2020-09-07T06:04:36","Local":"2020-09-07T07:04:36","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"06:16","Sunset":"19:19"}}
07:04:36 MQT: zbgateway/stat/STATUS10 = {"StatusSNS":{"Time":"2020-09-07T07:04:36"}}
07:04:36 MQT: zbgateway/stat/STATUS11 = {"StatusSTS":{"Time":"2020-09-07T07:04:36","Uptime":"0T13:10:38","UptimeSec":47438,"Vcc":3.278,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":17,"Wifi":{"AP":1,"SSId":"CASA","BSSId":"B8:3A:08:E8:73:D1","Channel":6,"RSSI":64,"Signal":-68,"LinkCount":23,"Downtime":"0T00:01:28"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

06:57:07 CMD: zbpermitjoin 0
06:57:07 MQT: zbgateway/stat/RESULT = {"ZbPermitJoin":"Done"}
06:57:07 MQT: zbgateway/tele/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
06:57:07 ZIG: {"ZbEZSPReceived":"4500050000360000000000000071FF000000FFFF03050001"}
06:57:07 ZIG: Internal ZDO message 0x0036 sent from 0x0000 (broadcast)
```

### TO REPRODUCE
Issue ZbPermitJoin 1

wait 60 seconds

expect "Disabled pairing mode" (never arrives)

Issue ZbPermitJoin 0

expect "Disabled pairing mode"

### EXPECTED BEHAVIOUR
Pairing mode should be disabled after configured period or when 0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

webfrank, 2020-09-08:
Hi, the same happens with 8.5.0 just upgraded.

s-hadinger, 2020-09-08:
Hi, unfortunately EZSP never sends any message to signal that pairing is disabled. I can't change it.

I understand therefore that the message is misleading, but `ZbPermitJoin 0` enables pairing for 0 seconds, which is supposed to disable pairing - although I didn't cross check if it actually stops a current pairing enabled.

derFrickler, 2020-09-23:
Anyhow, would be great to have a feedback of the state of the bridges
Would it possible that Tasmota generates the {"ZbState":{"Status":20,"Message":"Pairing mode disabled "}} message after the time has passed?

s-hadinger, 2020-09-24:
Yes will do. But only based on timer, which should be ok.

s-hadinger, 2020-09-24:
@derFrickler This is done, via #9378

derFrickler, 2020-09-26:
Works great!

ascillato2, 2020-09-26:
Closing this issue as it has been fixed.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-09-26 by ascillato2